bigtreetech/BIGTREETECH-SKR-mini-E3 #147: Hot end fans both always running at 100% speed

MrSnoebel, 2019-12-09:
Need some help, made a good working setup for Ender 5 with BLtouch v3.1.
All is working just fine, but the two fans are both running full speed all the time.

Would like to be able to control at least the part cooling fan, but need some help or direction how to do this.

Added both my config and advanced config.
[Configs.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/3941704/Configs.zip)

pmjdebruijn, 2019-12-09:
Did you wire them the right way around? If I recall correctly it's swapped around compared to the original Melzi board.

MrSnoebel, 2019-12-10:
I saw indeed also an post on this, but I have both fans running at full speed. So even swapping the wires would make no difference in this case.
I'm wondering if it has something to do with the suggested FAN_SOFT_PWM 0 I enabled, I thought this was a must for enabling the bltouch ?

pmjdebruijn, 2019-12-10:
The swapping of the fans does matter, as one is for the case fan, not the hotend, so it might just be the case that you're PWMing your casefan, as opposed to the part cooling fan.

FAN_SOFT_PWM is nice to have, as it seems to remove your high pitched noise, although you shouldn't define it with a 0 value. Not sure if that really matters though.

BTW, I have BLTouch enabled firmware available here (**DO** read the README):
https://github.com/pmjdebruijn/BIGTREETECH-SKR-mini-E3-V1.2/

MrSnoebel, 2019-12-21:
I feel stupid. I did see many reports that that fans needed to be swapped. I swapped the two fans used on the hot end. This didn't work. But you need to swap the fan for the motherboard fan and the hot end fan. The fan for motherboard is now always on. But the second fan on the hot end works now as intended. Issue can be closed.

closed 2019-12-21 by MrSnoebel