bigtreetech/BIGTREETECH-SKR-mini-E3 #148: Bl Touch and Filamentsensor

oOJoshOo, 2019-12-12:
How can I Install a BL Touch and a Filament sensor?

tatusah, 2019-12-13:
Connect filament sensor to the E0-STOP port and for use BLTouch SERVOS port + either Z-STOP or PROBE port.

oOJoshOo, 2019-12-15:
But the Endstop I have, has 3 pins and the E0-Stop has only two? And Marlin says to me, that I have to put the sensor in the servo 3 port.

tatusah, 2019-12-15:
Some of the filament sensors have only 2 wires and some 3. Even most of the 3 wire versions work with 2 wires.

The default config for SKR e3 mini & mini v1.2 sets the filament detect pin to PC15 which is the 2 pin port. Check Marlin/src/pins/stm32/pins_BTT_SKR_MINI_E3.h file

```
//
// Filament Runout Sensor
//
#ifndef FIL_RUNOUT_PIN
  #define FIL_RUNOUT_PIN   PC15   // "E0-STOP"
#endif
```

So either you find out which 2 pins are needed and customize the wiring. Alternatively if you want to use 3 wires you could install the filament sensor to PT-DET port and customize the pin config file. Refer to the wiring diagram of the specified filament sensor you have and the [control board pins schematics](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V1.2/BTT%20SKR%20MINI%20E3%20V1.2PIN.pdf).

oOJoshOo, 2019-12-15:
Ok I've changed the pin to PC12, checked if the sensor is aligned with the pins but it seems not to work.

oOJoshOo, 2019-12-15:
ok it also doesn't work on PC15 with two wires, and I've checked the sensor on its functionality

tatusah, 2019-12-15:
Check the state of endstops / filament sensor with M119 gcode. It might be that the sensor needs its state inverted. What sensor model do you have?

oOJoshOo, 2019-12-15:
I use the ender 3 z-sensor. But when he is just inverted shouldn't he activate the M600 GCode when I press it

tatusah, 2019-12-15:
I suppose you didn't forget to enable the `#define FILAMENT_RUNOUT_SENSOR` by uncommenting that line from the configuration.h file. If you have the firmware ok then the M119 command should help you to see whether you need to either invert the behaviour of the filament endstop in the firmware or alternatively by changing the wires between normally open / normally closed pin on the endstop switch. If you read the comments on the filament sensor feature in the configuration.h file you see that the default behaviour is that the filament runout pin is pulled high with the pullup resistor when the the filament is in the sensor. So this expects that the microswitch circuit should be open when filament is present and when filament is removed the switch would close the circuit connecting the sensor pin to GND.

I've just meen changing the extruder to DD and don't have the filament sensor installed currently but I can check it later. However, previously it's been working with the above logic. But start with the M119 to figure out the state of the switches.

oOJoshOo, 2019-12-15:
No, I didn't forget to uncomment that, M119 says that the Sensor is triggered when filament is in it that should also matching to my settings.
He opens when the sensor does.
But when I'm printing the printer does nothing when I release the trigger

oOJoshOo, 2019-12-15:
Ok, I don't know what happened, but now it works, and that without any change. Thank you very much.

closed 2019-12-15 by oOJoshOo