bigtreetech/BIGTREETECH-SKR-mini-E3 #149: [BUG] Flash impossible SKR mini E3 V1.2

Rhteam67, 2019-12-12:
Hello,

Same problem here, I managed to compile a firmware with platformio since marlin bugfix 2.0.
I put firmware.bin on my SD card and she installed marlin.

Now I just installed a bltouch so I take the firmware available by Bigtreetech and I put it on my pristine SD but the card does not want to install my firmware, it starts with the old firmware.

I have the impression that the boot loader does not want to read the SD

brew99, 2019-12-12:
If you were able to install marlin bugfix-2.0.x just fine, why don't you just alter your configuration.h to enable the BLtouch?   If you use the example config for the V1.2 from the marlin download as a starting config, then just enable these:

#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN    //assume using the probe as end stop
#define BLTOUCH
#define NOZZLE_TO_PROBE_OFFSET  // set according to your position
#define AUTO_BED_LEVELLING_BILINEAR
#define Z_SAFE_HOMING

try compiling in platformio, and see if it flashes.   Play will some of the BLtouch settings in configuration_adv.h to get your probe working if the defaults don't work.

I don't use BTT version of firmware, and prefer to alter the vanilla marlin

mikemgg123, 2019-12-15:
This is a problem with platformio if that is what you're using to compile. try downgrading that to see if that fixes the flash. open a terminal within the platform atom session and type 'sudo pip install -U "platformio<4.1"'

Rhteam67, 2019-12-15:
A big thank you to you. I finally managed to flash my card, the solution was to check the SD card from right click property.
But that I compile and flash the card to activate my bltouch, this one does a leveling in the middle and after a bottom left of the board but outside of it.
I have configured my #define NOZZLE_TO_PROBE_OFFSET

Rhteam67, 2019-12-15:
[image]

Rhteam67, 2019-12-15:
[image]

mikemgg123, 2019-12-18:
have you configured your offset?

Rhteam67, 2019-12-26:
Oui, offset Z bien configuré. Mais est ce que la tête E3DV6 demande a modifier l'offset Z ? Car le plateau tape contre le tendeur de courroie.
Peut on modifier l'offset Y ?

Aksthakor, 2019-12-29:
Hello, Can anyone help

I am also facing similar kind of issue

Not able to compile the BLTouch enabled code via VScode, getting below error, if i install the stock firmware with BLtouch z off set then, firmware gets installed on the ender3 pro but not able to print anything as BL touch is taking measurements but not started to print anything, i have to update Z_STOP_PIN PC2 To Z_STOP_PIN PC14 to try that setting but no luck compiling the code with changes, found above settings on below URL 

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/6

`Processing * in STM32F103RC_bigtree_NOUSB environment
---------------------------------------------------------------------------------------------------------------------------------------
Building...
PACKAGES: tool-stm32duino 1.0.1, toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0)
arm-none-eabi-ar: .pio/build/STM32F103RC_bigtree_NOUSB/libde0/U8glib-HAL_ID1932/U8glib.cpp.o: No such file or directory
*** [.pio/build/STM32F103RC_bigtree_NOUSB/libde0/libU8glib-HAL_ID1932.a] Error 1
OSError: [Errno 2] No such file or directory: '/Users/a1/Downloads/BIGTREETECH-SKR-mini-E3-master/firmware/V1.2/Marlin-2.0.x-SKR-Mini-E3-V1.2/.pio/build/STM32F103RC_bigtree_NOUSB/.sconsign.py27.dblite':
  File "/Users/a1/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1346:
    _exec_main(parser, values)
  File "/Users/a1/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1309:
    _main(parser)
  File "/Users/a1/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1091:
    nodes = _build_targets(fs, options, targets, target_top)
  File "/Users/a1/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1283:
    jobs.run(postfunc = jobs_postfunc)
  File "/Users/a1/.platformio/packages/tool-scons/script/../engine/SCons/Job.py", line 113:
    postfunc()
  File "/Users/a1/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1280:
    SCons.SConsign.write()
  File "/Users/a1/.platformio/packages/tool-scons/script/../engine/SCons/SConsign.py", line 109:
    syncmethod()
  File "/Users/a1/.platformio/packages/tool-scons/script/../engine/SCons/dblite.py", line 129:
    self._os_unlink(self._file_name)
Exception OSError: (2, 'No such file or directory', '/Users/a1/Downloads/BIGTREETECH-SKR-mini-E3-master/firmware/V1.2/Marlin-2.0.x-SKR-Mini-E3-V1.2/.pio/build/STM32F103RC_bigtree_NOUSB/.sconsign.py27.dblite') in <bound method dblite.__del__ of <SCons.dblite.dblite object at 0x10de5fb90>> ignored
===================================================== [FAILED] Took 74.03 seconds =====================================================

Test    Environment                     Status    Duration
------  ------------------------------  --------  ------------
*       megaatmega2560                  IGNORED
*       megaatmega1280                  IGNORED
*       at90usb1286_cdc                 IGNORED
*       at90usb1286_dfu                 IGNORED
*       DUE                             IGNORED
*       DUE_USB                         IGNORED
*       DUE_debug                       IGNORED
*       LPC1768                         IGNORED
*       LPC1769                         IGNORED
*       sanguino_atmega644p             IGNORED
*       sanguino_atmega1284p            IGNORED
*       melzi                           IGNORED
*       melzi_optiboot                  IGNORED
*       rambo                           IGNORED
*       STM32F103RE                     IGNORED
*       STM32F103RC_fysetc              IGNORED
*       STM32F103RC_bigtree             IGNORED
*       STM32F103RC_bigtree_NOUSB       FAILED    00:01:14.027
*       STM32F103RC_bigtree_512K        IGNORED
*       STM32F103RC_bigtree_512K_NOUSB  IGNORED
*       STM32F103RE_bigtree             IGNORED
*       STM32F103RE_bigtree_NOUSB       IGNORED
*       STM32F4                         IGNORED
*       STM32F7                         IGNORED
*       ARMED                           IGNORED
*       STM32F103VE_longer              IGNORED
*       mks_robin                       IGNORED
*       mks_robin_lite                  IGNORED
*       mks_robin_mini                  IGNORED
*       mks_robin_nano                  IGNORED
*       jgaurora_a5s_a1                 IGNORED
*       STM32F407VE_black               IGNORED
*       BIGTREE_SKR_PRO                 IGNORED
*       BIGTREE_BTT002                  IGNORED
*       teensy31                        IGNORED
*       STM32F103CB_malyan              IGNORED
*       chitu_f103                      IGNORED
*       teensy35                        IGNORED
*       esp32                           IGNORED
*       fysetc_f6_13                    IGNORED
*       linux_native                    IGNORED
*       SAMD51_grandcentral_m4          IGNORED
*       include_tree                    IGNORED
================================================ 1 failed, 0 succeeded in 00:01:14.027 ================================================
The terminal process terminated with exit code: 1
`

brew99, 2019-12-30:
If you are using the BLtouch and using the probe as the end stop, then I suggest you use the ZMin stop position on the board (the same as the old end stop switch) and keep the default PC2.  

Try moving the Marlin folder closer to the root of your hard drive rather than nested so far down.

SwiCago, 2020-01-01:
I compiled for an ender5 using STM32F103RC_bigtree_512K(someone made a pull in regards to this) and my configs(made pull for those), compiled fine and flashed fine. Currently doing a test print.

rarkhem, 2020-01-06:
> If you were able to install marlin bugfix-2.0.x just fine, why don't you just alter your configuration.h to enable the BLtouch? If you use the example config for the V1.2 from the marlin download as a starting config, then just enable these:
> 
> #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN //assume using the probe as end stop
> #define BLTOUCH
> #define NOZZLE_TO_PROBE_OFFSET // set according to your position
> #define AUTO_BED_LEVELLING_BILINEAR
> #define Z_SAFE_HOMING
> 
> try compiling in platformio, and see if it flashes. Play will some of the BLtouch settings in configuration_adv.h to get your probe working if the defaults don't work.
> 
> I don't use BTT version of firmware, and prefer to alter the vanilla marlin
> 
> Also, if you are getting nowhere with this, it might be a good idea for you to pick-up an ST-Link V2 clone device so that you can directly read/write to the MCU

Do you have a bootloader for mini e3 v1.2? I want to flash stlink.

HVJ17, 2020-01-15:
This is likely due to your (NOT A BLTOUCH) 3dtouch sensor to nozzle offsets. 

make sure to have 
#define X_BED_SIZE 235
#define Y_BED_SIZE 235

and adjust 
/**
 * Z Probe to nozzle (X,Y) offset, relative to (0, 0).
 *
 * In the following example the X and Y offsets are both positive:
 *
 *   #define NOZZLE_TO_PROBE_OFFSET { 10, 10, 0 }
 *
 *     +-- BACK ---+
 *     |           |
 *   L |    (+) P  | R <-- probe (20,20)
 *   E |           | I
 *   F | (-) N (+) | G <-- nozzle (10,10)
 *   T |           | H
 *     |    (-)    | T
 *     |           |
 *     O-- FRONT --+
 *   (0,0)
 *
 * Specify a Probe position as { X, Y, Z }
 */
#define NOZZLE_TO_PROBE_OFFSET { -44, -6, 0 } // values for BLTOUCH on stock Ender 3 pro