bigtreetech/BIGTREETECH-SKR-mini-E3 #150: [BUG} Description for compiling latest Marlin

okuemmel, 2019-12-15:
according to this [issue](https://github.com/MarlinFirmware/Marlin/tree/f786cc5145ddccdae02c29b0ccda315c86c159d9) 
`default_envs = STM32F103RC_bigtree_NOUSB `
needs to be changed to 
`default_envs = STM32F103RC_bigtree`

brew99, 2019-12-16:
This is not really a "BUG" with BTT's current version of Marlin.   The default_envs = STM32F103RC_bigtree_NOUSB is still valid for the current platformio on BTT github (21 days ago).

If you are compiling from Vanilla Marlin, then yes that env is no longer valid, and needs to be changed.

FrankM1916, 2020-02-18:
but hold on now ... im having the same issue with my compile ending in error code 1 except Ive tried both STM32F103RC_bigtree as well as STM32F103RC_bigtree_nousb as a matter of fact ive gone down the whole board env list trying all the stm32 variants all with the same issue in marlin 2.0.x and in the bugfix maRLIN AS WELL AS THE PRECONFIGURED  version of marlin bigtree posts here labeled skr_mini_e3_v1.2 master nothing will compile but when i switch to the atmega chip that was in the platformio file is there no support sor the stm chips ive literally tried compiling every one thats in the platformio.ini file in every version of marlin supporting 32bit processors i mean thats y im using vscode anyway otherwise id just be using the arduino ide ohh and yeah i tried this in atom as well same issue

BtreeEnderPalette, 2020-02-18:
In config/examples/bigtreetech/SKR1.2 folder, copy the 2 config files 1st.

Best to start from fresh unzips, in platformio paste STM321F103RC_bigtree as default env. 

My E3 1.2 hasn't the memory to deal with my BL Touch v3. I'd really like ABL active but seems to complie ok with #define MESH_BED_LEVELING .

I'm using the BTT TFT3.5 touch screen but I haven't figured out how to run the classic mode so I'm not sure how the MBL interface would be.

BtreeEnderPalette, 2020-02-19:
Apparently the 512l version can accommodate probe