bigtreetech/BIGTREETECH-SKR-mini-E3 #151: Controller Fan Speed

brianamato, 2019-12-16:
Been struggle to find a way to modify to the controller board (chassis) fan speed.  It seems to be running at max speed all the time.  It doesn't seem that there is any PWM control available to that pin based on what I was seeing in the schematic.  It just makes the unit so loud and sound level was a small reason for purchasing this board in the first place.  Has anyone been able to tweak this?  If you have any insight I would appreciate it.

Thanks,

Brian

baub26, 2019-12-18:
Hello,
I've bought a DC-DC buck converter (amazon, 5 for 6€) and I've lowered the voltage of the fan to make it less noisy. I think there is no other solution until fan is powered by 24V rail. . I have also printed a new cover for the motherboard case so the fan is closer to the steppers drivers and MOSFET. Sorry for my english (I'm French)
Regards
Bernard

Michalczyk-T, 2020-01-22:
Hey I have a similar issue.
It's nice that the board has silent TMC 2209 drivers but the mainboard fan is always on and makes my printer extremely loud. I read different things about it but it looks like the fan is connected to constant 24V. But when I switch on my printer the fan starts spinning with maybe 70% speed and after about 20 seconds it goes up to full speed. 
So I guess there is a controller for the fan.
Can someone tell me more about it?
The creality board v1.1.5 is much quieter. It the speed of the mainboard controlled/triggered with an event, or temperature, or current draw?
Thanks for the help.

ciidfrance, 2020-05-11:
Same issue and no solution at this time...