bigtreetech/BIGTREETECH-SKR-mini-E3 #153: [BUG] Board not honoring -Z offset w/bltouch

my95z34, 2019-12-18:
### Description

I have the 1.2 board in my Ender 3 using the provided BLTouch as Z endstop firmware. 

### Steps to reproduce

Following the exact steps from this video: https://www.youtube.com/watch?v=y_1Kg45APko After re-enabling the soft endstops, I move z to 10 and back down to 0 and it's right back where it started, not to my setting.

Also, my offset is -0.13 

**Expected behavior**
Z0 should be including the probe offset. 

**Actual behavior**
It does not.

Sonnenwerk, 2019-12-31:
I also have some probs with z offset, although not sure about issue with the firmware. There are some config limits in marlin how "minus" z can be. When i go to printer also it seems to be dependend on the printer state if z offset works..this is an issue wich i am currently not sure is a error, or a misconfig for me..

my95z34, 2019-12-31:
I found a workaround. I added have disabled soft endstops in my start gcode. I believe it's m211 but I can't remember off the top of my head.

HVJ17, 2020-01-01:
This is a feature of is a Marlin 2.0.x and a known issue. You need to edit out the PRINTCOUNTER.
In configuration.h change it to //#define PRINTCOUNTER
This is still a bug that wipes config after the 2nd reboot.

Then you should be able to fix the probe offset. AUTO HOME, move z down,it will not go past 0, that is intended. Go back and fourth a few times with HOME Z and then Z probe offset.
Store settings.

TheGadge, 2020-01-11:
Printcounter disabled by default in 2.0.1, so it's not that, for me anyhow.

Multiple homing and storing, no change. I'm going back to bugfix of old until this seems fixed :(
I was sat there blaming my BL with multiple swear words until I saw a few on FB posing this!

HVJ17, 2020-01-11:
It is enabled in the Bigtreetech build. Anyone using the files or .bin found in this Bigtreetech fork has it enabled.
Just look in the menus, if there is a print counter, then its on.

TheGadge, 2020-01-12:
> It is enabled in the Bigtreetech build. Anyone using the files or .bin found in this Bigtreetech fork has it enabled.

Well, yeah, as that's not 2.0.1

TheGadge, 2020-01-13:
Changed to STM32F103RC_bigtree_512K from standard, seems to be able to save now!