bigtreetech/BIGTREETECH-SKR-mini-E3 #154: Skr mini E3 v1.2 won't update

Rogerz79, 2019-12-19:
[Firmware.bin](https://drive.google.com/file/d/15yRnLRBVZLbPgK2iFlQiT97Ylr5IJ3d3/view?usp=drivesdk) 

I am stuck at the moment. I've had the skr mini v1.2 and a bltouch 3.1 for a few weeks now and I was printing fine with no issues. I just got the new e3d hemera and have been trying to install it. I believe I have all of the wiring done correctly as I did one to one swaps. I made the changes in the firmware for the hemera, increased the max temp and change the temp to 05 from 01. First time the firmware seemed to update. The temp for the nozzel was at about 20°c at rest. I was able to run pid and get my numbers. I was able to calibrate extruder steps too. It didn't go to 409 like how I set it in the firmware, I had to do it manually on the printer. This was my first clue it didnt update the firmware. I change the firmware to the new pid values and compile again. No errors. I load the firmware.bin file to the SD card and pop it in the printer and the printer seems to update. It creates the .cur file.

 The nozzle reads a very low temp of 3°c. It is not that cold in my house. I haven't printed outside since the neighbors complained. I unplug the thermistor and plug in the old one and it reads normal 20°c. I plug in the new one and it goes back to 3°c. When I completely unplug it, it drops to -14°c. I send a M 503 to see the pid numbers and the bltouch offset numbers. They did not change to what I set it to in the firmware. I use vscode and pio to change the firmware. I keep copies of my firmware.bin files. I load up the one before all this started and and my temp is still 3°c. I read that sometimes you need to delete the whole build folder but it didnt work. I reformatted the SD card but still see the low temp. I tried a new sd card but still a no go. I pressed the reset button on the board but had no luck. It seems to me that my board is not properly loading up the new firmware and just keeping the old one and that's why my new thermostat reads so low. I'm not sure what else to try. Any help is appreciated and sorry for the long post

TLDR: After installing new hemerra hot end my nozzle temp reads 3°c. It seems to be because my board doesn't want to load the new firmware although it looks like it does.

brew99, 2019-12-19:
Just to be certain that it is loading the firmware, remove the cover and look for the rapid flashing LED when the firmware is flashing.   Should be about 10sec to load new firmware.   If that seems OK, then on LCD, go to CONFIGURATION, ADVANCED SETTINGS, INITIALIZE EEPROM, and select that.   Reboot printer and see if its OK.   Are you saving info/settings in the flash (EEPROM) or is it all set in the firmware?

Also in your firmware, search for PRINTCOUNTER (I think it's in configuration.h), and ensure that is commented out, as it was at one point causing the EEPROM to not save/retain settings (is now fixed in bugfix-2.0.x).

Rogerz79, 2019-12-20:
i reinitialized the eeprom and that seemed to do the trick. i can see the changes in the firmware now. BTW i did see the led flashing while it was programming before. My nozzle temp is still 3°C though. i need to try and re run pid but the temp is so low that it wont run. i am running bug-fix 2.0.x. print counter is commented out

Yosuncle, 2019-12-31:
there are like at least 3 different 100k thermistor options in Marlin. Each with a little different temperature/resistance curves -- and if you've accidentally selected a 50k thermistor, that's definitely a problem. Since you were able to confirm the temp is "correct" using the old one and the new one is still showing the too low temp, it is possible the thermistor is "malfunctioning" or just needs to be properly calibrated and/or setup in your firmware.

Rogerz79, 2019-12-31:
I did get new thermistor and that was the issue. Once the new one was installed the temp was correct.

closed 2019-12-31 by Rogerz79