bigtreetech/BIGTREETECH-SKR-mini-E3 #155: [BUG] TMC Connection Error

damacguy, 2019-12-22:
I've been trying to get a BLTouch setup with my Ender 3 and E3 Mini 1.2 board. But the firmware won't load and i get a TMC Connection Error.

What does this actual error mean?

I've tried compiling the latest firmware from this repo. I've also tried using the firmware.bin that's also already posted here. I've even tried using the firmware that came on the SD card that CAME with my board and was working before this. All give the same error. I'm disconnected the BLTouch from my board as well putting me back before I tried anything and neither my compiled firmware, or any of the existing firmware.bin files or even my original firmware that came with my board will work.

I'd give building with the virgin Marlin bugfix firmware, but I'm not really clear on everything that needs to be changed. And I'm not even sure about the changes and build I made with the BTT firmware. But even their stock, pre-built, firmware isn't working.

So basically I'm totally hosed and about to toss my board in the trash and go back to the original Creality board that came with my Ender 3.

Suggestions?

baub26, 2019-12-22:
Hello,
Have you an USB cable plug into the printer with an octopi running ?

damacguy, 2019-12-23:
Nothing connected to the printer - no octopi or any5hing else over usb.

wisesokol, 2019-12-26:
I use the BOARD for 2 weeks. Today, a TMC error appeared when you turned it on. After the reboot, everything worked. Marlin 2.0 Official. USB not connected

esk8skunk, 2019-12-30:
has anyone found a solution to this yet?
i just keep finding dead ends.
i get my ender to start up and say ready to print then back to tmc error.
is this just DOA? ive tried reloading the latest firmware multiple times.
Im really out of my comfort zone trying   to solve this. any help or info would be appreciated

artmuse, 2020-01-04:
I use the BOARD for 2 weeks. Today, a TMC error appeared when you turned it on. 
Same Problem.  Does anyone solve it?

Nikcos, 2020-01-05:
send m122 via USB cable and pronterface and see the TMC status

JWolfJr, 2020-01-21:
I also started having the same issue after shutting down my printer for a day.  Started up and only my x axis wasn't moving, all others could be moved under control movement.  With my octroprint and the BTT TFT35 I have hooked up.  Then I started getting the TMC connection error.  I ran M122, and got back a error: all LOW on the x axis, other three were good.  I bought a new board, only thing I could think of after reading dozens of posts about this board.  I really like the performance of it, so decided to give it another shot.  Might be my fault though, I went with Teaching Techs cases for the back. The air flow design seemed not the best for the MB and a lot of the wiring was running over that heat sink for the x axis.  This time around I'll be putting a second noctua fan in blowing straight over the MB and do a better job with cable management.  Please let me know what you found!

TravisPeterson21, 2020-01-25:
I had the exact same issue earlier this week. Had Amazon send me a new board, got I plugged in last night, and everything was working great. Turn it on this morning, TMC Connection Error. Have tried the firmware that came with the machine, the latest firmwares from here and uninstalling BLTouch, still have the same error. Too much fussing around for me, I’m going to get my refund from Amazon and move onto something else.

JWolfJr, 2020-01-26:
I just got my second one, ran a test print.  So far so good.  I made sure the wiring was clean and nothing coming close to the heat sinks also added a fan blowing straight onto them.  I'm with you, if I get this problem again, I'm done.  Keeping my fingers crossed!

timothylrobb, 2020-02-24:
I came across the same issue.  It worked for about a month.  Then TMC Connection Error with X Axis no longer working.  I validated that the motor works by swapping its connection with the Y motor.  I do have a BLTouch and I often have this connected to an Raspberry Pi 2B+ with OctoPi running.  I think the board went bad.  I built a new firmware, but the error still occurs.  Even with the latest Marlin.  At the moment I don't think this can be fixed without replacing the board.  Very disappointing.

Thinkersbluff, 2020-02-24:
I also had the XM drive fail after 3-4 weeks.
I noticed that the XM drive is the farthest from the cooling fan and all 3 motors were running hot. The motors also became quite loud. I have since replaced the board, installed a larger cooling fan, lowered the drive currents to 530 and configured Marlin to monitor the drivers.

I hope these measures collectively prevent the new board from going the same way as the last...

Sent from my BlackBerry - the most secure mobile device - via the Rogers Network

Reply-to: reply@reply.github.com
Cc: subscribed@noreply.github.com

JWolfJr, 2020-02-24:
I also had this issue, after about a month of use, turned on one day and the x axis wasn't working.  Also getting the TMC error.  I replaced the board and put this time with better cable management and placed a cooling fan directly over the driver heat sinks.  I am using the Teaching Tech rear mounted cases with a raspi 3b+, a noctua fan over the pi and now another over the MB heat sinks.  So far so good, been about a month since.  I haven't touched the current levels yet, how did you configure the firmware warnings?

Thinkersbluff, 2020-02-24:
> I also had this issue, after about a month of use, turned on one day and the x axis wasn't working. Also getting the TMC error. I replaced the board and put this time with better cable management [...]

In configuration_adv.h, I found these settings, and uncommented "#define MONITOR_DRIVER_STATUS" :

   * Monitor Trinamic drivers for error conditions,
   * like overtemperature and short to ground.
   * In the case of overtemperature Marlin can decrease the driver current until error condition clears.
   * Other detected conditions can be used to stop the current print.
   * Relevant g-codes:
   * M906 - Set or get motor current in milliamps using axis codes X, Y, Z, E. Report values if no axis codes given.
   * M911 - Report stepper driver overtemperature pre-warn condition.
   * M912 - Clear stepper driver overtemperature pre-warn condition flag.
   * M122 - Report driver parameters (Requires TMC_DEBUG)
   */
  #define MONITOR_DRIVER_STATUS

  #if ENABLED(MONITOR_DRIVER_STATUS)
    #define CURRENT_STEP_DOWN     50  // [mA]
    #define REPORT_CURRENT_CHANGE
    #define STOP_ON_ERROR
  #endif

I took my cue from the default setting " #define CURRENT_STEP_DOWN     50  // [mA]" to reduce the max current setting from 580 to 530 for the X, Y, Z and drivers.  Since I use Mesh Bed Leveling, the Z motor used to get quite hot, until the Fade Height was reached. Now it just gets warm-ish.  The X and Y motors are cooler than Z.   I converted my system to Direct Drive and the retract is just 1.5mm at 25mm/s.  The 650mA drive to the extruder motor seems not to be a problem. The EM driver is also directly under the fan, so probably keeping quite cool.
The TMC2209 drivers also come with a feature called StepCool, which can dynamically reduce the driver current to the minimum needed to overcome the mechanical load (using StallGuard to monitor the drive) but Marlin does not seem to support that yet.

I should also mention that I am still using the stock electronics box, but I replace the lid with one that holds the larger fan.  I also had to add an LM2596 buck converter to drive the fan at 12VDC, so it actually added to my cable management issues.  While I was test-fitting things, I noticed that the Y tag on the YM cable had been pushed down onto the YM driver heatsink and seemed to be pulling that heatsink off the chip.  rerouting that cable to approach the connector from overtop of the other wires, rather than underneath them seems to have fixed that, but it is very tight in there.

JWolfJr, 2020-02-24:
Thank you for all the info, this is very helpful.  On the conversation about recompiling firmware, do you or anyone else change these lines in the platform.ini file, to match the current libraries?
  LiquidCrystal
  TMCStepper@>=0.6.2,<1.0.0
I noticed this in the new Marlin non bugfix state.  Once I had my firmware compiled and working correctly, I haven't messed with it too much.  Just don't want to bring up new bugs to have to deal with. If I'm guessing correctly the @>= points to the newest library installed by platform???

JWolfJr, 2020-02-24:
I already have MONITOR_DRIVER_STATUS enabled, just checked!  Yeah, I made sure none of the wiring passes over the board.  First board, against my gut feeling, the wires for the drivers were passing over and along the heat sinks.  I knew better, and paid the price.  With the my wiring and cooling setup now there shouldn't be any problems, if there is, it has to be in the board!

moonev, 2020-03-02:
Two boards through Amazon failed. First one burned up the Y motor and caused endless layer shifts. Second board has a TMC error and the X axis is non responsive. Resetting firmware does not clear the TMC error and it persists upon start up.

mashurex, 2020-03-11:
I've had two boards fail with a TMC Connection Error on the X-axis now (Ender 3). Both times the motor driver on the motherboard died. I have much better than stock airflow and cooling in my case over stock. I turned on monitoring and reduced the current the second time around. Still fried the X-axis. I'm starting to think that this board should be avoided.

TrashcanOfLove, 2020-03-22:
What TMC stepper library is currently in use? 0.6.0 has issues but 0.6.1 fixes them.

TrashcanOfLove, 2020-03-22:
Recompiled with new library, no fix. Interested in this fix by sjasonsmith: https://github.com/teemuatlut/TMCStepper/issues/93 but I don't have the skills to apply it to the TMC2209s on the SKR mini E3 v1.2

FloodMonkey, 2020-04-15:
Arrgghhh! I have been printing a lot of long prints recently on my Ender 3 with BTT-SKR Mini E3 v1.2 and BLTouch  and the stock BTT-SKR / BLTouch firmware with no issues at all. Then this morning I switch on the printer and get the "TMC Connection Error" message. Pronterface connected and the M122 command gives an "X axis: all Low" all other are Ok. I've swapped out the X stepper with the y one and still get the error message and the failed x stepper appears to be working fine with the y axis. 
I'm not overly literate with programming these things so a lot of what I'm reading is going over my head. 
My printer tends to be stand alone with no connections to a Pi or PC.
As the printer had been working well for the last couple of months is this likely to be a motherboard failure rather than firmware issue?

Thinkersbluff, 2020-04-15:
Sounds familiar. X driver tends to be farthest from the cooling fan in the electronics box. Would not be surprised if the driver has failed.

rbcross62, 2020-04-24:
I have been using my SKR Mini E3 V1.2 since October and it has worked well.  Until yesterday.  It was printing and was moving along the X-axis when it just stopped with a TMC CONNECTION ERROR.  I did some tests like everyone else here and have determined that the 2209 for the X-axis has failed.  I am working with BTT technical support and have sent various screenshots to them in regards to some of the things that they have had me check.  The M122 command shows that there is no response from the X-axis driver so I am assuming that the board is bad at this point in time.  Should I have chosen the SKR E3 DIP v1.1 with the plug-in DIP modules instead of the Mini E3 1.2?  If so, it would make sense that I could replace a bad driver and not the entire board.  I haven't heard of any issues with the DIP version of that board, anyone else?  At this point in time, I would recommend staying away from this board and getting the V1.15 Creality board if you want to run silent.  A friend of mine has 3 of these and they have been running longer than mine without any issues.  Thanks for listening!

kPATm, 2020-05-21:
> 
> 
> Recompiled with new library, no fix. Interested in this fix by sjasonsmith: [teemuatlut/TMCStepper#93](https://github.com/teemuatlut/TMCStepper/issues/93) but I don't have the skills to apply it to the TMC2209s on the SKR mini E3 v1.2

Did you ever get anywhere with this TrashcanOfLove?

TrashcanOfLove, 2020-05-21:
I went with returning the board, the other fix was beyond my ability to
apply with confidence. It's a shame, I miss my quiet steppers

jackharvest, 2020-06-29:
TMC error as well. Printed perfectly for 10s of hours, and then, in the middle of my print, the stepper motor quit turning. After testing things, the motors were fine... it was the board. :(   Tried everything above. No love. Going for a refund, and will begin looking for a higher quality alternative.

Alexkey2200, 2020-12-07:
Hello everyone, I wanted to tell you that for 8 months I have the SKR mini E3 1.2 working without problems with the SD card. However, a few days ago I started to place Octoprint via USB, and it was when suddenly one morning my SKR rebooted and the damn "TMC connection error" sign appeared. Indeed, the affected axis is the "X", so I got up the courage and with a USB mocroscope and an air welder I desolder the drivers for the "X" axis and the "Y" axis to exchange them and see what happens. Then I tried the board again and the "TMC connection error" sign kept coming up but this time the "X" axis was working and the "Y" axis did not. So I confirm that the problem was indeed in the "X" axis TMC2209 driver. Finally connect the skr to the usb with octoprint and through gcode make movements of the "X" axis, notice that the temperature of the driver was triggered above normal immediately, which from the printer menu did not, in my understanding This may be due to a bug when connecting the SKR to USB that causes an overtemperature in the drivers and as the "X" axis is the one that is furthest from the ventilation it is the one that suffered the most. These days I will be buying a TMC2209 driver and I will replace it, hopefully I can get this board running again. And I hope the people of BEGEETRETECH solve this.

Thinkersbluff, 2020-12-08:
I hope you have covered the 5volt pin or cut the wire from the 5volt pin on the USB cable before plugging into the USB port on the board.  It is a well-documented problem with that board.

Alexkey2200, 2020-12-08:
> I hope you have covered the 5volt pin or cut the wire from the 5volt pin on the USB cable before plugging into the USB port on the board. It is a well-documented problem with that board. Get Outlook for iOS<https://aka.ms/o0ukef>
> […](#)
,

yes, of course I had the USB cable with the 5v cut off.  I had it like this from the original board of the ender 3 that if you had the 5v it caused constant reboots.

Thinkersbluff, 2020-12-08:
Oh, good, although it is disappointing that you still had a driver failure, since I am using the same board.

I did two other things I hoped would protect the X motor driver on my 2nd E3 mini:

1. I reduced the max motor current settings to 580ma in Marlin
2. I replaced the motherboard cooling fan with a 50mmx50mm NoiseBlocker fan.

So far, touch wood, my mini is still working, at least 24x6 hrs a week, for over 12 months now.

Congratulations on being able to replace your own driver.  I hope it works for you.

________________________________ From: Alexkey2200 notifications@github.com<mailto:notifications@github.com> Sent: Monday, December 7, 2020 6:31:55
PM To: bigtreetech/BIGTREETECH-SKR-mini-E3 BIGTREETECH-SKR-mini-E3@noreply.github.com<mailto:BIGTREETECH-SKR-mini-E3@noreply.github.com> Cc: Thinkersbluff badback42@outlook.com<mailto:
,

Alexkey2200, 2020-12-08:
> Oh, good, although it is disappointing that you still had a driver failure, since I am using the same board. I did two other things I hoped would protect the X motor driver on my 2nd E3 mini: 1. I [...]

Amazing, I find your prevention methods very successful in your SKR mini.  In my case I had the currents of the Axes "X, Y, Z" in 560mA.  But I did have the same stock cooler as the ender 3, in addition to having the entire printer in a semi-closed box (I didn't dare to close it completely due to the electronics ventilation issue).  In addition, the ender I have is the one with the ventilation under the chassis.  I think that all these factors played a significant role.

I also think it convenient to discard the electrical in my case, since in my house in the general electrical panel I have an over and low voltage protector, in addition the printer is connected to a 3KVA Eaton full online UPS, so it is electrically 100%  sheltered.

mashurex, 2020-12-08:
I went through two different board with an X-axis failure. My printer does not have heavy demands, it rarely has seen prints longer than a few hours and speeds are kept saFe (20-60mm/s). I've always had my Octoprint hooked up via a USB power blocker so as not to get a voltage input leak, made sure everything is grounded correctly, I have a high quality power supply, and all connections are secure.

The second time I drastically increased the cooling and airflow in a completely modified motherboard enclosure and I enabled the Marlin settings for monitoring the drivers, lowered the voltage, and included larger heat sinks. Hit with an IR thermometer they were at least 10C cooler than stock.

None of that helped, my second board only lasted about a month longer than the first- and it never saw a printer longer than 90m.

I gave up on the board after this- but I gave BTT another chance. I went with their standard SKR DIP (v1.4 I think) board with swappable 2209 drivers (in case any burned up I wouldn't have to swap the whole board). It's been going strong for nearly a year now and I've not had one single issue.

I think the E3 mini just has too many design flaws, probably due to size and space constraints and the drivers running hotter than other versions. Giving BTT the benefit of the doubt, they were constrained by the original design of the Creality board to a certain degree, so there may only be so much they can (or are) willing to do.

My recommendation is that when (not if) your E3 mini board burns out a driver- go get the standard SKR DIP board. You're going to have to relocate things and move a lot of wires (probably) but it is a fantastically better board all the way around. 

You'll already be familiar with the firmware process and wiring, so it won't be nearly as daunting as you think. Plus you'll get a bunch of extras you could use in the future. The last benefit is that when new drivers become available you will be able to upgrade your board as/when they get supported in firmware vs. having to buy an entirely new board.

rcxrbt, 2021-01-07:
for me, seems to happened after updare from stock firmware to 2.0.7.2 and powering the SKR Mini E3 1.2 (new board from box!)

rinasek, 2021-01-11:
It worked for 3 hours and then just died mid print.

JWolfJr, 2021-01-11:
I had the same issue after installing my original skr 1.2 e3 into Teaching Techs rear mounted case. I admittedly had wired it a mess, and only cooling fan was on the raspberry pi side. Only after a month of printing, turned on the printer and got the TMC Error. Replaced the board and cleanly wired it, no wiring near the stepper heat sinks. Wired a noctua fan blowing straight down over the heat sinks. So far after a year of printing, many prints taking we'll over 12 hours, no issues. Hope this helps.

gottria, 2021-02-05:
Found this thread just now due to the same error. Running the SKR 1.2 Mini E# board for well over a year in the teaching tech rear mounted setup with a  Raspberry Pi on Octoprint. The last week I have been running it harder than I have in the past and after an update on the Octoprint for the Power toggle plugin I get the TMC Connection error. After checking things out if I leave the USB unplugged and then reboot I get the error, unplug the USB reboot all is good and then I can plug in the USB and prints good. USB does have the 5v pin taped.

UnterStrom, 2022-03-11:
Used the skr mini v1.2 with my old Ender 2 with Marlin and octoprint for about 3 years, no heavy printing. The USB cable to my RP had a taped 5v pin, so no current from the RP to the board. Installed BLtouch and tried Klipper worked without a Problem, but didn't use it too long. Changed my hotend and forgot to change the Z-Distance crashed in my buildtak. After an emergency stop I used probe calibration and immediately I got the TMC failure. Didn't use the Fan in the casing, so thought maybe the driver got hot because of the movement in my buildtak (got pretty bad grooves). Used my spare Skr mini v1.2 with cooling fan in my case connected (pointed directly on the X stepper driver). Tried to use probe calibration again and after some MCU connection failures my X stepper driver died again (TMC failure). In my cfg file (used with both skr's) I had the stepper drivers limited to max current 580. So I guess my NEMA on the X-axis fried my second driver? Just as a warning for others, that maybe something else can also kill the driver 🥲