bigtreetech/BIGTREETECH-SKR-mini-E3 #156: Y axis problem

Rhteam67, 2019-12-26:
Hello,

I have an Ender 3 or I have just modified the motherboard with a SKR mini E3, BMG extruder, E3DV6 head from créality, 3Dtouch.

I have a problem when the plate advances forward, suddenly it taps against the belt tensioner. This happens when it finishes printing, advancing the tray. So when the 3Dtouch does the 9 leveling points and when it reaches the top of the bed then it goes too far.

I have correctly entered my Z offset for the nozzle and the 3Dtouch in Marlin 2.0.X but can find nothing to enter a Y offset because I think there is a gap between the original head and the E3DV6 head.

I also moved my contact Y because otherwise the 3dtouch does not palpate from the start

If you have an idea how to prevent the chainring from hitting the belt tensioner.

Thank you

Nikcos, 2020-01-04:
i have a similar problem  of Y axis ,  im trying to print this test (https://www.thingiverse.com/thing:2948908) and the printing starts from the homing position (about 2-3 cm off the bed), also prints lower on the upper side bed , its like the Y axis needs to move about 2-3 cm upper , X/Y homing is normal , only when printing have this issue(the printings are perfect when no need to be bigger) , i have try several slicers both Cura or S3D , i get same issue

adrianmihalko, 2020-01-24:
Looks like we have same problem https://www.youtube.com/watch?v=JnDc88uBsIM