bigtreetech/BIGTREETECH-SKR-mini-E3 #159: RRD Full Graphic Smart Controller

MDFPereira, 2019-12-27:
Hello everyone.

Did anyone was able to use a RRD Full Graphic Smart Controller with the SMK E3 mini?

I'm looking for possible solutions before my TFT arrives (maybe 1 month)...

Thank you in advance.

MDFPereira, 2019-12-27:
After some experimenting I found it...
For the ones interested here is an image and PDF file from the connections.

[image]
[RRD FGSD SKR.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4003952/RRD.FGSD.SKR.pdf)

thisiskeithb, 2019-12-27:
There are design files for an AUX1 + AUX2 to AUX3 PCB floating around somewhere, so there will probably be adapters your can buy sometime in the coming year.

MDFPereira, 2019-12-28:
> There are design files for an AUX1 + AUX2 to AUX3 PCB floating around somewhere, so there will probably be adapters your can buy sometime in the coming year.

Yes, soon they will come out.
Meanwhile, if anyone needs, here it is how I get it working ;)
Just wanted to share if someone needs.

dp250f, 2020-01-19:
It bothered me that I couldn't use the SD card reader in 12864 mode on my TFT35 v3.0, so I figured out how to wire up the SD Card pins (SPI) and a couple other pins for Slave Select and SD Detect to make it work.

I'm not using a power supply relay or BL-Touch on this printer, so I used those pins for SD Detect and Slave Select. I used a 5 pin dupont cable to connect these pins like so:
```
TFT35 v3.0 EXP2 > SKR Mini
---------------------
SD-CLK --> SCK (PA5)
SD-CS ---> SERVOS (PA1)
SD-MOSI -> MOSI (PA7)
SD-MISO -> MISO (PA6)
SD-DET --> PT-DET (PC12)
```
I also tacked this onto the end of my pins_BTT_SKR_MINI_E3_V1_2.h file:
```
// Hack to use LCD SD Card on BTT-TFT35-E3-V3.0 controller
#ifdef CR10_STOCKDISPLAY
  #undef SDCARD_CONNECTION
  #define SDCARD_CONNECTION LCD
  #undef SD_DETECT_PIN
  #define SD_DETECT_PIN  PA1  //PT-DET GPIO pin
  #undef LCD_SDSS
  #define LCD_SDSS       PC12 //SERVOS GPIO pin
#endif
```

It works exactly like it would if it had EXP1 and EXP2 connected instead of EXP3

ahx948, 2020-02-20:
Hey guys, I'm trying to upgrade my Wanhao D6 with skr e3 1.2 and want to keep stock lcd/sd card controller.  It's using an ultimaker 2 clone mainboard and lcd. Default setting in marlin config for the lcd is TinyBoy2 128x64 OLED / Encoder Panel

I tried the wiring above but it doesn't work. Has anyone know if it's possible to make this lcd and sd card controller work?

usk-johnny-s, 2020-02-25:
PC12 conflicts USBSerial of maple.
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/144#issuecomment-590925789

Fred3Dprints, 2020-09-23:
Hi @MDFPereira , many thanks for your contributions. It was exactly what I needed to adapt the SKR to and older printer. Much appreciated.

wikolii, 2020-12-04:
Hi guys
Is this viable for the RRD 2004 smart controller also?
Does it need more pins than the full graphic smart controller?
I was about to just try the wiring @MDFPereira shared but i'm a little concerned i might damage the board or the display

Thanks!

wikolii, 2020-12-06:
Hi again
I was able to get the RRD 2004 LCD working, there are a several more pins that need space for this screen (LCD5, 6 and 7 on EXP1. I could barely get it working by disabling functions like PS-ON and PWR-DET.

@dp250f 's guide for using the LCD's SD slot was super handy, thanks

I'm workng on an adaptor to just connect the L shaped adaptor that comes with all these controllers. If and when i have more info i shall post here.

Thanks!

michael725stone, 2021-05-05:
I LoVe you all for the support. 

I currently have a Skr Mini V1.2 hooked up to a Velleman K8200 converted to CoreXY with a 12864 RRDFGSCfrom BTT.

Nevermind the frankenstein printer the board replaced my old ramps 1.4 and everything was fairly smooth for setup and config (although Neopixels could have been easier). With printer running fine and printing cubes no problem I decided to tackle the LCD concern. I attempted the wiring for my 12864 to convert to one plug and everything boots up fine screen looks great and encoder works when rotated BUT I have no menus. Seems encoder button signal is not being recognized.

Questions
Should I expect to have menus when performing the wiring that @MDFPereira was so kind to figure out?
Will those menus allow me to print from the SD on the actual Skr Mini or no SD access at all?
I would like to use a BLTouch at some point so should I abandon this screen all together?

jdmusica, 2024-02-24:
updated diagram
[image]

3dprintpt, 2024-02-24:
> updated diagram ![DIAGRAMA LCD REPRAP DISCOUNT FULL DISPLAY_BTT SKR MINI E3 V2](https://private-user-images.githubusercontent.com/91290524/307519336-116c4df4-a2de-4107-9533-b45ca49e1831.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg3NzkyNDcsIm5iZiI6MTcwODc3ODk0NywicGF0aCI6Ii85MTI5MDUyNC8zMDc1MTkzMzYtMTE2YzRkZjQtYTJkZS00MTA3LTk1MzMtYjQ1Y2E0OWUxODMxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjI0VDEyNDkwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBmMzE0ZDM0Mzk5NDVlNTZhMzdhNWZiZGI0N2QxNzVkM2IxNzA1MWZkZjg0MTNiNmNiOTY2ZTYxN2I1NGU1YzImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.2Vbt0sNxo6g8rY6P7rb-t_cjCi0jTxHaN4YU_Z99rtI)

Awesome...
Thank you