bigtreetech/BIGTREETECH-SKR-mini-E3 #160: after compiling firware and attempting to set z offset it freezes up and z stepper is clicking.  I finally gave up and was gonna go back to stock sd card and no bltouch and it does the same just before it reaches temp.  I have tried several times and it just freezes up.  When i push reset it does so but within 5 minutes it freezes again.[BUG] (short description)

monkeybutt13, 2019-12-30:
### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

monkeybutt13, 2019-12-30:
I used  http://github.com/morningreis/Marlin-SKR-E3-Mini-1.2

Dracrius, 2020-01-11:
@monkeybutt13 did you try using the official marlin build? I honestly don't see why you would post your issue here since your using that build why not report it to the maintainer? That build is also out of date, many SKR mini e3 problems have been and are being worked on in the main branch. I would suggest building this branch or the official your problem is most likely related to said build as I am running Marlin 2.0.1 bugfix on my SKR mini e3 with a bltouch without major issue.

monkeybutt13, 2020-01-11:
Hello, 
Yes I used the official build and this one and the original one that came with the board. I’m glad you are having luck with yours. That doesn’t mean I’ve done something wrong. I have dealt with support and they have agreed that something is wrong with the board and are sending me a new one. Do you work for big tree tech?  Or do you not have anything better to do?

Dracrius, 2020-01-11:
Why do people think this is a Bigtreetech general support page?? No I'm I don't work with them many/ most people who will actually maintain this build probably don't that's what open source is. I haven't been involved in this branch in the least I've just been watching it since purchase for updates that may be unique or to help out. so I don;t know what your getting at with the nothing better to do I was trying to help. If you post an issue this vague referencing a complete different Fork how can you be sassy with me lol. I was just trying to help bud update your post if you got new info its obviously not a bug related to this Fork then so also close the damn ticket!!