bigtreetech/BIGTREETECH-SKR-mini-E3 #161: Run Away Bed Temperature!

Augcliffe, 2019-12-30:
I just installed this card.  Immediately after turning on the power for the first time, the bed started increasing temperature and would not stop.  The bed quickly became too hot to touch (the stock card  never did that).  I tried going into the configurations and setting the max temperature to 60 and it ignored that and continued to climb.  I power cycled, I checked everything I could think to check.  I even flipped the "hot end thermistor" and "bed thermistor" cables on the circuit board because they aren't labeled and I wasn't sure which was which.  Regardless of that, the same symptom persisted.  Someone else on Amazon gave this board a 1 star review with the same problem I have.

drphil3d, 2020-01-02:
You need to put the correct firmware on it. It comes with test firmware from the factory for automated testing.

drphil3d, 2020-01-02:
Or you wired the bed into the power input instead of bed output

Augcliffe, 2020-01-03:
> Or you wired the bed into the power input instead of bed output

This is what the problem was.  To be fair, all the diagrams and pictures I googled had it wrong too.  On the board I noticed the postive and negative different.

SwiCago, 2020-01-04:
Can you not close your own issue request? Since you figured it out.

closed 2020-01-05 by Augcliffe