bigtreetech/BIGTREETECH-SKR-mini-E3 #163: [BUG] (power loss recovery )

spooky79, 2020-01-01:
the function does not work, after the power failure and confirmation of resuming printing the printout starts again. Does this function work on SKR E3 mini v1.2? on ender 3 .Tested on different versions of marlin
[Configuration_advh.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4014455/Configuration_advh.txt)
[configurationh.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4014456/configurationh.txt)

cjsoong, 2020-01-08:
@spooky79 
You need to enable #define SAVE_EACH_CMD_MODE in power_loss_recovery.h & select power_loss_recovery to enable in LCD menu.

power_loss_recovery.h
#if !defined(POWER_LOSS_STATE) && PIN_EXISTS(POWER_LOSS)
  #define POWER_LOSS_STATE HIGH
#endif

#define DEBUG_POWER_LOSS_RECOVERY
#define SAVE_EACH_CMD_MODE
//#define SAVE_INFO_INTERVAL_MS 0

spooky79, 2020-01-26:
512kb version uploaded, only with the 512kb version the function power loss recovery works correctly

closed 2020-01-26 by spooky79

Talha909, 2020-06-01:
> 512kb version uploaded, only with the 512kb version the function power loss recovery works correctly

what do you mean by the 512 kb version

rakeshbarman, 2020-12-13:
> @spooky79
> You need to enable #define SAVE_EACH_CMD_MODE in power_loss_recovery.h & select power_loss_recovery to enable in LCD menu.
>
> power_loss_recovery.h
> #if !defined(POWER_LOSS_STATE) && [...]

As of Marlin-bugfix2.0.x
-------------------------
marlin\src\feature\powerloss.h

#if !defined(POWER_LOSS_STATE) && PIN_EXISTS(POWER_LOSS)
  #define POWER_LOSS_STATE HIGH
#endif

#define DEBUG_POWER_LOSS_RECOVERY // uncommented
#define SAVE_EACH_CMD_MODE // uncommented to enable Power Loss Recovery in LCD menu.
#define SAVE_INFO_INTERVAL_MS 0  //  uncommented

labotecno, 2022-06-20:
what is mean this option?
can you explain me better