bigtreetech/BIGTREETECH-SKR-mini-E3 #166: Y axis problem on Ender 3

Nikcos, 2020-01-05:
### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

Nikcos, 2020-01-05:
Im trying to print this test (https://www.thingiverse.com/thing:2948908) and the printing starts from the homing position (about 2-3 cm off the bed), also prints lower on the upper side bed , its like the Y axis needs to move about 2-3 cm upper , X/Y homing is normal , only when printing have this issue(the printings are perfect when no need to be bigger) , i have try several slicers both Cura or S3D , i get same issue 
[image]

tatusah, 2020-01-06:
You could use `#define MANUAL_Y_HOME_POS -20` in configuration.h file to move the y 0 position to start 20 mm back after homing. Measure for yourself how far you need it to be set. It's known that at least on some batch of Ender 3 the Y endstop switch screw holes are drilled bit too far back. If you don't want to drill new holes or make a diy adjustable endstop mount then adjusting the home position in firmware is a viable option. My printer has also the endstop switch too far and use this firmware adjustment.

Alternative you can set `#define Y_MIN_POS -20`. Difference between these two is that offsetting the home position blocks moving the nozzle to the negative values after homing. In comparison the minimum position setting tells that the home position is actually -20 instead of 0, but still allows to move the nozzle to the negative values area. Personally I prefer the home position option as it helps to prevent collision with bed clips.

Nikcos, 2020-01-07:
At first thanks for the answer , i tried to change the #define Y_MIN_POS , with no luck , nothing happened , the #define MANUAL_Y_HOME_POS  worked , but i have to adjust it  , because bed hit a little bit, when getting in front , do i have to change anything else , or just the firmware ?