bigtreetech/BIGTREETECH-SKR-mini-E3 #167: [BUG]Bed leveling stops after 6 probes (G29)

kramer04, 2020-01-05:
### Description
I'm using Ender 3 + SKR-MINI-E3 V1.2 + TouchMi and latest BTT Marlin

### Steps to reproduce
Launching G29 or a print bed level stops after 6 probes

**Expected behavior**
bed leveling with 36 probes

### Additional Information
M503 gives this

G21    ; Units in mm (mm)
M149 C ; Units in Celsius
Filament settings: Disabled
M200 D1.75
M200 D0
Steps per unit:
M92 X80.00 Y80.00 Z400.00 E412.00
Maximum feedrates (units/s):
M203 X500.00 Y500.00 Z5.00 E25.00
Maximum Acceleration (units/s2):
M201 X500.00 Y500.00 Z100.00 E5000.00
Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
M204 P500.00 R500.00 T500.00
Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
M205 B20000.00 S0.00 T0.00 J0.08
Home offset:
M206 X0.00 Y0.00 Z0.00
Auto Bed Leveling:
M420 S0 Z0.00
G29 W I0 J0 Z-1.98250
G29 W I1 J0 Z-1.27250
G29 W I2 J0 Z-0.55000
G29 W I3 J0 Z0.21000
G29 W I4 J0 Z0.93000
G29 W I5 J0 Z1.64750
G29 W I0 J1 Z0.00000
G29 W I1 J1 Z0.00000
G29 W I2 J1 Z0.00000
G29 W I3 J1 Z0.00000
G29 W I4 J1 Z0.00000
G29 W I5 J1 Z0.00000
G29 W I0 J2 Z0.00000
G29 W I1 J2 Z0.00000
G29 W I2 J2 Z0.00000
G29 W I3 J2 Z0.00000
G29 W I4 J2 Z0.00000
G29 W I5 J2 Z0.00000
G29 W I0 J3 Z0.00000
G29 W I1 J3 Z0.00000
G29 W I2 J3 Z0.00000
G29 W I3 J3 Z0.00000
G29 W I4 J3 Z0.00000
G29 W I5 J3 Z0.00000
G29 W I0 J4 Z0.00000
G29 W I1 J4 Z0.00000
G29 W I2 J4 Z0.00000
G29 W I3 J4 Z0.00000
G29 W I4 J4 Z0.00000
G29 W I5 J4 Z0.00000
G29 W I0 J5 Z0.00000
G29 W I1 J5 Z0.00000
G29 W I2 J5 Z0.00000
G29 W I3 J5 Z0.00000
G29 W I4 J5 Z0.00000
G29 W I5 J5 Z0.00000
Material heatup parameters:
M145 S0 H190 B60 F0
M145 S1 H240 B110 F0
PID settings:
M301 P25.27 I2.01 D79.52
Power-Loss Recovery:
M413 S1
Z-Probe Offset (mm):
M851 X-50.00 Y-1.00 Z0.00
Stepper driver current:
M906 X580 Y580 Z580
M906 T0 E650
Driver stepping mode:
M569 S1 X Y Z
M569 S1 T0 E
Linear Advance:
M900 K0.15

cyberdriveautomotive, 2020-01-12:
I too am having this issue as well

I am using a BL Touch clone, unsure of which type and version, it's a black plastic casing.

My clone was working perfectly fine with the stock E3P board but now it does exactly as described by this ticket.

I was able to get it to probe 20 out of 25 points by forcing it to probe the Y axis first.

It seems to run into an issue when attempting to probe the right side of the bed, and I've adjusted firmware bounds settings to attempt to fix that to no avail.

HVJ17, 2020-01-15:
I suggest disabling multiple probes 
//#define MULTIPLE_PROBING 2
and uncomment
#define DELAY_BEFORE_PROBING 100
#define PROBING_FANS_OFF   

You could also try changing the probe points to 16.
// Set the number of grid points per dimension.
  #define GRID_MAX_POINTS_X 4
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

cyberdriveautomotive, 2020-01-16:
I was able to figure out why mine wasn't finishing probing.

My BL Touch is x-38.15 and y-7.2 but my min probe edge was set to 20mm. I set MIN_PROBE_EDGE to 40 and I was able to get it to probe my entire bed successfully.

Hopefully this helps!

kramer04, 2020-01-16:
> I was able to figure out why mine wasn't finishing probing.
> 
> My BL Touch is x-38.15 and y-7.2 but my min probe edge was set to 20mm. I set MIN_PROBE_EDGE to 40 and I was able to get it to probe my entire bed successfully.
> 
> Hopefully this helps!

Thanks it works. I've a TouchMi
But i'm sure there is a bug somewhere