bigtreetech/BIGTREETECH-SKR-mini-E3 #169: BTT SKR Mini E3 1.2 with BTT Smart Filament Sensor an BLTouch ABL and TFT35 E3 3.0Possible?

Sn4k3E4t3r, 2020-01-10:
Hi,

at first sorry for my bad english :) I am now the proud owner of the following products. I would like to connect the BTT Smart Filament Sensor and a BLTouch and a TFT35 E3 V3.0 to the SKR mini E3 V1.2 is that possible? I have an ender 3

dziabi3, 2020-03-07:
Yes, you can connect, I have the same hardware configuration but I temporarily disconnected TFT 35 due to strange behavior.
It is possible that it was due to my ignorance. You will use this display again as soon as I get more experience from Marlin. I am currently devoting time to octoprint, which compensates me for the lack of TFT 35.

demonkazuya, 2020-04-23:
> Yes, you can connect, I have the same hardware configuration but I temporarily disconnected TFT 35 due to strange behavior.
> It is possible that it was due to my ignorance. You will use this display [...]

Could you please confirm what strange behavior you are having? I have the same setup but i'm unable to have the Filament sensor work. i have attached the sensor to PT-DET but every time I remove the filament the printer shuts down..