bigtreetech/BIGTREETECH-SKR-mini-E3 #172: Z axis is only moving down, even after z endstop is reached. Ender 2 TMC connection Error

marijnnos, 2020-01-17:
So I installed the SKR mini a while ago and after configuring and updating the firmware I noticed some odd behavior in the z axis. First it locked up a few times and made some banging noises and then after a few tries of homing, the z axis would only go down. 
Ive tried (i think) every different configuration possible, changing direction, changing endstop triggering, enable/disable software endstop. But still no luck. 
It is also displaying TMC Connection Error.

Is the driver fried or does anyone has an idea what the problem could be?

Thanks in advance,

Marijn

JWolfJr, 2020-01-21:
I see it's been a few days, have you found anything out?  I don't know if this helps, but, I just started running into a similar issue.  After shutting down for a night and turning back on the next day, my z axis will only move up a bit before homing when starting a print.  Then freezes up and octoprint gives an error message and I have the TMC connection error on my lcd screen.  Also I can move all axis with the control menu except the x axis.  When I run the M122 command, I get back all ok except the x axis reports error: all low.  I ordered a new board, since I have the integrated stepper drivers.  Try running the M122 command and see what you get, then go from there.

goos766, 2020-01-23:
I had a similar situation, it turned out that the cable from BL touch was damaged after replacing the cable, everything started working again

drphil3d, 2020-03-21:
I purchased an skr mini e3 Rev1.0 6 months ago and the x axis would only move in one direction. however that was rev1 of the board so who knows.

tolitski, 2020-04-03:
> I purchased an skr mini e3 Rev1.0 6 months ago and the x axis would only move in one direction. however that was rev1 of the board so who knows.

I am hoping we don't have the same batch of boards. I am having a hard time getting this to work.