bigtreetech/BIGTREETECH-SKR-mini-E3 #173: [BUG] (short description)

m0rtiZz, 2020-01-17:
On firmware for skr mini e3 1.2 "firmware-bltouch-for-z-homing.bin" i'm inserting xyz offsets with m851 command, saving with m500 comand, and after rebooting the printer offsets are set to default, not these what i entered.

### Steps to reproduce

1. M851 X-44 Y-16 Z-2.8
2. M500
3. Reboot printer

**Expected behavior**
Set my offsets and have them saved

**Actual behavior**
Offsets reseting to default

### Additional Information
using configuration.h provided with marlin downloaded directly from marlin site

USAFRichards, 2020-01-17:
I have the same problem. Thank you.

Thinkersbluff, 2020-01-17:
Please confirm whether you are using Marlin 2.0.1.  There is a BUG in 2.0 that corrupts EEPROM saved to flash memory.

m0rtiZz, 2020-01-17:
> Please confirm whether you are using Marlin 2.0.1. There is a BUG in 2.0 that corrupts EEPROM saved to flash memory.

I'm using firmware created by btt, "firmware-bltouch-for-z-homing.bin", it's marlin 2.0.x

Thinkersbluff, 2020-01-17:
Closed bug #16532 or #16118 may help, then. 
If you have print counting on, you could try turning that off.
Forcing Marlin to use the SD card as the emulated EEPROM works.
Upgrading Marlin to 2.0.1 works.

USAFRichards, 2020-01-18:
How do you force Marlin to use the SD card?  

Also, is there a way to modify the "firmware-bltouch-for-z-homing.bin" to make fixes?

I was able to add emulated EEPROM to use the flash memory via the pins file when compiling, but then my stepper motors were louder than with the "firmware-bltouch-for-z-homing.bin".  If I could just modify the pins file of this firmware I would be able to keep all the other settings.

Thinkersbluff, 2020-01-18:
> How do you force Marlin to use the SD card?

"When I comment-out the line "#define FLASH_EEPROM_EMULATION" in the file pins_BTT_SKR_MINI_E3.h, then Marlin 2.0.x reverts to using the SD card to store EEPROM.dat. In that configuration, the EEPROM does work as expected."

I realize there is a lot to read in closed bug #16532.  The above info is included there.
NOTE also, if you do decide to upgrade to Marlin 2.0.1, from the bugfix branch, there are other edits and issues documented in #16532.

USAFRichards, 2020-01-18:
I read this and I also removed the SD card, deleted the bin file and reset the machine. Seems to hold.

Thinkersbluff, 2020-01-23:
Cool!  Does that mean this bug can be closed?

USAFRichards, 2020-01-29:
Now, if I don’t “Save Settings” before shutting down the machine I lose my offsets. If I perform the “Save Settings” before shutting down they are save for the next boot up.

brew99, 2020-01-30:
That is how it is suppose to work.  You need to "save settings", either to SD or use emulated EEPROM, then they will be recalled on next boot-up

krakow2000, 2020-02-05:
> Now, if I don’t “Save Settings” before shutting down the machine I lose my offsets. If I perform the “Save Settings” before shutting down they are save for the next boot up.

But what with PID nozzle and bed if you use repetier or other soft to calibrate?

brew99, 2020-02-05:
@krakow2000 just use M500 command in repetier to save the settings.   M503 to recall them and make sure they were saved

huzargarden, 2020-02-11:
Can bigtreetech update the file "firmware-bltouch-for-z-homing.bin" to eliminate this bug? I have the same issue and i can't save the setting on the BTT TFT35 screen

brew99, 2020-02-11:
Of course they can, but when is the real question!!   Best to take some time and learn how to compile it yourself using the Vanilla Marlin not BTT version

huzargarden, 2020-02-12:
then what is the point to keeping files with errors  if everyone will go to compile its own version? 

not everyone have time and wants to spent time to config and compile marlin if there is frlirmware file availble to upload. Some poeple here are actually using printers to print stuff and not to consistly improve them and print parts for the printer ;)

brew99, 2020-02-12:
I totally agree it is not very useful to have a firmware that has such a major issue in it.  Support is not a very strong point for BTT, so sometimes you need to take it into your own hands and correct it yourself

Xenon-2510, 2020-04-06:
I have this issue with the BTT firmware too, I have put the M851 for Z offset in my Slicer script so it sets the Z offset prior to each print. More of a workaround than a fix but for those who don't want to compile from Vanilla Marlin this will work.