bigtreetech/BIGTREETECH-SKR-mini-E3 #174: Source for "firmware-bltouch-for-z-homing.bin"

USAFRichards, 2020-01-18:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I am new. I would like to enable linear advance but I dont have source files for "firmware-bltouch-for-z-homing.bin". 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Is there such a source file so I don't lose the current settings contained in it and I can enable linear advance?

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
I am new to compiling so I have tried but I dont know if all the features contained in "firmware-bltouch-for-z-homing.bin" are in the compliled file. For instance my Z is much noisier with the firware I compile. I would like to ad linear advance to "firmware-bltouch-for-z-homing.bin".

**Additional context**
Add any other context or screenshots about the feature request here.

HVJ17, 2020-01-19:
Are you compiling 2.0.1? It has a HYBRED on by default, This makes the Z scream. Who wants faster movement at the cost of noise??? I bought this board for the silence more than anything. I suggest going to adv.h and 
//#define HYBRID_THRESHOLD
Linear advance is about useless. I know there are some uses. Do you have a specific use for it?  I think it is enabled in the pre-compiled firmware. Have you printed the test file from Marlin? That will show if it is enabled or not.

USAFRichards, 2020-01-20:
So I did figured out that Linear Advance was enabled in my build. I wanted to use it so I could get rid of or reduce the amount of bulging at the corners. Thank you for responding.

closed 2020-01-20 by USAFRichards