bigtreetech/BIGTREETECH-SKR-mini-E3 #177: section `.rodata' will not fit in region `rom'

SergeySn, 2020-01-20:
I am trying to debug the code on BigTreeTech SKR Mini E3 board.  When I click the build button - it compiles fine.  And when I click the "Platform IO: Upload" button it compiles and uploads via STLINK fine. But when I click "Start Debugging" menu, it gives me an error:

> Executing task: platformio debug 
> Processing STM32F103RC_bigtree_NOUSB (platform: ststm32; framework: arduino; board: genericSTM32F103RC)
> Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
> PLATFORM: ST STM32 6.0.0 > STM32F103RC (48k RAM. 256k Flash)
> HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
> DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
> PACKAGES: framework-arduinoststm32-maple 1.10000.190819 (1.0.0), toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), tool-stm32duino 1.0.1
> Converting Marlin.ino
> LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ soft
> ...
> Building in debug mode
> Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduinoVariant/board.cpp.o
> Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduinoVariant/wirish/boards.cpp.o
> Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduinoVariant/wirish/boards_setup.cpp.o
> ...
> Compiling .pio/build/STM32F103RC_bigtree_NOUSB/lib479/EEPROM/EEPROM.cpp.o
> Compiling .pio/build/STM32F103RC_bigtree_NOUSB/lib479/EEPROM/flash_stm32.c.o
> Archiving .pio/build/STM32F103RC_bigtree_NOUSB/lib479/libEEPROM.a
> Indexing .pio/build/STM32F103RC_bigtree_NOUSB/lib479/libEEPROM.a
> Linking .pio/build/STM32F103RC_bigtree_NOUSB/firmware.elf
> /home/serg/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/STM32F103RC_bigtree_NOUSB/firmware.elf section `.rodata' will not fit in region `rom'
> /home/serg/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: region `rom' overflowed by 12244 bytes
> collect2: error: ld returned 1 exit status
> [.pio/build/STM32F103RC_bigtree_NOUSB/firmware.elf] Error 1

What might be wrong?

brew99, 2020-01-23:
change your default env. in platformio.ini to "STM32F103RC_bigtree_512K_NOUSB" rather than the 256k.   It is telling you that there are too many things enabled and the code wont fit.   It works with STlink because it is doing a raw data upload without any sanity checks

SergeySn, 2020-01-23:
@brew99  Thanks a lot for the info. Though my new board is version 1.2 with STM32F103RCT6 chip, which has has only 256k.
I also have the previous version of the board with STM32F103RET6 chip, which has has 512k. So, I tried it, but got this error:

```
DATA:    [===       ]  26.4% (used 12968 bytes from 49152 bytes)
PROGRAM: [=====     ]  46.9% (used 245724 bytes from 524288 bytes)
...
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x00000003 pc: 00000000 msp: 0x464c4544
Info : device id = 0x10036414
Info : flash size = 512kbytes
0x00000000 in ?? ()
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00010100 msp: 0x464c457c
Loading section .text, size 0x3614c lma 0x8007000
Loading section .ARM.exidx, size 0x190 lma 0x803d150
Loading section .data, size 0x10b0 lma 0x803d2e0
Loading section .rodata, size 0x4c4c lma 0x803e390
Info : Padding image section 0 at 0x0803d14c with 4 bytes
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00010100 msp: 0x464c457c
Start address 0x80073f8, load size 245720
Transfer rate: 19 KB/sec, 13651 bytes/write.
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00010100 msp: 0x464c457c
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00010100 msp: 0x464c457c
Temporary breakpoint 1 at 0x800ad62: file /home/serg/.platformio/packages/framework-arduinoststm32-maple/STM32F1/cores/maple/main.cpp, line 38.
PlatformIO: Initialization completed
PlatformIO: Resume the execution to `debug_init_break = tbreak main`
PlatformIO: More configuration options -> http://bit.ly/pio-debug
Note: automatically using hardware breakpoints for read-only addresses.
```
Also, even with the STM32F103RC_bigtree_NOUSB board selected, it compiles fine, and I could upload it to the board via the microSD card, and it worked. And the compiler says the code is 245724 bytes. (See above)  So, the code is smaller than 256k.

But when I uploaded the code via STLINK - it stopped working.  I wonder if STLINK overwrote the bootloader?
Somebody said that my STLINK starting address is wrong.  But how do I check or fix it?

brew99, 2020-01-24:
even though your new V1.2 board has the RCT6 label, it still has the 512k, so you should be able to use the "STM32F103RC_bigtree_512K_NOUSB" with it.   If you have the labelled RET6 chip then you need to use the "STM32F103RE" environment.   Basically the actual chips both have 512, but for marketing purposes, the RCT6 is market at 256.

The bootloader is 0x0 to 0x7000 I believe.  So if you load firmware via the STLINK, the start address needs to be 0x7000, with a size of 80000.   You should always read via STLINK (start at 0x000 in order to capture the bootloader also), then save the file, before starting to alter or load a new file.

SergeySn, 2020-01-24:
@brew99 Thanks a lot for the clear explanation.  Now I know how to solve the size issue.  I removed some unused code, so it fits in both configurations.  Now it compiles in debug mode perfectly.
But when I click the debug button - it goes to Start.S file line 51:

> ldr r1,=__msp_init

shows the current execution point there for a second, and then exits the debugger.
In debug console it shows the following:

```
Reading symbols from /a/proj/VSCode/BIGTREETECH-SKR-mini-E3-/firmware/V1.2/Marlin-2.0.x-SKR-Mini-E3-V1.2/.pio/build/STM32F103RC_bigtree_NOUSB/firmware.elf...
done.
PlatformIO Unified Debugger -> http://bit.ly/pio-debug
PlatformIO: debug_tool = stlink
PlatformIO: Initializing remote target...
xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-11:25)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
hla_swd
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : tcl server disabled
Info : telnet server disabled
Info : clock speed 1000 kHz
Info : STLINK V2J32M22 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.254104
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : accepting 'gdb' connection from pipe
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x00000003 pc: 00000000 msp: 0x464c454c
Info : device id = 0x10036414
Info : flash size = 256kbytes
0x00000000 in ?? ()
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00010100 msp: 0x464c457c
Loading section .text, size 0x18a24 lma 0x8007000
Loading section .ARM.exidx, size 0x8 lma 0x801fa28
Loading section .data, size 0x1058 lma 0x801fa30
Loading section .rodata, size 0x176c lma 0x8020a88
Info : Padding image section 0 at 0x0801fa24 with 4 bytes
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00010100 msp: 0x464c457c
Start address 0x80073f8, load size 111088
Transfer rate: 18 KB/sec, 11108 bytes/write.
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00010100 msp: 0x464c457c
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00010100 msp: 0x464c457c
Temporary breakpoint 1 at 0x800ab16: file /home/serg/.platformio/packages/framework-arduinoststm32-maple/STM32F1/cores/maple/main.cpp, line 38.
PlatformIO: Initialization completed
PlatformIO: Resume the execution to `debug_init_break = tbreak main`
PlatformIO: More configuration options -> http://bit.ly/pio-debug
Note: automatically using hardware breakpoints for read-only addresses.
```
What might be wrong?
Also, how do I change the start address to 0x7000 ? (or make sure it's already there)

brew99, 2020-01-24:
I've not used VSC/platformio other than to compile Marlin, and then put on SD card.   I do not use it with STLINK hardware.

I have usde the STlink hardware to read/write via the STM ST-link utility found here:
[utility](https://www.st.com/en/development-tools/stsw-link004.html)

that is where you put the start address 0x7000 if you only want firmware without bootloader, otherwise if you want it all, then start at 0x0000.  0x80000 should be the size with 32 data bits.

It suggest reading, then saving a default file before you start messing with it and writing new firmware/bootloader

SergeySn, 2020-01-25:
Ah, I see.  Thanks.
BTW, if you are not using Platformio and STLINK, how do you debug it?

brew99, 2020-01-25:
Are you talking about debugging marlin code?  If so, I don't debug it, just change/enable a few things or parameters, then compile/build.

The only thing I have used the STlink for was to read my V1.0 board, then alter an address to allow it to get to 512k (it was hard locked to 256k in the bootloader as it was one of the first production runs of V1.0 board).   I also tried loading a v1.2 bootloader on it, which also worked.

SergeySn, 2020-01-25:
Got it.  Thanks!

closed 2020-01-25 by SergeySn