bigtreetech/BIGTREETECH-SKR-mini-E3 #184: [BUG] (problem with the PSU_CONTROL function)

dziabi3, 2020-02-15:
When I set  "default_envs =  STM32F103RC_bigtree" or "default_envs = STM32F103RC_bigtree_512K" pin PS_ON works correctly and the printer does not turn off.

When I set "default_envs = STM32F103RC_bigtree_NOUSB "default_envs  = STM32F103RC_bigtree_512K_NOUSB" pin PS_ON does not work properly and the printer turns off after a few seconds.

Can anyone check this?

I have a motherboard SKR mini e3 V1.2 and dedicated software.

Currently I have to use "default_envs = STM32F103RC_bigtree" and this excludes using "EPROM" emulation.

bryan-pl, 2020-02-17:
Confirmed,  had the same board behaviour.
PS_ON_PIN is set LOW after board startup regardless of PSU_ACTIVE_HIGH setting. 
The workaround is to add call for function PSU_ON(); before end of setup() function body in MarlinCore.cpp

dziabi3, 2020-02-18:
I solved the problem by setting "default_envs = STM32F103RC_bigtree_512K".
After several tests, it turned out that my motherboard has 512kB of flash memory. Thanks to this, the PSU_CONTROL option works and also the EPROM emulation department works.

usk-johnny-s, 2020-02-25:
PC12 conflicts USBSerial of maple.
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/144#issuecomment-590925789

dziabi3, 2020-02-25:
Can you explain it ?
I don't see it in the SKR mini E3 V1.2 diagram.
To remind you, I quote myself:

When I set "default_envs = STM32F103RC_bigtree_NOUSB or "default_envs = STM32F103RC_bigtree_512K_NOUSB" pin PS_ON does not work properly and the printer turns off after a few seconds.

This is just the opposite.

usk-johnny-s, 2020-02-26:
I just reviewed souce code. (I did not build and execute with except "STM32F103RC_bigtree_NOUSB". )

"_NOUSB" is just use compiler options "-DUSE_USB_COMPOSITE" or not. (not defining  "USE_USB_COMPOSITE" at "_NOUSB".)
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/5123ce60d1f7a88e9e6686934db38046e9360b3f/firmware/V1.2/Marlin-2.0.x-SKR-Mini-E3-V1.2/platformio.ini#L280-L383

But USB Serial functioning at both.
Detailed diffalence of PC12 as follows.

If difined "USE_USB_COMPOSITE" , "USBSerial" is "MarlinCompositeSerial", otherwise "Serial".
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/5123ce60d1f7a88e9e6686934db38046e9360b3f/firmware/V1.2/Marlin-2.0.x-SKR-Mini-E3-V1.2/Marlin/src/HAL/HAL_STM32F1/HAL.h#L63-L67

"MarlinCompositeSerial" is USBSerialComposite class.
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/5123ce60d1f7a88e9e6686934db38046e9360b3f/firmware/V1.2/Marlin-2.0.x-SKR-Mini-E3-V1.2/Marlin/src/HAL/HAL_STM32F1/msc_sd.cpp#L24

Then, USBCompositeSerial.begin() is nothing to touch  "BOARD_USB_DISC" pin.
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/0383384f4cc91aadee1501e11bf08dd3e62d9fb9/STM32F1/libraries/USBComposite/USBCompositeSerial.cpp#L45-53
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/0383384f4cc91aadee1501e11bf08dd3e62d9fb9/STM32F1/libraries/USBComposite/USBComposite.cpp#L126-128
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/0383384f4cc91aadee1501e11bf08dd3e62d9fb9/STM32F1/libraries/USBComposite/USBComposite.cpp#L100-114

The other "Serial" is USBSerial class. I commented at that.
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/144#issuecomment-590925789

dziabi3, 2020-02-26:
I was blind ;-)
Thank you very much, you are wonderful.

closed 2020-02-26 by dziabi3