bigtreetech/BIGTREETECH-SKR-mini-E3 #187: I get the following error when trying to ompile the firmware.[BUG] (short description)

KyleLampkin, 2020-02-19:
### Description
Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../../../../../inc/MarlinConfigPre.h:52:49: fatal error: ../HAL/HAL_STM32F1/inc/C
onditionals_LCD.h: No such file or directory

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

Branden0502, 2020-02-20:
I am getting the same error. have you had any luck?

lamosca01, 2020-02-23:
Yo es que cada vez veo más difícil copilar los nuevos update de marlin para esta placa. Creo que sera culpa de los cambios que han hecho en el propio marlin.

Skybeat, 2020-02-26:
I have the same issue. Any soulution for this one?