bigtreetech/BIGTREETECH-SKR-mini-E3 #189: [BUG] Cannot enable BL Touch without rom overflow

VanKurt, 2020-02-26:
I have downloaded the latest Marlin 2.0.4.3, and patched in the BigTreeTech config*.h files.
Then I enabled BL_TOUCH in the configuration.

The build then fails with the following error message:

> region `rom' overflowed by 588 bytes

Is there any way to get this working?
Do I need to disable other features (which ones won't hurt that much?)?
Why has the board so little memory?

brew99, 2020-02-27:
set the default environment in the platformio.ini file to "STM32F103RC_bigtree_512K_USB"

qwewer0, 2020-03-25:
[Guide]( https://www.reddit.com/r/ender3/comments/e894j7/marlin_20x_guide_for_ender_3_using_skr_mini_e3_v12/ ) for the board.

I think this can be closed.