bigtreetech/BIGTREETECH-SKR-mini-E3 #190: [FR] Add BLTouch firmware source to repository

pdearman, 2020-02-27:
**Is your feature request related to a problem? Please describe.**
I've recently installed a SKR Mini E3 v1.2 in my Ender-3 Pro and no matter how many times I try, none of the firmware that I compile myself works properly, specifically with my BLTouch. Every time I compile I get some sort of a different issue with either homing and/or auto leveling. It is a confirmed genuine BLTouch v3.1 and just to be safe I bought a second one which has the same issues.

I don't remember all the different error messages I've received but most of the time the two most common issues I've ran into are:
1. When homing it will move to center of bed (Z_SAFE_HOMING) as expected, deploy the BLTouch's probe, move up a bit, then have a fatal error requiring me to power cycle the printer.
2. Homing works here, but when auto leveling it moves to it's first point (lower left corner), deploys the probe and immediately says it's saved the settings and no probing actually takes place. The probe remains deployed until manually stowed. All other functions work properly after this happens and running auto leveling again results in the same behavior every attempt.

**Describe the solution you'd like**
Nothing seems to work except your pre-compiled "firmware-bltouch-for-z-homing.bin". It works like a charm, however I was disappointed that the source for it is not available for download to compare and figure out what I've been doing wrong. So if possible, please release the source for the pre-compiled BLTouch firmware files. I believe that this would be a huge help to others who are facing similar issues to troubleshoot their issues more efficiently.

**Describe alternatives you've considered**
I have followed the requirements in the readme, as well as many different tips and tricks found in the Issues here and other sites. The only thing that has worked reliably is the pre-compiled BLTouch firmware found here.

**Additional context**
I've already spent the better part of a month working on this and am invested to the point where I'd rather still try to understand the issue myself instead of asking for help (a bit hard headed that way). That is why I'm here asking for the source instead of directly asking for someone to help me fix it. Which in the long run might hopefully help others too.

Thank you for reading.

dziabi3, 2020-03-07:
In the attachment you will find my configuration file, compare it with your file.
I'm a newbie but I was able to launch BLTouch the first time.
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4302354/Configuration.zip)

pdearman, 2020-04-04:
> In the attachment you will find my configuration file, compare it with your file.
> I'm a newbie but I was able to launch BLTouch the first time.
> [Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4302354/Configuration.zip)

Hey thanks for that. Sorry for the late response! I had to take a break from the 3D printing stuff. It was driving me insane and apparently BTT doesn't seem to monitor this apparently... I'll give your config a try today probably and let you know how it goes. So far there are a few things I see that are different in yours. Like the Z Safe Homing option you have disabled. Before when I had that disabled I'd ALWAYS get bed crashes because the probe itself hangs over the edge of my bed. I'll give it a shot though. Thanks again.

pdearman, 2020-04-05:
Messed around with this for a while last night and discovered that the entire time I was having issues, it was the SD card causing issues somehow. If I have the SD card in while turning on the printer (updating firmware or not) BLTouch fails to initialize every time. If I take the card out and boot it up, works perfectly fine. No idea what could possible cause that, but for now I can at least print.