bigtreetech/BIGTREETECH-SKR-mini-E3 #191: [wont build] (The terminal process terminated with exit code: 1)

henriquetirolli, 2020-03-03:
Im trying to build my marlin and i do know why i cant do it. Please someone help me.
here is what is showing up.

Executing task: platformio run <

Warning! `env_default` configuration option in section [platformio] is deprecated and will be removed in the next release! Please use `default_envs` instead
Can not remove temporary directory `/Users/henriquetirolli/Desktop/Marlin-test-pontin/.pioenvs`. Please remove it manually to avoid build issues
Processing LPC1768 (platform: nxplpc@<3.4.0; board: lpc1768)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc/lpc1768.html
PLATFORM: NXP LPC 3.3.2 > NXP mbed LPC1768
HARDWARE: LPC1768 100MHz, 32KB RAM, 512KB Flash
DEBUG: Current (custom) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
LibraryManager: Installing dev
Error: Please read http://bit.ly/package-manager-ioerror
SSLError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /MarlinFirmware/U8glib-HAL/archive/dev.zip (Caused by SSLError(SSLError(1, u'[SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)'),)):
  File "/Users/henriquetirolli/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py", line 156:
    env.SConscript("$BUILD_SCRIPT")
  File "/Users/henriquetirolli/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/henriquetirolli/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "/Users/henriquetirolli/.platformio/platforms/nxplpc/builder/main.py", line 125:
    target_elf = env.BuildProgram()
  File "/Users/henriquetirolli/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/Users/henriquetirolli/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 62:
    env.ProcessProjectDeps()
  File "/Users/henriquetirolli/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/Users/henriquetirolli/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 131:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "/Users/henriquetirolli/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/Users/henriquetirolli/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 1029:
    project.install_dependencies()
  File "/Users/henriquetirolli/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 861:
    lm.install(uri)
  File "/Users/henriquetirolli/.platformio/penv/lib/python2.7/site-packages/platformio/managers/lib.py", line 278:
    force=force,
  File "/Users/henriquetirolli/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 698:
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "/Users/henriquetirolli/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 480:
    dlpath = self.download(url, tmp_dir, sha1)
  File "/Users/henriquetirolli/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 195:
    raise e
============================================================================ [FAILED] Took 0.64 seconds ============================================================================

Environment           Status    Duration
--------------------  --------  ------------
megaatmega2560        IGNORED
megaatmega1280        IGNORED
at90usb1286_cdc       IGNORED
at90usb1286_dfu       IGNORED
DUE                   IGNORED
DUE_USB               IGNORED
DUE_debug             IGNORED
LPC1768               FAILED    00:00:00.639
melzi                 IGNORED
melzi_optiboot        IGNORED
rambo                 IGNORED
sanguino_atmega644p   IGNORED
sanguino_atmega1284p  IGNORED
STM32F1               IGNORED
STM32F4               IGNORED
teensy35              IGNORED
malyanm200            IGNORED
esp32                 IGNORED
====================================================================== 1 failed, 0 succeeded in 00:00:00.639 ======================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

brew99, 2020-03-04:
Are you using the BTT version of Marlin, or the Vanilla Marlin?   Are you actually using a SKR mini E3 board?

Right now, you don't have the correct board defined in your configuration.h and also the default environment in platformio.ini.

henriquetirolli, 2020-03-04:
hi @brew99, im using a mks sbase. it was fine since last monday when i tried to build the soft and wont build anymore.

im sharing the link of my software with you cam help me i would apreciate so much.

https://drive.google.com/drive/folders/1oh9zumbug5Ob9-Wi-ceQaECWM8Z_Cog_?usp=sharing

brew99, 2020-03-04:
Not sure why you posted in the Bigtreetech github as I don't believe they make that board. 

You might try looking/posting [here](https://github.com/makerbase-mks/MKS-SBASE/issues)