bigtreetech/BIGTREETECH-SKR-mini-E3 #192: [BUG] Compiling with Marlin 2.0.4.4 fails (outdated configuration files)

patoberli, 2020-03-06:
When using Marlin 2.0.4.4 as the source, compiling fails with the provided configuration files here in the repository.
Error message in vs code:

In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/HAL.cpp:30:
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:45:4: error: #error "You are using an old Configuration.h file, update it before building Marlin."
   #error "You are using an old Configuration.h file, update it before building Marlin."
    ^~~~~
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/HAL.cpp:30:
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:49:4: error: #error "You are using an old Configuration_adv.h file, update it before building Marlin."
   #error "You are using an old Configuration_adv.h file, update it before building Marlin."
    ^~~~~
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:316:4: error: #error "LEVEL_CORNERS_INSET is now LEVEL_CORNERS_INSET_LFRB . Please update your Configuration.h."
   #error "LEVEL_CORNERS_INSET is now LEVEL_CORNERS_INSET_LFRB . Please update your Configuration.h."
    ^~~~~
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:1231:6: error: #error "Z_SAFE_HOMING is recommended when homing with a probe. Enable it or comment out this line to continue."
     #error "Z_SAFE_HOMING is recommended when homing with a probe. Enable it or comment out this line to continue."
      ^~~~~
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/HAL_SPI.cpp:35:
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:45:4: error: #error "You are using an old Configuration.h file, update it before building Marlin."
   #error "You are using an old Configuration.h file, update it before building Marlin."
    ^~~~~
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/HAL_SPI.cpp:35:
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:49:4: error: #error "You are using an old Configuration_adv.h file, update it before building Marlin."
   #error "You are using an old Configuration_adv.h file, update it before building Marlin."
    ^~~~~
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:316:4: error: #error "LEVEL_CORNERS_INSET is now LEVEL_CORNERS_INSET_LFRB . Please update your Configuration.h."
   #error "LEVEL_CORNERS_INSET is now LEVEL_CORNERS_INSET_LFRB . Please update your Configuration.h."
    ^~~~~
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:1231:6: error: #error "Z_SAFE_HOMING is recommended when homing with a probe. Enable it or comment out this line to continue."
     #error "Z_SAFE_HOMING is recommended when homing with a probe. Enable it or comment out this line to continue."
      ^~~~~
*** [.pio/build/STM32F103RC_bigtree/src/src/HAL/HAL_STM32F1/HAL.cpp.o] Error 1
*** [.pio/build/STM32F103RC_bigtree/src/src/HAL/HAL_STM32F1/HAL_SPI.cpp.o] Error 1

thisiskeithb, 2020-03-06:
Don't mix & match configs with different versions of Marlin. Either use [stock Marlin](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) & [configs](https://github.com/MarlinFirmware/Configurations) or use this repo & configs.

patoberli, 2020-03-06:
I know that I should not mix it :) But I hope with this bug report that bigtreetech could release a newer config file version that is compatible with 2.0.4.x.
In the readme here: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2 they also state (more or less) under point 3 to use the official firmware alternatively and then under point 4 they continue with copying the configure files to the directory. 
Yes I know, this is meant for the firmware here, but it would still be nice if it would work with newer Marlin builds :)
Thanks

Thinkersbluff, 2020-03-06:
> it would still be nice if it would work with newer Marlin builds :)

One of the big surprises for me, when I thought how much fun it would be to learn to 3D print, was how much programming and hardware engineering I would also need to learn, if I expected to be my own service desk... 

If you use a Windows PC and if you would like to be able to easily and carefully update your Config files to use the latest Marlin, I can recommend [WinMerge](https://winmerge.org/).  It lets you do a side-by-side, line-by-line comparison of old and new, automatically highlighting all differences and allowing you to make the changes as you go, then save the updated file.  That's how I recently "moved up" to the latest BugFix version.

patoberli, 2020-03-07:
Thanks for the tips, I guess I have it to do this way. 
Could you maybe upload your updated configuration for the less capable users?

BiOs2k, 2020-03-07:
Hello, It's my first time I compile a clean Marlin FW (2.0.4.4), I'm new to this but I did alot of homework before playing with that :). 
Ok for my part here is what I did. 
configure PlatformeIO for the env_default = STM32F103RC_bigtree_512K
and then, modify configuration.h and configuration_adv.h from the Marlin 2.0.4.4 source. using the original Bigtreetech config file.  (side by side windows)
this at the same time give the the ability to see what is the difference between all the config and read all the parameter to learn what do what and why.

took me about 1 hours and flash to my board and I have a perfect running firmware.
I took the take to set the LIN_ADVANCE and my E-STEPS Setting directly in the firmware.

Buy the way. I set TMC Current to 550 and activate monitoring and TMC Debug.

TheDoc97, 2020-03-11:
I did same as @BiOs2k and have it running fine, 2.0.4.4 on ender 3 Pro with bltouch v3.1 original.

lizaoreo, 2020-03-15:
@BiOs2k @TheDoc97 Would either of you mind sharing your configs?  I had mine working and then it suddenly stopped working, I'm getting stuttering on the Z stepper.  Was told it looks like voltage issue.  If I flash the precompiled firmware from BTT, it's working, but I had set my offsets and stuff for my BLTouch in the firmware and can't do that on the precompiled as far as I know.  I'm not sure why it's not working now, even going to an older version of my builds isn't working.  Tried going to 2.0.5 and redoing the whole config (using the BTT config on Marlin's git as a template), but no luck.

TheDoc97, 2020-03-15:
[Marlin-2.0.x.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4335111/Marlin-2.0.x.zip)

TheDoc97, 2020-03-15:
Here you are

lizaoreo, 2020-03-15:
Awesome, thanks @TheDoc97 , that was fast :)

lizaoreo, 2020-03-15:
And that worked, now to compare and see what's different because I want to know _why_ it wasn't working.  Thank you, got me going again.

TheDoc97, 2020-03-15:
you’re welcome

Chak10, 2020-03-26:
Compiling with Marlin 2.0.5.2

Error: Unknown environment names 'STM32F103RC_bigtree_512K'. Valid names are 'melzi_optiboot, at90usb1286_cdc, STM32F407VE_black, rumba32_mks, STM32F103RC_btt_512K_USB, mega2560, mks_robin_pro, sanguino1284p, STM32F7, teensy35, STM32F103VE_longer, STM32F103RC_fysetc, STM32F103RC_btt_512K, BIGTREE_GTR_V1_0, STM32F401VE_STEVAL, esp32, STM32F103RC_btt, STM32F103CB_malyan, STM32F103RC_btt_USB, melzi, FLYF407ZG, linux_native, FYSETC_F6_13, FYSETC_F6_14, jgaurora_a5s_a1, mks_robin_mini, mks_robin, STM32F103RC, DUE_debug, LPC1769, rumba32_f446ve, STM32F103RE, BIGTREE_SKR_PRO, STM32F103RE_btt_USB, ARMED, mks_robin_nano, SAMD51_grandcentral_m4, mks_robin_lite, FYSETC_S6, BIGTREE_BTT002, at90usb1286_dfu, teensy31, LPC1768, STM32F103VE_GTM32, rambo, STM32F4, include_tree, mega1280, sanguino644p, DUE, chitu_f103, STM32F103RE_btt, mks_robin_lite3, DUE_USB'
The terminal process terminated with exit code: 1

GhostlyCrowd, 2020-03-26:
> STM32F103RC_bigtree_512K

your error tells you exactly what the problem is if you read it, your trying ot use a platform name that doesn't exist in the platformio enviroment.

Use STM32F103RC_btt_512K

Chak10, 2020-03-28:
> > STM32F103RC_bigtree_512K
> 
> your error tells you exactly what the problem is if you read it, your trying to use a platform name that doesn't exist in the platformio environment.
> 
> Use STM32F103RC_btt_512K

Yes, I saw it later, sorry

GhostlyCrowd, 2020-03-28:
> > > STM32F103RC_bigtree_512K
> > 
> > 
> > your error tells you exactly what the problem is if you read it, your trying to use a platform name that doesn't exist in the platformio environment.
> > Use STM32F103RC_btt_512K
> 
> Yes, I saw it later, sorry

No need to be sorry, I may have came across as abrasive, some times written response can do that, but I was simply trying to point out your error, explain your mistake and then guide you in the right direction.

patoberli, 2020-03-29:
> No need to be sorry, I may have came across as abrasive, some times written response can do that, but I was simply trying to point out your error, explain your mistake and then guide you in the right direction.

I also had this issue, sadly nowhere in the release notes it was stated that there was a renaming done between 2.0.4 and 2.0.5.

Jonni95, 2020-04-18:
Hello all,
actually i have the same problems with vscode. Is there anybody who can help me? I already tried to fix this Problems with  few different Marlin versions, without a solution.

**Terminal Output:**
     Executing task: C:\Users\user\.platformio\penv\Scripts\platformio.exe run <

    Processing STM32F103RC_btt_512K_USB (platform: ststm32; board: genericSTM32F103RC; framework: arduino)
    ------------------------------------------------------------------------------------------------------------------------------------------------

    Verbose mode can be enabled via `-v, --verbose` option
    CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
    PLATFORM: ST STM32 6.1.0 > STM32F103RC (48k RAM. 256k Flash)    
    HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 512KB Flash
    DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
    PACKAGES:
     - framework-arduinoststm32-maple 2.10000.200103 (1.0.0)        
     - tool-stm32duino 1.0.2
     - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
    Converting Marlin.ino
    LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
    LDF Modes: Finder ~ chain, Compatibility ~ soft
    Looking for TMCStepper library in registry
    Found: https://platformio.org/lib/show/5513/TMCStepper
    LibraryManager: Installing id=5513 @ >=0.6.2,<1.0.0
    Using cache: C:\Users\user\.platformio\.cache\72\6a70f906c548b2cee9147f8f4c763872
    Unpacking...
    FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\user\\Downloads\\BIGTREETECH-SKR-mini-E3-master (1)\\BIGTREETECH-SKR-mini-E3-master\\firmware\\V1.2\\Marlin-2.0.x-SKR-Mini-E3-V1.2\\.pio\\libdeps\\STM32F103RC_btt_512K_USB\\_tmp_installing-le80jq0g-package\\.\\examples\\Calibrate_spreadCycle\\Calibrate_spreadCycle.ino':
      File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 156:
        env.SConscript("$BUILD_SCRIPT")
      File "C:\Users\user\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 597:
        return _SConscript(self.fs, *files, **subst_kw)
      File "C:\Users\user\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
        exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
      File "C:\Users\user\.platformio\platforms\ststm32\builder\main.py", line 98:
        target_elf = env.BuildProgram()
      File "C:\Users\user\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
        return self.method(*nargs, **kwargs)
      File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
        env.ProcessProjectDeps()
      File "C:\Users\user\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
        return self.method(*nargs, **kwargs)
      File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 131:
        project_lib_builder = env.ConfigureProjectLibBuilder()
      File "C:\Users\user\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
        return self.method(*nargs, **kwargs)
      File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1035:
        project.install_dependencies()
      File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 867:
        lm.install(uri)
      File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 280:
        force=force,
      File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 700:
        pkg_dir = self._install_from_piorepo(name, requirements)
      File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 130:
        requirements,
      File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 482:
        self.unpack(dlpath, tmp_dir)
      File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 222:
        return fu.unpack(dest_dir, with_progress=False)
      File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 149:
        self._unpacker.extract_item(item, dest_dir)
      File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 82:
        super(TARArchive, self).extract_item(item, dest_dir)
      File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 39:
        self._afo.extract(item, dest_dir)
      File "C:\Users\user\.platformio\python37\lib\tarfile.py", line 2044:
        numeric_owner=numeric_owner)
      File "C:\Users\user\.platformio\python37\lib\tarfile.py", line 2114:
        self.makefile(tarinfo, targetpath)
      File "C:\Users\user\.platformio\python37\lib\tarfile.py", line 2155:
[... 38 lines cut ...]
    mks_robin_pro             IGNORED
    mks_robin_lite            IGNORED
    mks_robin_lite3           IGNORED
    jgaurora_a5s_a1           IGNORED
    STM32F103CB_malyan        IGNORED
    chitu_f103                IGNORED
    STM32F401VE_STEVAL        IGNORED
    FLYF407ZG                 IGNORED
    FYSETC_S6                 IGNORED
    STM32F407VE_black         IGNORED
    BIGTREE_SKR_PRO           IGNORED
    BIGTREE_GTR_V1_0          IGNORED
    BIGTREE_BTT002            IGNORED
[... cut ...]
    teensy35                  IGNORED
    esp32                     IGNORED
    linux_native              IGNORED
    SAMD51_grandcentral_m4    IGNORED
    rumba32_f446ve            IGNORED
    rumba32_mks               IGNORED
    include_tree              IGNORED
    ==================================================== 1 failed, 0 succeeded in 00:00:02.369 ==================================================== 
    The terminal process terminated with exit code: 1

    Terminal will be reused by tasks, press any key to close it.

BiOs2k, 2020-04-18:
Hello, 
may be because to Long directory structure.
create a c:\marlin
and put your source over there.
make is shorter as possible.

remove all project in platformio and open the marlin project..
then try to compile.

BiOs2k, 2020-04-18:
use this page as reference. there is some tweak you can do depending on your setup and requirement but this a good starting document
https://www.reddit.com/r/ender3/comments/e894j7/marlin_20x_guide_for_ender_3_using_skr_mini_e3_v12/

Jonni95, 2020-04-18:
> Hello,
> may be because to Long directory structure.
> create a c:\marlin
> and put the marlin build source over there.
> make is shorter as possible.
> remove all project in platformio and open the marlin project..
> then try to compile.
> and by the way, don't play with the bigtreetech source, it is old and not correctly tweaked.
> I Have the latest Marlin 2.0.5.3 source.

Thanks a lot man!!! I created a Shorter Directory and that's it! Took me a few days...

BiOs2k, 2020-04-19:
> Thanks a lot man!!! I created a Shorter Directory and that's it! Took me a few days...

Yeah, Happen to me when I start compiling. I've found the solution somewhere on internet. 
I saw you error with this directory path:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\user\\Downloads\\BIGTREETECH-SKR-mini-E3-master (1)\\BIGTREETECH-SKR-mini-E3-master\\firmware\\V1.2\\Marlin-2.0.x-SKR-Mini-E3-V1.2\\.pio\\libdeps\\STM32F103RC_btt_512K_USB\\_tmp_installing-le80jq0g-package\\.\\examples\\Calibrate_spreadCycle\\Calibrate_spreadCycle.ino':

That's too long.! make sure that you working directory is at shortest as possible.

ipatch, 2020-07-15:
try bumping the below [line](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/Configuration_adv.h#L34)

```c
#define CONFIGURATION_ADV_H_VERSION 020005
```

from that 👆, to this, 👇

```c
#define CONFIGURATION_ADV_H_VERSION 020006
```

Personally, i clone the bugfix branch of vanilla marlin, then reapply (patch) in my configuration changes related to the hardware of my 3d printer. when patching in changes, if you run into build errors, look at the error messages and they can be considered quite obvious and point you to what configuration setting requires updating in order to build.

I was running into the same error trying to compile the lasted bugfix branch of marlin, doing a quick search yielded this github issue, and vaguely remembered the configuration files are _versioned_ so if your trying to compile from upstream master, make sure to update the _version_ of your configuration files. good luck.

GhostlyCrowd, 2020-07-15:
there are more changes then just that @ipatch between x5 and x6