bigtreetech/BIGTREETECH-SKR-mini-E3 #193: Issues compiling marlin to work with skr mini and bltouch[BUG] (short description)

RamboTron, 2020-03-12:
I keep getting these errors when compiling marlin 2.0x and im not sure how to fix them. I have no idea what im doing :)....

#error "TEMP_BED_PIN must be defined using the Pn_nn or Pn_nn_An format. (See the included pins files)."

#error "Oops! Select an STM32F1 board in 'Tools > Board.'"

'PA1' was not declared in this scope

'PA4' was not declared in this scope

RamboTron, 2020-03-12:
update! i changed the default environment in platform.ini from lpc1768 to STM32F103RC_bigtree_512K and it builds successfully...however it wont upload and i dont know where to find the firmware.bin file

TheDoc97, 2020-03-12:
In Marlin folder, .pio subfolder, STM32F103RC_bigtree_512

bskinnell, 2020-03-12:
I am having the exact same issue trying to compile.  I have tried every option of "STM32F103RC_*" and all error out.  I have confirmed that I'm using the most up to date PlatformIO core as well as the latest Marlin 2.0 bugfix.

NickdeK, 2020-03-20:
`STM32F103RC_btt_512K` works for me! This should be changed in the readme.

qwewer0, 2020-03-25:
[Guide]( https://www.reddit.com/r/ender3/comments/e894j7/marlin_20x_guide_for_ender_3_using_skr_mini_e3_v12/ ) for the board.