bigtreetech/BIGTREETECH-SKR-mini-E3 #194: [BUG] Plugin in the usb while PSU is ON

Dimitar-88, 2020-03-15:
When Plug in the USB in the board and the board is external powered (from the PSU) the printer freezes and starts to make a strange noise  like it is outputting PWM modulated signal ot the stepper motors.  If i have the board plugged in only via USB and no external power it's fine it power's up and there is no problem.

The board is SKR Mini E3 V1.2
I have my firmware attached.
I used TF35 LCD screen in combo with the board.

### Steps to reproduce

1. Power up the board with external power.
2. Plug in the USB to the PC  (windows 10) and to the board

**Expected behavior**
it should be possible to print from USB

**Actual behavior**
Not possible to print over USB

### Additional Information

[SKR Mini E3 V1.2.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4333948/SKR.Mini.E3.V1.2.zip)

Dimitar-88, 2020-03-20:
@bigtreetech Do you guy's have any suggestion ?

dziabi3, 2020-03-20:
The problem may be USB power supply.
Me and others from this forum seal the tape or pull out the 5V pin.

[image]

Dimitar-88, 2020-03-22:
@dziabi3  Thanks for the suggestion but it did not help :(

mjimm, 2020-03-29:
Are you using Cura?  Is Cura open when you plug in USB? 
(https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/198)

Dimitar-88, 2020-04-04:
@mjimm Yes i use Cura. I have tried both with and without it open:
1. When Cura is not open the printer continues to work until i start it.
2. When Cura is open it directly freazes
3. I have tried to upload the firmware from the VSCode but when i hit upload it dies.

I have tested all scenatios with USB +5 V line isolated or connected.

mjimm, 2020-04-05:
Hi Dimitar,

1. Disable Cura serial/usb programmer.  Comment out the folowing lines in file *\Ultimaker Cura 4.4\plugins\USBPrinting\AutoDetectBaudJob.py
# try:
# programmer.connect(self._serial_port)
# serial = programmer.leaveISP()
# except ispBase.IspError:
# programmer.close()
[https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/198]
[https://community.ultimaker.com/topic/30043-does-not-connect-with-ender3-and-skr-mini-e3-board/]

2. Follow the BigTreeTech firmware compilation guide.  The firmware is compiled to BIN file, and then copied to/loaded via the SD card.  The serial/usb bootloader is not used for the SKR platform.
[https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2]

Hope this helps :)

Dimitar-88, 2020-04-13:
Worked great :)

closed 2020-04-13 by Dimitar-88