bigtreetech/BIGTREETECH-SKR-mini-E3 #196: Programing via USB instead of SD?

drphil3d, 2020-03-16:
I would like to see if anyone has had luck with programming the SKR mini e3 via usb, I tried using flymcu but it doesn't seem to work sense it doesn't have a usb to serial chip just direct usb to the STM32. I'm not sure if the programming usb port on the STM32 is exposed or not. I'm designing an adapter kit to put this board in old printrbots with the least amount of modification. I have the choice of the USB or SD card being easy to access. 
[image]

This board uses a CH340 which makes it possible to program via usb easily. 
 https://wiki.fysetc.com/Cheetah_Board/

B0M0A0K, 2020-03-17:
I have tried for quite some time to get my SKR-Mini-E3 to talk to VSC or even an Octopi without any success. As far as I am aware this is a MAJOR limitation of this board. It's very strange because I can talk to the board using USB via Pronterface, just not VSC/PlatformIO. I don't know if this is something that Bigtreetech could fix via firmware update, but I don't think they are interested and have moved on to their latest and greatest board. It's a shame because this and not being able to connect to Octopi for control are the two main things that let this board down.

drphil3d, 2020-03-17:
you just didn't set the serial port config correctly in marlin. It looks like the STM32 would need a different bootloader or serial programmer which is what the Cheetah uses.

dziabi3, 2020-03-20:
I send gocd from Octoprin to the SD carriage.
Initially I had problems but when I turned off the AUTO_REPORT_TEMPERATURES option, the write errors disappeared.

drphil3d, 2020-03-21:
That's true huh, you could just leave the SD card plugged in and upload the new firmware.bin file to the sd card and power cycle...But the only problem is, if for some reason you accidently flash incorrect firmware or leave out the host drive feature... well then if the sd card slot is hard to access you're going to be taking things apart.

closed 2020-03-21 by drphil3d