bigtreetech/BIGTREETECH-SKR-mini-E3 #197: [BUG] (SKR m. E3 (stock)LCD crashes/ random pixels or blank/ clicking by itself)

Gavalakis, 2020-03-20:
### Description

### Steps to reproduce

1. No clue how to reproduce
2. Both times it crashed after printing for more than an hour and the occasional menu clicking was becoming frequent.

**Expected behavior**

**Actual behavior**

ShadowBr0ther, 2020-04-11:
Same issue here

Gavalakis, 2020-04-11:
> Same issue here

Somehow it fixed itself after those 2 times and it hasn't happened since. Same for the menu clicking. I really don't know what fixed it though :(

kubik256, 2020-11-06:
Same issue here.

Sometimes garbage is shown on LCD and rotary encoder is incidentally self-clicking.
I've just flashed Marlin release 2.0.7.2 - and the same is happening right now as I'm typing :-)

thekdub, 2021-01-20:
Same happens for me with SKR Mini E3 V1.2 using Marlin Bugfix-2.0.x (2.0.6 potentially?) and an otherwise stock Ender 3.

Display randomly glitches and becomes unusable during long prints, only solution I've found is to power cycle. Does not seem to cause any issues with printing, aside from being unable to use the display or change settings mid-print.