bigtreetech/BIGTREETECH-SKR-mini-E3 #198: cura print via usb cable

jpokorny12, 2020-03-20:
Hello
There is a possibility to print directly from Cura via USB cable. With the stock plate on ender 3 it went. Now I have skr3 mini. Repetier hosts the COM port and controls the printer.
Thanks

dziabi3, 2020-03-20:
Several times I tried to use CURA, each time the SKR Mini E3 motherboard stopped responding, and the heaters reached 100% power.

I also used Repetier-Host and Octoprint.
In both cases, the 3D printer stopped for several hundred ms.
The problem disappeared when I turned off the "AUTO_REPORT_TEMPERATURES option"

I recommend you use Raspberry Pi and Octoprint.
It works perfectly ;-)

Dimitar-88, 2020-03-22:
I have same problem. When i plug in the USB the board is fine. As soon as Cura load's up it dies

mjimm, 2020-03-29:
Try this reddit thread - Cura usb upsets the mainboard somehow.  Works for me.
(https://www.reddit.com/r/BIGTREETECH/comments/dp6k6r/skr_mini_e3_v12_and_curausb/)

Adjust the AutoDetectBaudJob.py file in *\Ultimaker Cura 4.4\plugins\USBPrinting
Comment out following lines:
# try:
# programmer.connect(self._serial_port)
# serial = programmer.leaveISP()
# except ispBase.IspError:
# programmer.close()