bigtreetech/BIGTREETECH-SKR-mini-E3 #201: [BUG] Firmware guide missing Z_SAFE_HOMING for BLTouch

tgadberry1, 2020-03-26:
### Description

The guide for configuring the firmware is missing a critical step that will cause many hotends to crash when homing the z-axis. You must uncomment "#define Z_SAFE_HOMING" for the printer to safely move to the center of the bed before homing z.

### Steps to reproduce

1. Simply include this step in the README. Uncomment line 1306

**Expected behavior**
BLTouch should act as the Z-probe if configured to. 

**Actual behavior**
The hotend crashes into the build plate because the BLTouch is too far off and cannot probe the surface. 

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

HVJ17, 2020-03-27:
This is not really a bug. It probably should be on but it is optional. There are about 100 other changes to make not listed either when using your own 2.0.x build. I've made probably 200 changes to my build.

tgadberry1, 2020-03-27:
Sorry, relatively new to GitHub. Wasn’t sure where to note this.

mjimm, 2020-04-05:
Except if using BL touch for Z homing, you get a bed crash.

If the guide says "if you want use bltouch for z-homing, please enable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN", it shoud also say to enable safe homing..
#define Z_SAFE_HOMING