bigtreetech/BIGTREETECH-SKR-mini-E3 #203: [BUG] (short description)

Hxspider, 2020-03-28:
### Description

I updated my skr mini e3 v1.2 with the newest firmware-bltouch.bin ond my tft35 with the update 24 from github and I had the possibility to correct my extruder output to correct underextruding. 
Then I tried to update to the update 25 on my tft35, then I lost the correction options in my settings? I have tried to downgrade but it won't come back...
The only thing showing under machine settings/custom is home and flash from... 
And the icons under machine settings/custom is distorted. 

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**
Being able to correct extruder output to correct underextruding.

**Actual behavior**
The machine/custom settings is missing

### Additional Information
[image]

Clowntrigger, 2020-03-28:
This looks like a TFT35 problem with the icon name and firmware being incorrect. See [here](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/430#issuecomment-604664793) for a fix. 

I've found the BTT fork of Marlin has issues. You may want to follow [this guide](https://www.reddit.com/r/ender3/comments/e894j7/marlin_20x_guide_for_ender_3_using_skr_mini_e3_v12/) to get the latest version compiled from the mainline.

oldman4U, 2020-09-05:
@Hxspider 

Close this ticket and also https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/522 now.

closed 2020-09-05 by Hxspider