bigtreetech/BIGTREETECH-SKR-mini-E3 #204: Burned Mosfet

oOJoshOo, 2020-03-28:
Hello, i want to ask you whats connectet on the Mosfet in my Picture, because its burned and i want to check the Parts of my printer.

I also use the dcdc 5v adapter for Neopixels.
[image]

GhostlyCrowd, 2020-03-28:
Thats the hotbed mosfet. inspect your hotbed.

oOJoshOo, 2020-03-28:
Ok the insolation on the hotbed wasnt that tight on the contacts i will insolate it new and close the issue when it works with the new board

HVJ17, 2020-03-29:
Um, this is not a code issue. This should be posted on another website all together.

oOJoshOo, 2020-03-31:
HVJ17 i'm Sorry but it helped and now it works again thank you.

closed 2020-03-31 by oOJoshOo

esebuenjona, 2020-07-21:
I also use a dcdc for the cooling fan, and the same thing happens to me.

esebuenjona, 2020-07-21:
my bed was working perfectly for two months. I can't understand what is wrong, and what should I do in case I get another board. I have burned 2 already. One was the steper driver.

Moorviper, 2020-07-22:
This is a software related bug tracking system NOT a support website!

esebuenjona, 2020-07-22:
> This is a software related bug tracking system NOT a support website!

Thanks for letting me know. I was lacking on that knowledge.  Anyway. If is anything I can contribute let me know. I was just trying to get help, and if possible to help in some way. I'm going to try to replace the mosfet. bigtree  only offer a replacement instead of telling what could be the cause. Maybe is hardware, maybe software. I wish I could help.

HVJ17, 2020-07-22:
Go to ender 3 sub reddit and post your issue. Include your exact setup. Mods and addons.

GhostlyCrowd, 2020-07-22:
> This is a software related bug tracking system NOT a support website!

Sure, but you do realize the only reason they posted any of this to github is because of pressure from the community, its not like they take any effort in addressing anything on their repo for any of their repositories.  

Just food for thought before your next off the cuff remark on an absolutely Ancient bug report.