bigtreetech/BIGTREETECH-SKR-mini-E3 #205: Part cooling fan has no control for speed settings

srogers0277, 2020-03-29:
I recently upgraded my ender 3 with the SKR Mini E3 1.2 and so far it works great.  The one problem I am having is I do not have control of the part cooling fan.  I have read all the setting and gone through the videos with no luck.. I have tried swapping the fan1 and fan0 with no luck either.

Is there a setting in Marlin that I have to enable to get this working.  This is the only issue I have and I would love for it to be resolved.

Please help,,,,,,

Steve

Klaus3d, 2020-04-08:
Ich habe genau das selbe Problem.
Ist schon eine Lösung vorhanden?

merson316, 2020-04-20:
Having the same issue. Confirmed that both fan1 and fan0 are in the correct position (swapped from stock) and still seeing my part cooling fan at 100%, ignoring the setting on the screen.

Klaus3d, 2020-04-20:
Hallo, ja leider immer noch.Vielleicht hat das Board einen Fehler Gruß Von meinem Samsung Galaxy Smartphone gesendet.

  control for speed settings (#205) 

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/205#issuecomment-616745116",
"url": "https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/205#issuecomment-616745116",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

sregister, 2020-05-02:
Sounds like the same issue as https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/238 Anyone found a solution or heard back from BTT?

dturner, 2020-05-05:
Same issue here. The only thing I would add is that this is the same behaviour I was experiencing on my original Creality board. From marlin settings 0% = cooling fan off, 1-100% = cooling fan on 100% with voltage fixed at 24V. Would be good to get some control over the cooling fan.

ciidfrance, 2020-05-11:
Same issue here !

CRTified, 2020-05-23:
> Anyone found a solution or heard back from BTT?
@sregister - I made a comment at the referenced issue on how to replace the MOSFET, if you want to try it. For me, it works again.

@dturner It _might_ be possible that your problem with the missing control is caused by something else. You could try `FAN_SOFT_PWM`, but I've also read the suggestion to add a capacitor in parallel to the fan (but that would just smooth the PWM signal, and makes not that much sense at all).

txt4nk, 2020-05-25:
Have any of you tried to turn the fan off via the printer menu or a USB host (octoprint, etc)?