bigtreetech/BIGTREETECH-SKR-mini-E3 #206: Micro-pausa ao imprimir

Emersonrh, 2020-04-01:
apos atualizar a minha está com algum problema nos steps/jerk, ela não consegue fazer um arco grande sem dar várias micro-pausas no caminho. 
agora voltei a versão anterior mais o erro continua 

 ajude-me por favor 

[image]
[image]

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the BigTreeTech Facebook group: https://www.facebook.com/groups/505736576548648 or BigTreeTech Subreddit: https://www.reddit.com/r/BIGTREETECH/.
-->
### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

edoardoo, 2020-04-05:
I'd suggest you to write in english and to compile the bug template.
Are you using Octoprint? Are you printing from the display sdcard or from the controller sdcard?

Emersonrh, 2020-04-05:
Hello, I'm printing from the display using sdcard, I don't use octoprint, circular pieces look like this ( anexo) .

 pieces with straight lines are perfect