bigtreetech/BIGTREETECH-SKR-mini-E3 #208: Skr mini e3 v1.2 + tft 35 v2 problem

youneslak, 2020-04-02:
Hi,
I have connected my tft 35 v2 to my skr mini e3 v1.2 with marlin 2.0, but the screen doesn't show anything, it's on but nothing is showing.
I have followed the right wiring and still nothing.
Any help will be great
Thank you

rydol, 2020-04-02:
Set in config.h
#define SERIAL_PORT 2
#define SERIAL_PORT_2 -1

Update the firmware for TFT

If you have it correctly wired up, it should work

youneslak, 2020-04-02:
Thanks you @rydol 
Is there a way to update the firmware even if the screen is not showing anything?

rydol, 2020-04-02:
@youneslak  The screen should light up with its basic menu if its wired correctly. Even if you just power it up with 5v. Then you should see message:  printer connected  or no pritner connected at least. 
How did you plug it to the board ?

youneslak, 2020-04-02:
@rydol 
I have connect it just like this wiring scheme
[image]

but still the screen is just blank no menu or anything

rydol, 2020-04-02:
@youneslak  Seems good to me, what happens when you connect only 5V and GND ?