bigtreetech/BIGTREETECH-SKR-mini-E3 #211: Noisy stepper drivers

ddmkr666, 2020-04-05:
I purchased SKR Mini E3 a couple of months ago and it was working fine and as expected until a couple of days ago, when I noticed noise from the steppers that I haven't heard before. It's nowhere near stock board stepper drivers levels of noise, but it's definitely noticeable. That's in stealthchop of course. When I switched to spreadcycle the noise was unbearable, way louder than the stock board and sounds like grinding/crunching, definitely not like a healthy machine. 
Can anyone advise what can be the issue and is there any fix?

HVJ17, 2020-04-07:
What FW? BTT or your own Marlin?  Marlin 2 has #define HYBRID_THRESHOLD by default, its makes the motors scream with the default settings.

ddmkr666, 2020-04-08:
Marlin 2.0.3, 2.0.5 and default BTT firmware behave the same. Hybrid threshold is disabled, I know how it sounds and it's not that.

AnakinSpain, 2020-04-15:
I have the same noise, so, I deleted 2.0.5.3 and install the version of this repository modificated by me. But It is an old version, not actual.

GhostlyCrowd, 2020-04-15:
I have a friend running on  2.0.5.3 and no noises for him. Manually set your stepper plus width PPS for the 2209's and what not in advanced config.  Also dont use Junction deviation its still broken.

ddmkr666, 2020-04-16:
I tried on 2.0.5, 2.0.3, stock, klipper, no change. I already confirmed
with the vendor that it's hardware fault and I got sent a new board.

ddmkr666, 2020-04-16:
BTW how is junction deviation broken? I had zero issues with it.

AnakinSpain, 2020-04-16:
> Marlin 2.0.3, 2.0.5 and default BTT firmware behave the same. Hybrid threshold is disabled, I know how it sounds and it's not that.

Here there is a video with sound I say. 

https://youtu.be/FinCK_6aYf8

GhostlyCrowd, 2020-04-16:
> BTW how is junction deviation broken? I had zero issues with it. >

It still has issues with printing high detail models, especially circular ones  without chattering the steppers which causes print defects on anything curved or circular, the faster you print the worse it gets, so you may not notice it at 60mm/s i notice it at 150mm/s as do others.

ddmkr666, 2020-04-16:
Ok, I didn't notice anything like that and I almost never print below
100mm/s, but I'll take a look.

GhostlyCrowd, 2020-04-16:
> Ok, I didn't notice anything like that and I almost never print below 100mm/s, but I'll take a look.
> […](#)
> -- Marcin Jażdżewski ...the end of the beginning marks the beginning of the end... Blank Faces - http://blankfaces.pl

github.com/MarlinFirmware/Marlin/issues/17342 < this has been an ongoing issue. Use classic jerk trust me, it will eventually catch up with you.

ddmkr666, 2020-04-16:
I use klipper nowadays, so that's a non issue for me

HVJ17, 2020-04-17:
Has this been resolved? 
Junction deviation works flawlessly for me at all speeds.

ddmkr666, 2020-04-17:
Caused by a faulty hardware.

closed 2020-04-17 by ddmkr666

fftkatarn, 2020-05-18:
Same here, the board worked like a charm for about a month, now all the stepper motors make A LOT of noise... I asked the seller to send me a new board, I hope they do..

fftkatarn, 2020-05-20:
> I tried on 2.0.5, 2.0.3, stock, klipper, no change. I already confirmed with the vendor that it's hardware fault and I got sent a new board.

@ddmkr666 sorry if I revive this closed topic but I have the same hardware fault and I can't contact bigtreetech in any way. I tried to email them at info@bigtree-tech.com and write them on Aliexpress but I never had a reply.
How did you manage to get in touch with them?
Thank you!

ddmkr666, 2020-05-20:
I never got in touch with them. I never buy directly from any
Chinese company, as their customer service is basically non-existent. I
ordered from amazon and that's where I got my replacement.