bigtreetech/BIGTREETECH-SKR-mini-E3 #218: TMC Connection Error

skrap1r0n, 2020-04-16:
So a few days ago I installed my SKR mini E3 on my Ender 3 with BLTouch, and after some initial troubleshooting, got it printing. Yesterday afternoon, I started getting a TMC Connection Error.

I have Octopi hooked up and I know that if power from the pi is going to the board when you start up, it can cause this. I am still getting when I unhook the Pi.

M122 gives me the following:

Recv: Testing X connection... Error: All LOW
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing E connection... OK
I have double checked all connections and they are solid. I have done nothing with any jumpers. I have verified that the Marlin build is set to using the proper drivers.

#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define E0_DRIVER_TYPE TMC2209
I have a multimeter, could test the connection, but that is a bit north of my expertise without some guidance.

Any ideas where to go from here?

JWolfJr, 2020-04-16:
I had the same issue with my X axis as well.  I had also tethered the Pi to the SKR through USB at first, after a little research, because my printer wasn't acting right, I found out that the SKR could be damaged by the Pi's 5 volt going through the USB cable. I just placed a piece of electrical tape over the 5 volt connection on the usb cable, and the printer was fine.  About a month later I got the TMC error.  I researched the issue and found that I had to replace the board.  Have not had any issues since, it's been about 6 months now.  I don't know if it was the extra voltage coming in from the Pi that the SKR can't handle.  I think I read somewhere that it may have something to do with the design of the SKR.  It always seems to be the X axis that has the issue.  I have not found any work arounds to date, holding on to my first board in case I do find one.

danstreeter, 2020-04-16:
@skrap1r0n - I had exactly the same thing yesterday... Has been printing fine since ~10th Dec when I got my SKR and have been running the full SKR Mini E3 with firmware that I burned a few days after that, OctoPi WITH a taped 5v USB lead as advised.

Been printing without any issues consistently since then, and over the last few weeks have been printing PPE visors 8 hours of the day - then boom. X stepper fails with a TMC Connection Error on the LCD.

My M122 output is the same as yours.

skrap1r0n, 2020-04-16:
So am I looking at a new board then?

danstreeter, 2020-04-16:
I'm going to spend some time at the weekend trying to diagnose what is the problem and I'll report back here if I find anything.

JWolfJr, 2020-04-16:
Please let us know if you find anything.  I've only seen someone suggest to replace the actual driver chip, if you have the soldering skill and tools to do so.  But, at $35, I just bought another.  When I did install my 2nd board, I changed around my hardware setup.
1) made sure my wiring was clean and not passing anywhere near the driver heat sinks.
2) connected a 2nd noctua fan blowing straight down onto the drivers.
3) connected my Pi to the Uart pins instead of USB.
Keeping my fingers crossed, no problems yet and I print almost every day. Not 24/7 Covid-19 PPE, but definitely a lot of over night prints.  Hope this helps.

danstreeter, 2020-04-16:
Thanks @JWolfJr - do you have any resources on how to get the Pi talking to the SKR over UART rather than USB? I’d be interested in getting that sorted as the only thing hanging out the back of my setup is now the USB cable which really annoys me, and with the potential that its 5V pin is causing issues, would be great to remove that from the equation.

Did a bit of a quick look over this evening, and removing the pi from the mix and just powering the SKR still has the same error.
Wires are all neat and tidy, without any sign of any wear and tear...
Not got time for a full tear down tonight, but I’ll do that at the weekend.

JWolfJr, 2020-04-16:
No problem @danstreeter hope this helps.  Worked for me. 
https://community.octoprint.org/t/simple-3-wire-serial-connection-from-pi3-to-skr-v1-3-board/10166.
There will be pdf link a few posts down, click on that.
This is a pdf from bigtreetech, go to page 12.

danstreeter, 2020-04-16:
Thanks @JWolfJr 

For others reading here, also see #155

skrap1r0n, 2020-04-16:
@JWolfJr

I am also using the Teaching Tech case with all the electronics moved to the rear. Out of curiosity, how are you powering the Noctua? Those are 12v fans aren't they? Are you using a buck converter or are you pulling power from elsewhere? Also, did you stick with the TT case or did you move to something different?

JWolfJr, 2020-04-17:
I have 2 Noctuas, one at the designed case location by the Pi and the other I modified the lid to point straight down above the driver heat sinks.  I did place a small DC/DC buck converter on the Pi side just for the Noctua fans. I'm using these https://www.amazon.com/gp/product/B01MQGMOKI/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1. I thought about moving all the electronics including the PSU off the frame to a separate enclosure with a larger/quite fan.  That way I can use an enclosure and not worry about everything heating up.  I would like to start using PC, Nylon and other high temp filaments.

skrap1r0n, 2020-04-17:
okay, cool. I stuck with 24v fans because I didn't want to mess with buck converters. I may have to add another fan though, I don't want to overheat that board, if that is in fact what may be causing this issue.

JWolfJr, 2020-04-17:
I hear you, thought about it myself.  Less components to wire in, but I had them laying around from other projects and didn't want to add more fan noise.  I just wanted to eliminate all possibilities on my end, so careful wiring placement and plenty of cooling and no possibility of 5 volts from the Pi.  I'm also careful not to manually move my axis by hand too fast, I am using hybrid threshold and haven't messed with the amp settings.  So, if it goes again, has to be in the board then!?  Good luck.  It really is a nice board.

themixer2000, 2020-04-22:
having the same issue was printing fine and then went to start a print and tmc error did the m122 and got the exact message about x.

themixer2000, 2020-04-26:
I bought a new skr board and did one test print bed leveling thought everything was going great until the TMC Connection Error showed up I flashed the board with the newest firmware from Github and same issue so I did some test I switched the Y and X Stepper motors around same issue and Y works which tells me its not a stepper motor issue. Now I'm thinking power or maybe Bltouch or my outreached idea bad endstop on x?

jeeperjang, 2020-04-30:
My board started giving the dreaded TMC Connection Error too. I was told that commenting out the "#define MONITOR_DRIVER_STATUS" in the Configuration_adv.bin file might solve the issue. Problem is I can't get anything to recompile the firmware. I've tried Ardinuno IDE, Atom, and VSCode. They all fail to compile with an error stating the STM32F103RC board environment is no good. I've given up on the BTT SKR Mini E3 V1.2 board for now and have gone back to the stock Creality board on my Ender 3. I've also started posting warnings on various forums to keep people from buying these crappy boards.

WScottCross, 2020-05-05:
I now have 2 SKR Mini E3 boards that ran for a few days and now display TMC Connection Error.  They were in the same machine and will home X and Y but not Z axis.  Not sure what's going on.  I bought a 3rd board intending to put it in another machine, but now I don't know if I trust these boards any more.  I may try swapping out the Z stepper and the board to see if the motor is killing the drivers.  Am I looking in the wrong direction?

jeeperjang, 2020-05-07:
Sorry to hear that. I've given up on the Mini E3 board with the integrated stepper drivers and will not buy another one. Instead I ordered a SKR Mini E3 DIP which has sockets for plugging in discrete driver modules. This way when a driver dies I can replace just the driver module and not the entire control board.I don't think it's the motors that are killing the drivers since mine have been fine with the stock control board for almost 3 years. Might be a cooling issue or maybe their voltage setpoint is set too high causing them to overheat. This setpoint can be programmed in the firmware or manually set if you have plug-in driver boards. On your machine, you might want to check to make sure there is no mechanical binding going on with your Z axis. This could cause the Z motor to have to work extra hard and overdrive the driver chip.I've modified the firmware to bypass the driver error checking to see if that was an issue and although I no longer get the TMC Connection Error message, my X driver still doesn't work which tells me it really is a hardware problem with the driver chip/circuit.Good luck with whatever you do to solve the problem.-Wes-----Original Message-----
From: WScottCross 
Sent: May 5, 2020 12:26 PM
To: bigtreetech/BIGTREETECH-SKR-mini-E3 
Cc: jeeperjang , Comment 
Subject: Re: [bigtreetech/BIGTREETECH-SKR-mini-E3] TMC Connection Error (#218)

I now have 2 SKR Mini E3 boards that ran for a few days and now display TMC Connection Error.  They were in the same machine and will home X and Y but not Z axis.  Not sure what's going on.  I bought a 3rd board intending to put it in another machine, but now I don't know if I trust these boards any more.  I may try swapping out the Z stepper and the board to see if the motor is killing the drivers.  Am I looking in the wrong direction?

WScottCross, 2020-05-08:
Thanks Wes! I am also considering the SKR E3 DIP.  I know the Z axis is not binding. If I disable the steppers, the whole gantry slowly drops on it's own so it is super smooth.  I'm also suspecting overheating. but it's a bit curious that it's the z axis, unless maybe it's actually due to the z axxis holding the gantry and not moving much. Bummer that we have to spend even more to get boards with replacable stepper drivers. Hopefully that will be the answer.

SlothCroissant, 2020-05-11:
Add me to the list (tagging #155 as there's a long list there as well) - Printer worked for weeks and hundreds of prints, then magically failed today with `TMC CONNECTION ERROR`, and M122 shows "Recv: Testing X connection... Error: All LOW"

Asking BTT to replace the board, but I'm also going to go ahead and move away from the Mini-E3 due to the repeated reports of issues that people seem to have with this. It feels as though it's not "if" but "when" the replacement will fail. 

Could not recommend the board based on my and others' feedback.

jeeperjang, 2020-05-13:
I finally got a response from BigTreeTech. They basically told me they're up to their eyeballs in emails and it might take a while to get a real response to my inquiry. Not a lot of help but I guess it's better than nothing.

rize3boyut, 2021-01-24:
You don't need to buy new cards or drivers. Remove the diag pins on all drives. Included in E drive! You don't need to cut them for this. Heat it with a soldering iron. You can use it later, if you wish, by soldering it. Set the jumper arrangement of all drives to work in uart mode. I haven't tried Marlin with bugfix 2.0 yet, but the problem has disappeared with the latest release 2.0.7.2. Get M122 report. All drivers should be ok.

murdoow, 2021-01-29:
> You don't need to buy new cards or drivers. Remove the diag pins on all drives. Included in E drive! You don't need to cut them for this. Heat it with a soldering iron. You can use it later, if you [...]

Can you explain this fix in a bit more detail?  Not sure which diag pins you're referring to.

rize3boyut, 2021-01-30:
https://user-images.githubusercontent.com/12804127/68514625-5225a880-028f-11ea-91bd-9af971f5d11d.png

Don't forget to take your machine out of usb mode. Operate with power supply 12 or 24 v.

If it still fails, adjust your jumper settings to uart mode.
https://ae01.alicdn.com/kf/H832a0a839acc4e6c9162372bf47e0850p.jpg

Shand359, 2021-02-26:
Please forgive me for jumping on this thread....
I am receiving the same TMC CONNECTION ERROR message... I have checked the jumpers etc and they are in the correct place. 

One thing i notice from above is your comment regarding take the machine out of USB mode and into 12/24V power...... will remaining on the USB mode cause this TMC issue?

Endracion, 2022-04-08:
Reporting that my X Stepper Driver on the 1.2 also seems dead, Forced to swap back to original creality 1.1.4 board, ew.
Consensus seems to be that the X Stepper is prone to failure on SKR Mini E3, right?