bigtreetech/BIGTREETECH-SKR-mini-E3 #220: SKR Mini E3 V1.2 Firmware won't compile, even if I make no changes to it.

Koyodee, 2020-04-16:
Here is the [error message](https://pastebin.com/3fdb4cGG) I'm getting in Visual Studio (With Platformio extension)

GhostlyCrowd, 2020-04-16:
"Please install Git client from https://git-scm.com/downloads"

Koyodee, 2020-04-17:
I have installed the program, but it's [still not compiling](https://pastebin.com/Tt7B0crn). :(

elhits, 2020-04-18:
you have to use STM32F103RC_btt_512K_USB

Koyodee, 2020-04-19:
> you have to use STM32F103RC_btt_512K_USB

"Error: Unknown environment names". I've even tried the suggested names such as "STM32F103RC_bigtree_512K_NOUSB" and "STM32F103RC_bigtree_512K", but neither of them work either.

Koyodee, 2020-04-19:
Here is the [error message](https://pastebin.com/PBYEqC4H)

elhits, 2020-04-19:
I have downloaded the new firmware (2.0.5.3) and in platfotm.io the name is STM32F103RC**_btt_**512K_USB 
I have download to my C root directory because the default path was too large... may it helps you

Koyodee, 2020-04-19:
> I have downloaded the new firmware (2.0.5.3) and in platfotm.io the name is STM32F103RC**_btt_**512K_USB
> I have download to my C root directory because the default path was too large... may it helps you

I did actually get it to compile using the latest version of marlin, but it made my printer a lot noiser (which I think is due to a bug in the latest version that makes the extruder consistently go above the stealthchop threshold). That's why I was hoping to get the BTT firmware working, as it's much quieter, and then I can change what I'd like to have enabled form there.

HVJ17, 2020-04-21:
This is not a bug. If you are compiling your own firmware, disable hybridthreshold.

//#define HYBRID_THRESHOLD

Koyodee, 2020-04-21:
You are a lifesaver! Thank you so much.

closed 2020-04-21 by Koyodee