bigtreetech/BIGTREETECH-SKR-mini-E3 #221: [BLtouch not deploying when Auto Homing]  (Latest Marlin 2.0.5.3 firmware Bltouch for z homing)

antbarbato, 2020-04-18:
### Description

### Steps to reproduce

1. New firmware installed
2. click motion
3. click auto home
4. printer homes x and y
5. printer moves nozzle to center of bed and begins descend.
6. bltouch does not deploy and nozzle crashes into bed

**Expected behavior**
1. click motion
2. click auto home
3. printer homes x and y
4. printer moves nozzle to center of bed and begins descend.
5. bltouch is deployed
6. bltouch probe hits bed and homes the z axis

**Actual behavior**
1. click motion
2. click auto home
3. printer homes x and y
4. printer moves nozzle to center of bed and begins descend.
5. bltouch does not deploy and nozzle crashes into bed

### Additional Information
Ender 3 with genuine bltouch. SKR mini e3 v 1.2 board. Bltouch is wired as z-end stop. Bltouch functions as it should when it is manually controlled. i.e. "self test", "stow", "deploy" functions through LCD. There were not issues with auto home when the board had the previous marlin installed (downloaded from this page).

poysh, 2020-04-18:
Same here, with a brand new SKR mini e3 v1.2 and and original BLTouch 3.

rolge, 2020-04-18:
Same here, with a SKR mini e3 v1.2 and and original BLTouch 3.1

elhits, 2020-04-18:
Same here, with a SKR mini e3 v1.2 and and original BLTouch 3

arduibag, 2020-04-18:
Same here ! Very dangerous !!! :(((

FJJWFP, 2020-04-18:
Might be a stupid question, but have you guys enabled Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN?

poysh, 2020-04-18:
It was using the precompiled firmware, using the firmware-bltouch-for-z-homing.bin directly with a sd-card.

FJJWFP, 2020-04-18:
Oh well, guess it's best to wait for more people to pick up on this. in the mean time you could try and compile your own firmware to see if that fixes it.

elhits, 2020-04-18:
> Oh well, guess it's best to wait for more people to pick up on this. in the mean time you could try and compile your own firmware to see if that fixes it.

It fixes it. But i'm noob with marlin and i am having problems with other configurations. But in precompiled firmware-bltouch-for-z-homing.bin there is disabled Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN because i enable it and compile and works. But i need to know how many other configurations are changed to test them

FJJWFP, 2020-04-18:
> It fixes it. But i'm noob with marlin and i am having problems with other configurations. But in precompiled firmware-bltouch-for-z-homing.bin there is disabled Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN [...]

Did you follow the guide for compiling with BL touch? I assume you have because otherwise it will not work at all, but if you did follow the instructions then there should not be any difference between your current firmware and the precompiled one.

elhits, 2020-04-18:
> > > Oh well, guess it's best to wait for more people to pick up on this. in the mean time you could try and compile your own firmware to see if that fixes it.
> > 
> > 
> > It fixes it. But i'm noob with marlin and i am having problems with other configurations. But in precompiled firmware-bltouch-for-z-homing.bin there is disabled Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN because i enable it and compile and works. But i need to know how many other configurations are changed to test them
> 
> I am glad to hear it worked. I thought there might be some errors with precompiled codes. Did you follow the guide for compiling with BL touch? I assume you have because otherwise it will not work at all, but if you did follow the instructions then there should not be any difference between your current firmware and the precompiled one.

in the guide doesn't appear which of auto bed leveling is used by bltouch, can you help me?
//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
//#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING

FJJWFP, 2020-04-18:
> in the guide doesn't appear which of auto bed leveling is used by bltouch, can you help me?
> //#define AUTO_BED_LEVELING_3POINT
> //#define AUTO_BED_LEVELING_LINEAR
> //#define AUTO_BED_LEVELING_BILINEAR [...]

May I ask if your bed is flat or warped? depending on which it can impact the speed of your bed leveling. 3 point only probes 3 point, which is fastest but is only suited to a very flat bed. Linear is the same, however uses the number of points you have specified in the "GRID_MAX_POINTS_X " definition, which the guide said to do 5. If your bed is warped, use Bilinear, which creates a mesh of however many points specified in "GRID_MAX_POINTS_X " in each axis, so 5 will result in 25 measurements. UBL is very complicated and shouldn't be used in this case. Mesh bed leveling is used for manual leveling only, so isn't necessary.

elhits, 2020-04-18:
> May I ask if your bed is flat or warped? depending on which it can impact the speed of your bed leveling. 3 point only probes 3 point, which is fastest but is only suited to a very flat bed. Linear [...]

Thank you!.
I have a creality glass but i have see in some vids a mesh of 3x3. I will test BILINEAR tomorrow

justintanbs, 2020-04-19:
I had re-compile it with Auto Bed Leveling to Bilinear, and activated BL Touch. everything seems fine. but when it heat up the bed for 105 degree, it halted on 95 degree. and if heat up to 100 degree and then it halted on 90 degrees.

archerbobby, 2020-04-24:
> ### Description
> ### Steps to reproduce
>
>     1. New firmware installed
>
>     2. click motion
>
>     3. click auto home
>
>     4. printer homes x and y
>
>     5. printer moves nozzle to center of bed and [...]

I have this exact problem as well after I tried to install a bl touch on my skr board today.  I am completely new to the programming side and have been trying all day to figure out how to build a code from an existing with the fix but with no luck.  I admit I am good at finding a bin and pasting it, but not at trying to manipulate it.  Any help or a fix for this would be great.

Ender 3 with genuine bltouch. SKR mini e3 v 1.2 board. Bltouch is wired into the prob pins and not the z stop pins.

justintanbs, 2020-04-24:
Yes... it is doing that for auto home. When i start to print then the Y
will start not moving.

HeyRay2, 2020-04-25:
Another "fix" for this is to open the "pins" file for your board and set the "Z_STOP_PIN" to same value as shown for the "Z_MIN_PROBE_PIN"

`#else
  #ifndef Z_STOP_PIN
    #define Z_STOP_PIN                     P0_10  // Same as Z_MIN_PROBE_PIN - Fix Probe not deploying on Z Home (G28
  #endif
#endif

//
// Z Probe (when not Z_MIN_PIN)
//
#ifndef Z_MIN_PROBE_PIN
  #define Z_MIN_PROBE_PIN                  P0_10
#endif`

https://github.com/MarlinFirmware/Marlin/issues/14328#issuecomment-504499973

crashdoc, 2020-05-06:
Just fitted a probe to my Geeetech i3 Pro B - GT2560 A+ running Marlin 2.0.5.3 and getting exactly the same issue. Probe just crashing into the board without deploying. All the manual tests work OK to deploy / retract, its just on auto home/level when it fails.

The above suggestion doesn't fix it for me

cody82, 2020-05-17:
Same for me with Marlin 2.0.5.3 on Ender 5 Plus and SKR 1.4 Turbo.
Auto home works for X and Y, then the BLTouch is not deployed when homing Z.
The BLTouch works when testing it with the LCD.

cody82, 2020-05-17:
Looking into the code this could be a bug in Marlin:
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/inc/Conditionals_LCD.h#L583

> #if DISABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
>    #define HAS_CUSTOM_PROBE_PIN 1
>  #endif
>  #if Z_HOME_DIR < 0 && !HAS_CUSTOM_PROBE_PIN
>    #define HOMING_Z_WITH_PROBE 1
>  #endif

So if Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN is disabled, you can not home with the BLTouch because it will not be deployed.

NeoMod, 2020-06-29:
I was losing my mind over this issue for about a week now when I finally came across this issue and the last comment by cody82.

I had the same problem with Marlin 2.0.5.3 with MKS SGEN L. Once I edited the Conditionals_LCD file removing the ` && !HAS_CUSTOM_PROBE_PIN` string finally the BLTouch is deploying when homing.

Since everything else is properly configured in my Configuration, Configuration_adv, and PINS file I too would guess that this is a BUG.

R3D8U1L, 2020-07-03:
NeoMod, what exactly did you remove?

sophof, 2020-07-04:
> I was losing my mind over this issue for about a week now when I finally came across this issue and the last comment by cody82.
> 
> Once I edited the Conditionals_LCD file removing the ` && !HAS_CUSTOM_PROBE_PIN` string finally the BLTouch is deploying when homing.
> 
> Since everything else is properly configured in my Configuration, Configuration_adv, and PINS file I too would guess that this is a Marlin BUG.

I think we're missing something here, there must be a reason that statement is there? In general I find it a bit weird that replacing the z-stop is the 'standard' way of doing this.

cody82, 2020-07-05:
Although I dont think this code makes much sense, I am also not sure what would be the right way to fix this.
So I did not create an issue on the Marlin repo.
Maybe it is a feature that you have to connect the BLTouch to the Z-endstop pin if you only have a BLTouch and no Z-endstop?

NeoMod, 2020-07-05:
> NeoMod, what exactly did you remove?

I only removed this portion of the string (from the file mentioned): `&& !HAS_CUSTOM_PROBE_PIN`

The string was `if Z_HOME_DIR < 0 && !HAS_CUSTOM_PROBE_PIN` and now is just "if Z_HOME_DIR < 0". 

> I think we're missing something here, there must be a reason that statement is there? In general I find it a bit weird that replacing the z-stop is the 'standard' way of doing this.

Indeed. I think the whole problem arises when one doesn't follow the "standard rule" to sacrifice the z_min_endstop and instead use the z_max_endstop for the bltouch.

Musicdad12, 2020-07-08:
I have the same problem. But I use the SKR 1.4 turbo which has its own bltouch pins NOT the Z_USES_MIN_ENSTOP_PIN. On the SKR 1.4 boards, the bltouch does not use the min endstop pin. Marlin should allow for this.

NeoMod, 2020-07-08:
> the bltouch does not use the min endstop pin. Marlin should allow for this.

Actually **Marlin does allow for using any other pin for the probe**. In the latest _Marlin 2.0.x bugfix_ release, the `Configuration.h `file allows for commenting out the `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` (_if your probe is not connected to the aforementioned endstop)_. 
Then you can de-comment the `#define Z_MIN_PROBE_PIN` and specify the PIN you are using on your board. 
Also, it may be a good idea to de-comment the `#define USE_PROBE_FOR_Z_HOMING` option in order to force the use of the problem for Z-Axis Homing, just in case.

kishkaru, 2020-07-09:
Looks like this issue was [recently](https://github.com/MarlinFirmware/Marlin/pull/17359) fixed:
```
  #if Z_HOME_DIR < 0 && (!HAS_CUSTOM_PROBE_PIN || **ENABLED(USE_PROBE_FOR_Z_HOMING))**
      #define HOMING_Z_WITH_PROBE 1
  #endif
```

Using `#define USE_PROBE_FOR_Z_HOMING` fixed the issue for me. Thanks!

rccypher, 2020-09-11:
Now that this is fixed. Is there any chance we can get a firmware_bltouch.bin file that is updated with the fix? Not to be a pain here but at this moment I don't have time to dig in and figure out how to properly compile for this board.