bigtreetech/BIGTREETECH-SKR-mini-E3 #222: Today i started not having part fan control on my mini e3

MarceloFS13, 2020-04-18:
I dont know what happened but yesterday the board was working fine with control over part cooling fan but today it would not work anymore, the fans are at full speed all time. 
Any ideia what might happened?

ciidfrance, 2020-05-11:
Same issue here !

vajonam, 2020-05-16:
make sure your part cooling fan is attached to the FAN0 port. FAN1 is not controllable is better for the case fan. This is bit different from the E3 board layout.

CRTified, 2020-05-23:
This might be the same issue as #238 - it might be a broken MOSFET, but it is somewhat hard to tell. I made a comment there on how to replace the MOSFET, if you want to try it yourself.

lilasiian8, 2020-07-12:
my HO MOSFET died last night and took the parts cooling fan along with it.

CRTified, 2020-07-13:
@lilasiian8 That sounds unlikely, except when you use a fan that is not rated for the supply voltage (which might work fine if you do not run it at 100% speed, but will easily fail if you exceed the tolerances too much). If that's not the case, it is probably a bad coincidence (or the other way round, so that the fan failed and killed the MOSFET).

lilasiian8, 2020-07-13:
> @lilasiian8 That sounds unlikely, except when you use a fan that is not rated for the supply voltage (which might work fine if you do not run it at 100% speed, but will easily fail if you exceed the [...]

I was trying to install a new controller fan; 12V with a buck converter. Parts fans are (2) 24V in parallel. So, maybe the controller fan wasn’t working and the board overheated causing the MOSFET to burn out?