bigtreetech/BIGTREETECH-SKR-mini-E3 #224: Blank blue LCD screen on stock LCD

lassespilling, 2020-04-19:
I have two boards that were working but now show blue blank screen on stock Ender 3 LCD screen.
Board one:
Went blank after flashing new firmware. (using firmware.bin from Github), this board seems to flash and renames to firmware.cur succesfully.

Board two:
Went blank after just turning power on and off without even flashing or doing anything.
Tried to flash after LCD went blank (using firmware.bin from Github), but this board won't rename to firmware.cur so i do not think it is flashing at all.

### Steps to reproduce

1. Downloaded Firmware.bin from BIGTREETECH-SKR-mini-E3 github
2. Put firmware.bin on card
3. put card into skr mini e3
4. turn off and on power

**Expected behavior**
Board powers on and LCD should work

**Actual behavior**
Board powers on, LCD shows blank blue.

I now have two boards i cannot use.

ylexot, 2020-05-01:
I also have this problem.

seth995, 2020-05-13:
Check the board.  This may not be related to the firmware but rather a hardware failure.  I had a similar  issue with using stock firmware, but found components in the cluster around Y1( between the reset button and U2) to be burned.

BTT claims I let voltage leak between a thermistor and the hot end or bed heating circuits.  I did no such thing.

jhstatewide, 2020-05-14:
Same problem here. SKR Mini E3 -- used for a few months with no problems. Was preheating today and suddenly LCD froze. After power cycle stock LCD is blank. Board will not accept new firmware. Power light is on -- no other responses. I checked connections and did not discover anything odd.

I believe the board is dead.

avsmithy, 2020-05-14:
I have identical issue to @jhstatewide, except this happened immediately on first print.

Returning to seller.

Shizzy713, 2020-05-18:
I have the same problem as the OP.  I installed the board yesterday and flashed the firmware.bin from github on my Ender 3 Pro today and my board is now bricked?  FML

sultanjawad, 2020-05-24:
So what exactly are we supposed to do with a bricked board?

nmg196, 2020-05-26:
Mine did this but it wasn't bricked. I just reflashed the original firmware from the SD card which came with the BTT board and it came back to life.  Let me know if you need the file, but you should already have it on the 128MB micro SD.

nmg196, 2020-05-26:
> Same problem here. SKR Mini E3 -- used for a few months with no problems. Was preheating today and suddenly LCD froze. After power cycle stock LCD is blank. Board will not accept new firmware. Power [...]

That is not the same problem at all. This thread is a problem about no display after flashing the new firmware, not a failure of the board when you've not touched the firmware.

lassespilling, 2020-05-26:
> That is not the same problem at all. This thread is a problem about no display after flashing the new firmware, not a failure of the board when you've not touched the firmware.

If you read my post, this happend on my second board without touching the firmware.

jhstatewide, 2020-05-27:
> So what exactly are we supposed to do with a bricked board?

Toss it in the trash and spend another $40.

nmg196, 2020-05-28:
Is it definitley bricked?  Mine booted up fine once I'd reflashed the original firmware.

avsmithy, 2020-05-28:
Mine was most definitely either a) bricked or b) broken. It also killed two SD cards, so I pretty sure it's the latter.

jhstatewide, 2020-05-30:
> Is it definitley bricked? Mine booted up fine once I'd reflashed the original firmware.

Yes, it will not accept any firmware. Confirmed by keeping my SD card with firmware on it, buying another SKR board and using same card to reflash. The original board is definitely bricked.

arnoldcp, 2020-06-06:
I am having exactly the same problem, brand new board, straight out of the box hooked it up and upon boot, simply a blue screen.

Attempted to update the firmware thinking it maybe required an update and upon boot the activity light does flash but still no LCD. Strange.

federico77, 2020-06-10:
I have a firmware I compiled that is based on Marlin 2.0.x. It ran fine with the board for a few months. Yesterday night I turned the printer off after a quick session but as soon as I turned it back on today I experienced the symptoms described in this thread: printer apparently on, blank screen, no serial connection available (for Octoprint and such).

I shuffled a few firmware versions and SD cards but the printer doesn't even start using the new firmware (.bin->.cur). Should I simply ditch the board for a new one?

greenbox6, 2020-06-11:
ok update here: after taking out the board I examined the sd card slot and turns out one of the pins has been bent back. so now I have to try and bent it back to normal from a tiny enclosure. I've been trying to use a pin but it's frustratingly hard. I wish I could just use the usb slot instead.

ErickFourie, 2020-06-13:
Day 2 that my board is in and boom blank screen. Doesn't want change the .bin to a .CUR and absolutely no LED's working on the board. However, heat shrink fan goes on and the electronics fan goes on.

lassespilling, 2020-06-15:
> Check the board.  This may not be related to the firmware but rather a hardware failure.  I had a similar  issue with using stock firmware, but found components in the cluster around Y1( between the [...]

No visible hardware damage, already sent images to confirm with BTT

lassespilling, 2020-06-15:
> Mine did this exactly this but is was **not** bricked - it simply had no display (all pixels black). I just reflashed the original firmware from the SD card which came with the BTT board and it came back to life.  Let me know if you need the file, but you should already have it on the 128MB micro SD.

Yeah please share the file, and I will test myself.

ErickFourie, 2020-06-15:
I have tried everything now on the board and there's absolutely no body on the inside. Plugged the board in and even measured a few points and there is definitely components toasted.

rvegele, 2020-06-16:
Used my board since Dec 2019. Today after a print the screen just froze. After a power-cycle of the printer the screen is blue-blank.

rvegele, 2020-06-16:
Tried replacing the skr mini board with my original ender3 board. And it worked just fine, so not a screen or cable issue I guess. Something's up on the skr board side :/ Any ways to debug this?

zandervix, 2020-07-17:
Replaced Creality 1.1.3 board with the SKR (and added a BLTouch). Worked beautifully for about 4 days (printing round-the-clock). Then, going into the next print, the home command ran the nozzle straight into the bed. Turn off and back on, and the screen is on but no boot-up. Tried re-flashing (multiple firmwares) with no luck. 

Replaced the board, and it seemed to be DOA (I also tried replacing the screen, ribbon cable, disconnecting the BL, firmwares, new PSU, voltage checks, etc...). Sent that board back and bought another from a different seller. While waiting, put the original 1.1.3 board in, and it worked. 

New board (3rd SKR) came today. There is nothing I can do to get this to work. 

Giving up and going to try the Creality 1.1.5 silent board. Really wanted this to work, but after 3 boards and weeks of downtime, I'm done.

winstonf, 2020-07-17:
I have a Bricked board too. Mine worked well for 3 odd months. But i know the reason it stopped. While trying to clean up the hotend from a filament leak, I accidentally shorted some wires I guess. My mistake :-) . I realized that the STM32 was getting hot just with USB power. I managed to replace the STM32 mcu and I am able to Flash Marlin using a STLINK programmer. But the LCD still does not come up. Just by looking I don't see damaged components. Any Clues as to what more I could check will be helpful.

DarthAlaan, 2020-07-29:
I encountered this same problem on both my Ender 3s in the past week after months of no issues with the v1.2. In both cases I was simply preheating and swapping filament and when I looked back I see the screen is froze. I try a power cycle and the screen goes blank. Then it wont accept a flash of the original firmware or a new firmware from sd card as the .bin is never renamed to .cur. Anyone find a good fix yet? Is everyone just chucking these boards and ordering new? If so what, the new Mini E3 V2?

kozip, 2020-08-29:
I just installed a new SKR mini E3 V2 into my ender 3 pro. only blank screen. I put firmware.bin from BTT git on the sd card. On power up you can see the red LED flashing, still just blue screen. when examining the sd card firmware.bin is gone and FIRMWARE.CUR is there. I've tried both bltouch and stock firmware from BTT GIT with no luck....

denisao1998, 2020-09-01:
Hello. I have the same problem: the printer worked with original microSD (original firmware.bin). I put the latest 2.0 firmware.bin and the printer is stucked on blue screen. Can anyone share the original firmware.bin. Thank you

LearnIIBurn, 2020-09-04:
Exact same issue. On my second v2.0 board and right as I power it on its a blank screen (all pixels black) and will not flash firmware of any SD card I try. Bought board brand new off Amazon and it did not include a 128MB SD card anywhere in the package which is odd. But none the less this is a very big problem and I'm about to give Bigtreetech the permanent 86 from all my printers.

Hunner63, 2020-09-04:
I was unable to get beyond the blank blue screen after upgrading the
firmware.  The board appeared to work fine before the upgrade.  I
eventually re-installed the original 8-bit board and put the new board to
the side until this problem gets sorted out

denisao1998, 2020-09-04:
Hello. I found 2 solutions to bring to life the SKR mini S3 v2 after firmware upgrade (blank blue screen problem):
1. i downloaded the latest marlin open source project, i modified it for SKR mini S3 v2 (there are guides about this), i compiled it and i used the resulting firmware.bin
2. i asked bigtreetech support to send me the original microSD firmware.bin

Both solutions are succesufully. In first solution i enabled 3D Touch sensor and it is working.

DTSParsley, 2020-09-10:
I have the same problem with my ender 3 pro skr v2.0 and blue screen. I created a firmware for the ender 3. When creating the firmware for the ender 3 pro I get errors when compiling.
But I managed to get the ender 3 pro to work with the firmware for ender 3 without pro. Has one of the original firmware that was included with the SD card? I'm sorry for my english. Google translator
[firmware ender3.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5200144/firmware.ender3.zip)

tthamel, 2020-09-10:
Can we stop saying the boards are bricked?

i had the same thing happen, tried to flash the BLTouch version of the firmware.  bam, blue screen.

went back and got the original BTT firmware it shipped with and it booted up no problem.. ergo the issue is the BT firmware

my point is, has everyone that is claiming they have a bricked board tried re flashing the original firmware?

LearnIIBurn, 2020-09-11:
@tthamel I don't know if everyone is having the same symptoms resulting in just a blue screen with nothing happening, but I know for me that firmware is not flashable at all from the SD card slot. Doesn't matter if it's official from their github, official from Marlin's github, compiled from bugfix branch or the original firmware that came with the board on the 128MB SD card; it just doesn't flash anything. I'd say that calling it a bricked board is quite alright.

kozip, 2020-09-11:
> Can we stop saying the boards are bricked?
>
> i had the same thing happen, tried to flash the BLTouch version of the firmware.  bam, blue screen.
>
> went back and got the original BTT firmware it [...]

I contacted support and theyr sending a new board. I'm not sure why people aren't first contacting BTT through their support email.

SonnyJim, 2020-09-12:
Just had this happen to me.  Was printing something, went to change the filament using the menu option, when I inserted the new filament and pressed OK it went back to the main status screen, but it had frozen.  Dial wouldn't do anything, no beeps.  So I power cycled the printer.  Now it just sits on a blue screen and tried various different firmwares, nothing loads.  No firmware.cur, no lights on the mainboard apart from a single red LED.  No damage to the board as far as I can see, it just decided to stop working.

getz97, 2020-09-18:
> Mine did this exactly this but is was **not** bricked - it simply had no display (all pixels black). I just reflashed the original firmware from the SD card which came with the BTT board and it came [...]

pleace, send me original .bin file.

kabone, 2020-09-18:
Same. Ive been printing good for months. Then i turned off and returned on the printer and it just shows black pixels. If i remove tft pins and move to exp1 then it shows pixels. Im at a loss. I just ordered another...

Hunner63, 2020-09-18:
Me too, please. Thanks!

blhamon, 2020-09-19:
Ok, I just had this problem, tried tried to flash the firmware, specifically the firmware_bltouch.bin.    turned the printer off, inserted SD card, firmware.bin renamed to firmware.cur (finally pulled the card out after waiting 10 minutes).    Blue screen, completely blank.

Got a new SD card, tried again, no good, still blank.   This time, didn't rename the file.

copied the firmware.bin that came on the SD card.  Inserted, nothing.   Didn't rename the file.    when I looked closer, I noticed that my file was named, FIRMWARE.bin.    All caps name, lowercase extension.   Made extension all caps, inserted.    Flashed back to original firmware that came with the board.

So, that solved my restore problem.   making sure the file was FIRMWARE.BIN.

Tried the process again, on the new SD card.   Same thing, blank screen.  Then I restored back to the original.

Then I noticed that the firmware.bin and firmware_bltouch.bin that I downloaded from github, via right-click, save as was only 82KB.   While the firmware that came with the board was 209KB.   This didn't seem right, so I went back to github and instead of downloading via right click, I clicked on the firmware_blthouch.bin.   This loaded a page that said the file ws 226KB, with a download button.   Clicked download.   Verified that the file was indeed 226KB when downloaded.    Renamed it to FIRMWARE.BIN, inserted into SD Slot, turned on printer.   15 seconds later, the screen loads and says I'm on the newer version of Marlin (2.0.6, I think).

So, I am a bit of a dumbass, but in case it might help others, make sure your downloaded file is 200+KB in size, and not 82KB.    And renaming to all caps fixed my restore, so I didn't even try the new firmware.bin file lowercase, I just made it all uppercase.

blhamon, 2020-09-19:
> pleace, send me original .bin file.

Here's the zipped original FIRMWARE.BIN from my original card.

[FIRMWARE.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5249161/FIRMWARE.zip)

getz97, 2020-09-19:
Thank you, but I have no success ;(  , still blue screen.

getz97, 2020-09-19:
i conect my old screen from ender 3 and it boot, it seems the ""TFT 35 E3 V3.0"" is a problem?!? 
How to reset this display?
i try with ""RESET"" button on it-no result.

getz97, 2020-09-19:
I find solution ;)
1. install some old firmware for TFT35 from bigtreetech gihub.
2. create a blank file named reset.txt and place in root folder of the sd card.
3. power on, it start screen calibration.. some red dot have to touch... 
4. calibration fail to me, i remove sd card and screen boot up
Good Luck to everyone ;)

kabone, 2020-09-19:
> **I find solution** ;)
> 
> 1. install some old firmware for TFT35 from bigtreetech github.
> 2. create a blank file named **reset.txt** and place in root folder of the sd card. put SD card to TFT sd slot.
> 3. power on, it start screen calibration.. some red dot have to touch...
> 4. calibration fail to me, i remove sd card and screen boot up
>    Good Luck to everyone ;)

Ill have to try this. I did very close to this before and it did not work. But i was having issues with the firmware update and nissing bmp files. So now that thats fixed it may work. Will report back later tonight.

getz97, 2020-09-20:
last firmware update show this ""missing bmp files"", that why i use some old file, and it work.

h3soy2m, 2020-09-20:
> last firmware update show this ""missing bmp files"", that why i use some old file, and it work.

Can you share the old file please.

getz97, 2020-09-20:
this on :11 sept.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/cb2430ca5b8bf1a50b0058f41b0c0c98fb114ff5

Anybid, 2020-10-11:
> Ok, I just had this problem, tried tried to flash the firmware, specifically the firmware_bltouch.bin. turned the printer off, inserted SD card, firmware.bin renamed to firmware.cur (finally pulled [...]

Thank you so much, blhamon!!!! Your are no dumpass but a genius, this was exactly my problem too, and you showed me the solution.

Ansari-Tech, 2020-10-20:
I'm also having this issue.

I have nothing connected to the board except power and LCD. 

When I upload the firmware.bin (confirmed to be 220kb) to the SD and boot the board up, the status LED flickers for a few seconds indicating that it's flashing, and then goes off. the LCD remains blank.

My FIRMWARE.bin does get changed to FIRMWARE.cur, so I know that it's flashing, but the LCD remains blank and both octopi/pc continue to not see the board whatsoever, as though it's not plugged in at all.

It was working fine for several months, I didn't change anything. I powered up the printer after not using it for a few months and this is the issue I'm having.

JoeDaddy7105, 2020-11-15:
> Then I noticed that the firmware.bin and firmware_bltouch.bin that I downloaded from github, via right-click, save as was only 82KB. While the firmware that came with the board was 209KB. This [...]

You are my hero! I was doing the exact same thing with downloading the bin files with right click save as from Github and getting ~90KB files and thought I had bricked my SKR mini E3 V2.0 since the screen was stuck on the blank screen no matter what I did and I had wiped the original SD card so I didn't have the old firmware to fall back on. Then I saw this and downloaded from Github by clicking the file and using all caps filename and it worked like a charm with the 2.0.7.2 version.

Gertsch, 2020-11-16:
I've got the exact same problem right now.
Printed flawlessly for 6 months and today tried to calibrate again - just for good measure.
After some initial test octo print lost the connection and the lcd turned solid blue.

Tried everything mentioned in here, but nothing worked for me unfortunately.

Are there any new insights?

Thanks!

SonnyJim, 2020-11-16:
What I did was contacted BTT support, who made me run through quite a few
troubleshooting steps.  Once it was clear the board was dead they sent me a
new one.  The board that died was about 7 months old iirc.

Gertsch, 2020-11-22:
As the BTT support didn't get back to me I purchased a SKR mini E3 V2.0 in the meantime. Worked fine from the beginning and did all the calibration. Did some print and then again the same issue like with the 1.3 board.
LCD went black and octoprint lost connection out of blue. Wasn't even printing at that moment.

What could cause this problem? I'm a little stuck right now.

tchilton, 2020-12-14:
**I have the solution to this problem from reading between the lines on this page.**

The key is the file size. 

If you go to the firmware.bin on the github page and right-click, save, you save a file of around 95K, this is simply the HTML of the next page, hence why it does not work as this is not firmware !!

To correctly get the firmware file, you must click on firmware.bin, which takes you to the download page, then click on "View raw file" and it will save the firmware.bin to your machine, this file will be around 224K.

You can also check the files by opening them in notepad. The firmware file will appear as a binary file (unreadable), whereas the 95K one will appear as simple HTML. You will see a line <!DOCTYPE HTML> in the start of the file. Just make sure you don't save the file if you open it in notepad as it would corrupt it.

You can also check the card is programming by looking at the two LED's. The status LED will flash when its programming the firmware.

Correctly downloading and programming resulted in a working machine again.

Gertsch, 2020-12-17:
I found the cause of my problem at least, and have heard form others with the same experience.
Two of my SKR mini E3 V2.0 were bricked by an electrostatic discharge of myself. I just found out because I heard a clicking sound doing some calibration prints and wanted to remove filament residue - this board then had the exact same issue like the ones before. 

So if anyone has the same issue with the board as I had (blank screen, no serial connection/octoprint, can't read MC with STLink), it is quite possibly came from an ESD and bricked the MCU/electronics.
So be sure to
- ground yourself before doing something on the printer, regardless where
- use an esd-bracelet
- don't keep the room humidity too low

tchilton, 2020-12-17:
The other thing to watch out for is manually moving the table too fast. When you move the table, you turn the motors into generators (dynomo's) and the resulting voltage (reverse EMF) can be far too large for the logic board and therefore permanently damage it. 

It would be good if SKR had put some clamping networks on the board to protect against this obvious problem, particularly given the need to manually move the bed to perform bed levelling and the risk of the bed moving quickly when turning over the printer to change the logic board.

Don't forget that this only has to happen once to spoil your day, so please be careful !!.

GDRPrint, 2020-12-20:
Just had this happen after 3 weeks of excellent printing. The 8 bit board worked fine before the upgrade. I have the original screen. Have tried uploading the original BTT firmware, no luck. Thank you for all the posts, have ordered a new board from Amazon, will wait for any results from BTT and post.
Ender 3 Pro printed daily, no issues and very good prints with the basic upgrades like glass bed, springs, extruder.

gknapp, 2021-06-11:
What you said here makes sense. I was changing my filament, usually I don't both with the 'change filament' option in Marlin, I just disable steppers, heat the hotend, hold the extruder clip open and pull the filament out myself.

This time I used the menu option to change filament. I did the same procedure, waited for the hotend to heat, pinched the extruder clip and began to pull the filament out. As I did, the extruder stepper motor began to assist. As the filament was coming out of the boden tube I noticed my BTT TFT 3.5 V3 flashed, then a second or so later, went off completely.

From there on, it would not power up. I think as the filament cleared the hotend and became easier to withdraw my assistance has overloaded the board.

I switched out the screen for the original, it just turns on, glows blue but is blank. I've tried placing a firmware.bin on an SD card, the board doesn't appear to red the card as the file is not renamed firmware.cur.

> The other thing to watch out for is manually moving the table too fast. When you move the table, you turn the motors into generators (dynomo's) and the resulting voltage (reverse EMF) can be far too [...]

MetalMinotaur, 2021-08-28:
Hi everyone,

I'll just leave this here, to see if it might help anyone.
I also had the issue of the stock LCD of my Ender 3 not working with two different SKR mini e3 v2 boards.
Turns out this might be an easy fix in the firmware.
After experimenting a bit, I found that not setting the BEEPER_PIN when building Marlin suddenly makes the stock LCD work.

I just changed line 143 in "Marlin\src\pins\stm32f1\pins_BTT_SKR_MINI_E3_common.h" 

from 
`#define BEEPER_PIN                      PB5
`

to
`    #define BEEPER_PIN                      -1 //PB5
`

You'll lose the annoying beeps but you get a working stock display in return. 
Not a bad deal I guess.

Credit for the hint in the right direction goes to this reddit post - seems like Klipper has the same issue with this board: https://www.reddit.com/r/klippers/comments/msgi7z/ender_3_pro_display_problems_on_skr_mini_e3_v20/

Let me know if this worked for you.

deputyd0ng, 2022-01-22:
> Hi everyone,
> 
> I'll just leave this here, to see if it might help anyone. I also had the issue of the stock LCD of my Ender 3 not working with two different SKR mini e3 v2 boards. Turns out this might be an easy fix in the firmware. After experimenting a bit, I found that not setting the BEEPER_PIN when building Marlin suddenly makes the stock LCD work.
> 
> I just changed this part in "Marlin\src\pins\stm32f1\pins_BTT_SKR_MINI_E3_common.h"
> 
> ```c
> #if HAS_WIRED_LCD
> 
>   #if ENABLED(CR10_STOCKDISPLAY)
> 
>     #define BEEPER_PIN                      PB5
>     #define BTN_ENC                       EXP1_9
> ```
> 
> to
> 
> ```c
> #if HAS_WIRED_LCD
> 
>   #if ENABLED(CR10_STOCKDISPLAY)
> 
>     #define BEEPER_PIN                      -1 // PB5
>     #define BTN_ENC                       EXP1_9
> ```
> 
> You'll lose the annoying beeps but you get a working stock display in return. Not a bad deal I guess.
> 
> Credit for the hint in the right direction goes to this reddit post - seems like Klipper has the same issue with this board: https://www.reddit.com/r/klippers/comments/msgi7z/ender_3_pro_display_problems_on_skr_mini_e3_v20/
> 
> Let me know if this worked for you.

I am experiencing this issue, and unfortunately, this did not resolve it.

[2 more replies]