bigtreetech/BIGTREETECH-SKR-mini-E3 #225: BLTouch not leveling correctly

Linkeb3, 2020-04-21:
### Bug Description

My BLtouch won't probe more than 9 spots regardless of config, and doesn't create accurate settings as the right side is carving into bed, and left is too high.

### My Configurations

[](https://gist.github.com/Linkeb3/a497ad90f4adecd53c6031fe999223c0)
[](https://gist.github.com/Linkeb3/faee492c414166d28b47cf8470f1386b)

### Steps to Reproduce

I'm using an SKR 1.2, the BLTouch is plugged into z-stop and servos. I was using a BTT35, but disconnected it as a troubleshooting step and plugged in the old LCD.
I've edited the config and advanced config every way i can think of to get the BLtouch to probe more than 9 points - of which it doesn't apparently create a mesh as one side is carving into glass, the other too high. i

**Expected behavior:** create and use a mesh with more than 9 points

**Actual behavior:** poor leveling at 9 points

GhostlyCrowd, 2020-04-21:
I have a friend with this exact same issue, Many others are reporting it as well. So I've started stepping back commits until its working.

Linkeb3, 2020-04-21:
it's the latest marlin, anyone stepped back to 2.0?

GhostlyCrowd, 2020-04-21:
I've checked out this commit others say it has fixed it for them https://github.com/MarlinFirmware/Marlin/commit/fc74c6a7b6646cf608f96bdf362a136eb306415e

SO ill post back once my friend tests on his SKR Mini E3

Linkeb3, 2020-04-21:
fails to build

GhostlyCrowd, 2020-04-21:
Building fine for me.  Just did a build from my friends E3 Mini and my SKR Pro both suffering weird BLtouch issues.

Linkeb3, 2020-04-21:
got it to build, missed a #.. doesn't fix the 9 point probe.

GhostlyCrowd, 2020-04-21:
So far its fixed issues for both of us, things liek this were happening on the latest pull of marlin 2.0.5

[image]

Linkeb3, 2020-04-21:
replacing 
#ifndef SHORT_BUILD_VERSION
  #define SHORT_BUILD_VERSION "2.0.5.3"
#endif
with
#ifndef STRING_DISTRIBUTION_DATE
  #define STRING_DISTRIBUTION_DATE "2020-03-28"
  #define STRING_DISTRIBUTION_DATE "2020-03-29"
#endif

GhostlyCrowd, 2020-04-21:
Their are more changes then just that, thats just the commit i rolled back to as it was the last known working on for another user in the marlin bug tracker.

GhostlyCrowd, 2020-04-21:
Well it made it further but we got the bed black holes back here again. going to roll further back to another suggested commit ill let you know if it works.

Linkeb3, 2020-04-21:
I've got a budyd that says 5.0.3 works, solves all the issues. but i honestly can't find it - and don't know how to build it myself

GhostlyCrowd, 2020-04-21:
5.0.3?
you mean 2.0.3?

Linkeb3, 2020-04-21:
yes, sorry - brain melting

Linkeb3, 2020-04-21:
if it's more complicated than replacing some lines in versions.h, afraid i'll have to be told what to do

HVJ17, 2020-04-21:
Have you made any changes the probing? I suggest the following in .h

**#define PROBING_FANS_OFF          // Turn fans off when probing
#define DELAY_BEFORE_PROBING 200  // (ms) To prevent vibrations from triggering piezo sensors**
#define MULTIPLE_PROBING 2

Then ober in adv.h
#define BABYSTEPPING
 #define BABYSTEP_MULTIPLICATOR_Z  6
 // i like to use 6 here, higher number is ok too. I like the adjustments a value of 6 allows

Linkeb3, 2020-04-21:
all tried, i reverted to someone elses 2.0.3 marlin and have been going through it customizing it to my printer - it is currently probing 25 spots. now i just have to figure out why the bed heat won't go above 46

GhostlyCrowd, 2020-04-21:
I pulled 2.0.3 compiled it and sent it off to my friend to test we will see.

Linkeb3, 2020-04-21:
i've got 2.0.3 running, it's doing everything as it should as far as the BL touch, just probed 25 points.

However.. now my heated bed wont hit 55.. it's been stuck at 53 for 30 minutes

HVJ17, 2020-04-21:
Which config files are you using for 2.0.3?  The PID settings are different than in 2.0.5.3.

Linkeb3, 2020-04-21:
https://gist.github.com/Linkeb3/12c92de46e27b47f126ac44a27ff3d4e
https://gist.github.com/Linkeb3/b68d1f81e6ed6461cdf55ba0cc0d766d

HVJ17, 2020-04-21:
Those correct even the right config files. You need to start with the BTT files in the config examples.

Linkeb3, 2020-04-21:
I'd love to, they're no longer on github for 2.0.3 - and the config examples in the marlin 2.0.3 downloads are gone as well.

HVJ17, 2020-04-21:
[SKR Mini E3 1.2.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4513047/SKR.Mini.E3.1.2.zip)

HVJ17, 2020-04-21:
The PID values are wrong, going to ruin the board or you house.

Linkeb3, 2020-04-21:
is that 2.0.3 ?

HVJ17, 2020-04-21:
Not 100% sure. But I think it is. its the last update for 2019. Its the official release after bugfix.

I can give you my 2.0.5.3 but i use zstop not servo. So you might to change that. Also I made quite a few other quality of life changes.

Linkeb3, 2020-04-21:
it built no problem, just need to enable the bltouch.. the wordings a bit different in here for using the servo for z-stop, i'm sure i can google that. Thanks a a bunch.

HVJ17, 2020-04-21:
Oh, here is the previous versions, Forgot about this bookmark.
https://github.com/MarlinFirmware/Marlin/releases

Linkeb3, 2020-04-21:
i found that, within that - the config examples are gone.

HVJ17, 2020-04-21:
Wow they are sneaky. I thought it was left in the older zips. Gotta use the branch release button.
**https://github.com/MarlinFirmware/Configurations/tree/release-2.0.3**

Linkeb3, 2020-04-21:
Marlin\src\feature\pause.cpp:355:20: error: 'FILAMENT_UNLOAD_PURGE_FEEDRATE' was not declared in this scope

is the only error in building this

Linkeb3, 2020-04-22:
however octoprint is now giving me the low voltage warning, i'll build one from the link you sent.

Linkeb3, 2020-04-22:
https://www.dropbox.com/s/f2uh0qabt56am4x/garbage.jpg?dl=0

I’d like to toss this whole thing out a window..

HVJ17, 2020-04-22:
No its ok. you need to redo the probe offset when you do fresh fw.

Linkeb3, 2020-04-22:
i did.. the left purge line is perfect. i used the repository config and rebuilt it again - octoprint is still pissy about voltage. it's got a buck converter on it, i'm afraid to adjust it, as it's probably an issue elsewhere.

Linkeb3, 2020-04-22:
I've got Z-Offset set to -2.45, i ended up babystepping an additional -2.5.

There's a noticeable difference between the front and the back.. A few days ago when i initially upgraded the firmware i had a moment of panick when the bit drove into the center and burned a hole - instead of just flipping it off, i pulled up on the extruder and pushed down on the bed. with the outragious -5ish z offset, i think i might need to replace the motors.. i have two extras here..

Jimmywwk, 2020-04-22:
I have the same issue with the new original bltouch Z home firmware, If I print with manual bed leveling, 4 corner is OK but if I do a ABL before printing, the nozzle will wringing to the bed when it goes to the upper left corner.

GhostlyCrowd, 2020-04-22:
Just a update ive rolled all the way back to https://github.com/MarlinFirmware/Marlin/commit/e6a7be922b300bc68030a5d301743ac30d6d1834

And it has fixed the issue, so this is the solution, when i have more time ill have to submit a proper bug  report to marlin proper.

GhostlyCrowd, 2020-04-22:
So I've had to roll all the way back to https://github.com/MarlinFirmware/Marlin/commit/e6a7be922b300bc68030a5d301743ac30d6d1834 to fix BLtouch going rogue on SKR Pro and Mini e3. Randomly not deploying probes driving tool head into the earths core, and "black holes" on bed meshes. Like the attached image. I don't have anything more to offer with debug probe as were just trying to get things moving we are printing PPE for local medical.  There are a few open bug reports now for bltouch issues, ill see if I can add to them when it is possible.

HVJ17, 2020-04-22:
Just do a new build with the newest marlin files and the newest btt config. Dont mix and match , dont mess to commits and rollbacks, wastes of time usually. Really if you have just 1 printer model its easy to setup an new build for each release.

GhostlyCrowd, 2020-04-22:
> Just do a new build with the newest marlin files and the newest btt config. Dont mix and match , dont mess to commits and rollbacks, wastes of time usually. Really if you have just 1 printer model its easy to setup an new build for each release.

Rolling back to track down a bug is not a waste of time. This is exactly how marlin devs ask people to help them trouble shoot. Then step forward a test if its still working step forward, when it stops working walk back.

Linkeb3, 2020-04-22:
i did the newest of the new - that's what caused this 4 days of crappy prints. I work in boston, i'm just trying to print a bunch of ear savers.. i just needed a working solution and rolling back to 2.0.3 worked - it's not perfect, but i can print again.

raidthebarn, 2020-04-27:
Linkeb3 - what config files did you end up using? i cant compile this one: https://github.com/MarlinFirmware/Marlin/commit/e6a7be922b300bc68030a5d301743ac30d6d1834

raidthebarn, 2020-04-27:
> Just a update ive rolled all the way back to [MarlinFirmware/Marlin@e6a7be9](https://github.com/MarlinFirmware/Marlin/commit/e6a7be922b300bc68030a5d301743ac30d6d1834)
> 
> And it has fixed the issue, so this is the solution, when i have more time ill have to submit a proper bug report to marlin proper.

what config files did you end up using? i cant compile this one: MarlinFirmware/Marlin@e6a7be9

so very frustrated, just want to get bltouch working

closed 2020-04-27 by Linkeb3

Linkeb3, 2020-05-02:
Any word on a fix? I didn’t close this, not sure why it says I did.

Masterzior, 2020-05-07:
> So far its fixed issues for both of us, things liek this were happening on the latest pull of marlin 2.0.5
> 
> [image]

Ive had this exact problem for like a month. You are the first person ive seen that has the same problem. I will try reverting back to an old version