bigtreetech/BIGTREETECH-SKR-mini-E3 #226: BL touch probing only once + failure to build

raidthebarn, 2020-04-22:
configuration:
skr mini e3 1.2
Bltouch clone (3d touch)

Trying to get ABL up and running for the first time using: firmware-bltouch.bin
Gcode sends G29, bl touch probes one time in the corner and begins the print.

If I start Auto bed leveling from the printer directly, it probes all 9 points, but doesn't appear to store the data as the prints are still clearly uneven.

I went to compile Marlin-2.0.5.3 to ensure AUTO_BED_LEVELING_BILINEAR is enabled and GRID_MAX_POINTS_X 3 but when compiling I get the following error:
#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (C:\Marlin\Mini-E3-V1.2\Marlin\src\MarlinCore.cpp).

Clearly I am very new at this - would love some guidance

raidthebarn, 2020-04-23:
Can anyone help me with the #include errors detected ussue?

GhostlyCrowd, 2020-04-23:
Run VS Code as administrator or it will have issues, try that first.

raidthebarn, 2020-04-24:
Unfortunately running as admin did not fix my issue...

Can a kind soul please upload a known good version of firmware-bltouch.bin with bilinear leveling enabled (ideally 3x3 but beggars can't be choosers)

GhostlyCrowd, 2020-04-24:
Try shortening the path to your source make C:\Marlin\Mini-E3-V1.2\ >  C:\Marlin\MiniE3\

raidthebarn, 2020-04-26:
Shortening the path worked great, thank you!

closed 2020-04-26 by raidthebarn