bigtreetech/BIGTREETECH-SKR-mini-E3 #228: Publish precompiled firmware-bltouch-for-z-homing.bin

Gcroberson, 2020-04-23:
This firmware seems to be working well for me but there are some settings that I'd like to change. Would it be possible to upload this firmware precompiled?

GhostlyCrowd, 2020-04-23:
what are you asking your using the precompiled but you want them to upload it again?

Gcroberson, 2020-04-23:
Maybe I'm confused but the firmware folders/files are not configured the same way firmware-bltouch-for-z-homing.bin is. I understand that we can make those changes ourselves but I'm having issues during bed leveling with bltouch. Just curious if there is a config.h config_adv.h with the same settings as firmware-bltouch-for-z-homing.bin....

GhostlyCrowd, 2020-04-23:
> Maybe I'm confused but the firmware folders/files are not configured the same way firmware-bltouch-for-z-homing.bin is. I understand that we can make those changes ourselves but I'm having issues [...]

ah now i get you. Yeah they didnt post their bltouch configs. unfortunately. What issues are you having? I had nothing but issues on 2.0.5.3 with bltouch so we rolled back to 2.0.3

11chevelle11, 2020-04-23:
Mid auto bed level it just stops leveling.  It runs perfect on the firmware-bltouch-for-z-homing.bin but I want to make changes to some things on there and I lose my esteps and zoffset after reboot which is extremely frustrating.

GhostlyCrowd, 2020-04-23:
Yes this is because their seems to be a bug with BLtouch and flash eeprom emulation on 2.0.5.3 Marlin I had the same issue so rtolled back to 2.0.3. It would seem that BTT disabled this eeprom emulation in their build which is why BLtouch works but your eeprom settings dont save.

11chevelle11, 2020-04-23:
So 2.0.3 seems to run okay? What version BLtouch do you have?

GhostlyCrowd, 2020-04-23:
> So 2.0.3 seems to run okay? What version BLtouch do you have?

Yes and the ender 3 is running V3.1 bl touch

11chevelle11, 2020-04-23:
Sweet. Would you happen to be able to upload your precompiled firmware files?

GhostlyCrowd, 2020-04-23:
> Sweet. Would you happen to be able to upload your precompiled firmware files?

you mean the config files?

11chevelle11, 2020-04-23:
> > Sweet. Would you happen to be able to upload your precompiled firmware files?
> 
> you mean the config files?

Sure that works.

GhostlyCrowd, 2020-04-23:
[Configs.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4524243/Configs.zip)

11chevelle11, 2020-04-23:
Thanks man

11chevelle11, 2020-04-23:
Something is running extremely hot with that config. Its either the psu or the z motor. I bailed out on it.

GhostlyCrowd, 2020-04-23:
> Something is running extremely hot with that config. Its either the psu or the z motor. I bailed out on it.

I have after market stepper motors, if youre running stock youre gonna want to change the miliamps to 580 from 750 in the TMC https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/228#issuecomment-618672760_SMART section of config_adv

HVJ17, 2020-04-23:
What is your Ender 3 setup? How is the bltouch mounted and connected? Ill give you my 2.0.5.3 configs that work great. You might need to adjust offset. Mine is -44,-6, 0. 
My esteps are 96,
Be sure to do zprobe offset, I have the menu enabled and the z can move below 0. Careful!
Also raise the z high before doing the first auto home to test the probe by hand.

elhits, 2020-04-24:
> [Configs.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4524243/Configs.zip)

I have an error with you r config.
#error "You are using an old Configuration.h file, update it before building Marlin."
Which marlin version is? i am using 2.5.0.3

GhostlyCrowd, 2020-04-24:
> > [Configs.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4524243/Configs.zip)
> 
> I have an error with you r config.
> #error "You are using an old Configuration.h file, update it before building Marlin."
> Which marlin version is? i am using 2.5.0.3

yes if you read the whole conversation, its my config for 2.0.3 i had to roll back to it because of various issues with bltouch on 2.0.5.3 such as probe inaccuracy and it missing probe points.

HVJ17, 2020-04-24:
Try these.
https://github.com/HVJ17/Ender-3-Pro-Bltouch-BTT-Mini-E3-V1.2
Marlin 2.0.5.3

archerbobby, 2020-04-24:
> Try these.
> https://github.com/HVJ17/Ender-3-Pro-Bltouch-BTT-Mini-E3-V1.2
> Marlin 2.0.5.3

how are your pins installed?  Are you using the servo and probe pins or are you plugged into the z stop pins?  I just tried your bin and when I tried to auto home the head shot to the right and shuddered against the rail, and the controls to test the probe didn't work anymore.  I have a ender 3 pro with a skr 1.2 board and a bl touch probe and no other upgrades.  Thanks for working to fix this latest update.

HVJ17, 2020-04-24:
Z-stop

archerbobby, 2020-04-24:
thanks,  so did you change the way the pins connect to now connect as yellow/brown/red, or did you leave it as it was which was yellow/red/brown?

HVJ17, 2020-04-24:
well that depends on the bltouch you received. If you have the wires in the wrong location it will not boot and you will get just the blue LCD. My wires are yellow/red/blue in that order.

archerbobby, 2020-04-24:
nope still didn't work for me, once the head moves to the right side, it likes to stay there.  doing a bed level it goes 3 probes left, 3 probes middle 3 probes right, moves the bed, then 3 probes right, then shudder 3 times (then probes 3 times then shudders again does this 3 times in this position), moves bed and again then shudder 3 times (then probes 3 times then shudders again does this 3 times in this position).

also, when level corners the head moves to the corners but does not probe, just moves with each choice of next corner.

this is a messed up issue.  I tried with both pin set up's now, with the bin's for both pin set ups and the same issue.  argggggg.  going to try the base bin with the bl touch unplugged to see if it goes back to working as a 3d printer again.

HVJ17, 2020-04-24:
This should be probing 25 points starting at front left moving 5 across to the right. Then move the bed forward and probe 5 more back to the left. 
Sounds like you have x and y motor wire crossed.

archerbobby, 2020-04-25:
thats what I thought but I double checked the wires a few times against the bigtree site, here is a pic of the wire set up as well.  
[image]

HVJ17, 2020-04-25:
The motors not the limit switches. Other side of the board.

archerbobby, 2020-04-25:
oh they are correct, I was printing with it for 4 days before I got the bl touch and everything worked fine.  it all went to hell right after I installed the bl touch yesterday and used the new firmware 2.05

HVJ17, 2020-04-25:
Do an init eeprom from the panel.
Place the head in the middle and turn it on. Do not auto home. Then use the panel to move the x axis. Does it work correctly?

archerbobby, 2020-04-25:
yup, i cam move the head around by move axis on the panel and it works well, unless I auto home it, then that become 0 and I can no longer move it to the left

HVJ17, 2020-04-25:
I just loaded the same file onto my Ender. Printing right now. I of course have the bltouch in the zstop.
Test the stops by manually activating them during auto home.

archerbobby, 2020-04-25:
the stops using the probe?  it stopped it from moving down.

HVJ17, 2020-04-25:
no. put the carriage in the middle up high. Then tell it to auto home. Before it reaches x-stop, use your finger to push the x-stop,

HVJ17, 2020-04-25:
Send: N12667 M119*46
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: open
Recv: z_min: TRIGGERED

archerbobby, 2020-04-25:
just did a check on the wires and 1 of the leads on the x axis trigger wire is dead

11chevelle11, 2020-04-27:
> Try these.
> https://github.com/HVJ17/Ender-3-Pro-Bltouch-BTT-Mini-E3-V1.2
> Marlin 2.0.5.3

You got that thing going 100mph on bed leveling dont you?!?! :D

11chevelle11, 2020-04-27:
> Try these.
> https://github.com/HVJ17/Ender-3-Pro-Bltouch-BTT-Mini-E3-V1.2
> Marlin 2.0.5.3

Also I'm getting an error on M503 in pronterface. Any clue why it would do that?

HVJ17, 2020-04-27:
I'd say it's a very slow bed leveling procedure. I set it up that way so I don't need to do it at the beginning of every print job. I only do it once every few days.

I believe in m503 is disabled in my settings. It saves memory. I've done a lot with custom boot animation as well as had all the games that enabled on more than one occasion.

roomme13, 2020-04-28:
> yes if you read the whole conversation, its my config for 2.0.3 i had to roll back to it because of various issues with bltouch on 2.0.5.3 such as probe inaccuracy and it missing probe points.

@GhostlyCrowd 
What kind of issue did you have in 2.0.5.3?
i think i face same issue of inaccuracy in latest Marlin.
The problem is not when you leveling. but when homing for me. its either higher or lower the endstop. it make the whole leveling mess up.
after i level my corner with steel feeler, and did 7x7 probe.
this is mesh visualizer.
[image].

GhostlyCrowd, 2020-04-28:
> @GhostlyCrowd
> What kind of issue did you have in 2.0.5.3?
> i think i face same issue of inaccuracy in latest Marlin.
> The problem is not when you leveling. but when homing for me. its either higher or [...]

Probe failing to deploy some times, probing inaccuracy, such as out of a 5x5 grid one random point was always a black hole on the mesh (way lower then the rest of the bed)  failing to probe and skipping probe points completely.  and when it did probe it was not respecting the mesh generated. (the nozzle was way too high or driving to the bed) rolling all the way back to 2.0.3 fixed all these issues immediately on both Minis on ender 3's i deployed it on.

roomme13, 2020-04-28:
> Probe failing to deploy some times, probing inaccuracy, such as out of a 5x5 grid one random point was always a black hole on the mesh (way lower then the rest of the bed) failing to probe and [...]

Same here, imagine probe 7x7 in the middle it have blackhole and at the end failed probe. of course you can always reprobe. then the main problem came, when it failed when homing, it just suck! 
i thought at the first my BLtouch defect. 
Maybe its not bad idea back to 2.0.3.

HVJ17, 2020-04-28:
Is that octoprint? You need to change the axis settings.   Its just visual but you do not have 0,0 in the lower left. Probe point 00 and 11 and 22 and so on should match up on the image. This will not effect how the printer uses the mesh as this is an external representation.

Also, check the probe. I've bent mine. It comes out easily. 
From antclabs:
"If push-pin deploy fails, turn the core by up to 180 degrees with an Allen-key so that the core is further inside the casing."
Make sure you have some delay for the sensor
//#define BLTOUCH_DELAY 500 
or
start code:
G4 P100

paul-1, 2020-04-28:
Working for me,  but I'm building my own firmware.  Been running 2.0.5.3, for a week, and I just updated to 2.0.5.3 + Bugfix.

I used this guide to setup my Configs initially.
https://www.reddit.com/r/ender3/comments/e894j7/marlin_20x_guide_for_ender_3_using_skr_mini_e3_v12/

roomme13, 2020-04-29:
@HVJ17 
No, i used another visualizer, 00 or 06 same, the value still have range +1 and -1
is that normal? i mean, i dont have any flat area, 
but when manual level the corner with steel feeler, is perfectly flat. 

And the problem is when homing, lets say you have good ABL mesh.
when you homing the probe stow at certain distance after pin touch the bed right? then you have 0 Z that ABL will compensate with the mesh.
But when there's an issue with probe while homing, that the pin stow at more/less distance than it should be. your ABL mesh will useless since your Z0 in the wrong place.

I use 500 Delay in marlin and already try to adjust the BLtouch Hex screw., still same result.

11chevelle11, 2020-04-29:
> I'd say it's a very slow bed leveling procedure. I set it up that way so I don't need to do it at the beginning of every print job. I only do it once every few days.
>
> I believe in m503 is disabled [...]

Then something must be wrong for me because z is moving ridiculously fast during level. I have tried lowering the yx_probe_speed to 6000 which didnt help. (I know thats not z) I'm just sort of shooting in the dark here to try to figure out the issue.

HVJ17, 2020-04-29:
Did you edit your own configs, build from files, or load the firmware.bin? I suggest doing an EEPROM init.  The EEPROM is kept through loading firmware from the previous working firmware. Z should be moving at 20mm/s. You can slow this down. The default was 4. Makes  a 25 point probe take way to long.

// X and Y axis travel speed (mm/m) between probes
#define XY_PROBE_SPEED 10000

// Feedrate (mm/m) for the first approach when double-probing (MULTIPLE_PROBING == 2)
#define Z_PROBE_SPEED_FAST HOMING_FEEDRATE_Z

// Feedrate (mm/m) for the "accurate" probe of each point
#define Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST / 2)

#define HOMING_FEEDRATE_Z  (20*60) //this is 20mm/s

This is the speed the z should move during home and probing.

11chevelle11, 2020-04-30:
> Did you edit your own configs, build from files, or load the firmware.bin? I suggest doing an EEPROM init. The EEPROM is kept through loading firmware from the previous working firmware. Z should be [...]

I got it figured out. It was Homing Feedrate. I slowed it down to 10. I was getting inconsistent meshes with it at 20. Thanks for the advice man. You've been a huge help.