bigtreetech/BIGTREETECH-SKR-mini-E3 #229: [FR] BTT laughs at us

JPCossu, 2020-04-24:
Good morning, everyone,
Does BTT check its firmware before offering it, it's incredible there's never a .bin that works properly.
Following the replacement of my Mini E3 (TMC X HS) I wanted to try the new firmware BLTouch Zhoming.
The problem is that my ZOOFSET PROBE doesn't work, the Z doesn't go up or down preventing the correct adjustment of the tray and the first layer.
Also I realize that BTT was not able to enter the right values of the tray (220 instead of 235).

In short it drives me crazy, I don't know how to compile firmware (I'm not the only one) and it should be simple for a beginner to take a working firmware.

I don't even know where to learn how to compile a FW.

Anyway if someone has a recent FW that works well on an original ender with a BLTouch installed in its original position and wants to share its .bin I would be super happy. 

Take care of yourself

Translated with www.DeepL.com/Translator (free version)

GhostlyCrowd, 2020-04-24:
You should learn how to compile, its not plausible for them to give a precompile firmware for a multi purpose board and expect it to work in every single application it can physically fit. 

If you are capable of reading this comment then your capable of reading the Marlin how-to on VS Code compiling Marlin from source. its literally a half dozen mouse clicks and your set up.

eiscorp, 2020-05-01:
bonjour,
ce que tu dit est complétement faut et les 3 pouces avec aussi ! Car vous êtes bien dans la section "BIGTREETECH-SKR-mini-E3" et E3 est pour ???? ENDER 3,  donc il n'y a pas de multiple carte mère !

Pour la compilation, si c'est si compliquer je pense que c'est parce que beaucoup monnayent ce qu'il imprime et qu'ils ne veuillent pas que se soit trop simple pour le commun des mortelles ou encore que tu soit obliger de passer par les fablabs !
En fait l'ideal est que tu prenne 2 ou  3 heures afin de lire le FW en entier, afin de voir ttes les options.
ensuite install visual studio code : https://code.visualstudio.com/docs/?dv=win , puis une fois dedans install platformio. 
ensuite tu doit definir :
MIN_SOFTWARE_ENDSTOPS -z offset
NOZZLE_TO_PROBE_OFFSET { -44, -6, 0 } // values for BLTOUCH on stock Ender 3 pro
define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN //assume using the probe as end stop
define BLTOUCH
define AUTO_BED_LEVELLING_BILINEAR
define Z_SAFE_HOMING
define AUTO_BED_LEVELING_BILINEAR

Vu que le precedent FW ne fonctionnait pas, j'ai laché l'affaire et comme il y a eu le confinement, on peu comprendre que bbt avait d'autre chat à fouetter ! je retente cet aprem et si c bon jle met en ligne et je reviendrai ici !
ps: je vois pas l'interet de mettre [fr], si tu parle en anglais ! J'aime pas l'anglais, n'ayez pas honte votre langue !

eiscorp, 2020-05-01:
[firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4565273/firmware.zip)

GhostlyCrowd, 2020-05-01:
> bonjour,
> ce que tu dit est complétement faut et les 3 pouces avec aussi ! Car vous êtes bien dans la section "BIGTREETECH-SKR-mini-E3" et E3 est pour ???? ENDER 3, donc il n'y a pas de multiple [...]

The board is not JUST for the Ender 3. It can be used on any printer one wants.  Which is way they give a basic precompiled anything custom has to be done by the end user.

txt4nk, 2020-05-14:
With 3D printing, coding is something comes along with it. Especially when you buy a universal board. It would be a great tool for hou to have in your arsenal. I wouldnt mind compiling a firmware file for you, or even walking you through it so you can see all of the customization and tweaking you can do.

DJbalou, 2020-05-16:
ouaipppp! Étant nouveau dans le domaine j'ai reçu ma première imprimante 3d , Anet A8 Plus, TOUT EN PIÈCES ... aoutch! 6 heures avec un pot de tylenol et une bouteille d'eau local fermer pour juré la machine comme bon me sembles. Je vous jure même pour 10 000K je ne monterai pas une 3d print pour personne. bref, déja un chialleux quebecois comme je suis! je me lance dans un casse tête plus fort, Tant qua y être! pourquois pas ce compliquer la vie quand sa ne peux qu'être si simple. Je me commande un board BIGTREETECH SKR MINI E3 V1.2 32 bits avec le Lcd qui vas avec touch screen. OUI OUI !!! j'ai juste fini today de codé mon board Anet V1.7 avec marlin 2.0 bugguer rare. qui manquais des options au controleur lcd, etc etc breff encore du chiallage! ... mais que me direz vous ??? POURQUOIS TU TE LANCE DANS ÇÀ???? 
le merveilleux monde de l'informatique c'est comme une femme ... si tu n'aime pas les problemes..... oublie sa .. garde tes mains dans tes poches! lache le clavier ( la femme y too) ^^ :)
so.. sur cette note ! voici mon premier post sur se site!
TADAM!!