bigtreetech/BIGTREETECH-SKR-mini-E3 #231: LCD keeps changing from home screen to options

CrossMan92, 2020-04-25:
When I send something to print my ender 5 with stock screen keeps beping and changing from the home screen to settings.
Any way to fix it?

mad73923, 2020-04-25:
I had the same problem.
I updated to d983789d72edfa047e4dd1c96eabf5124b088cb2 and the error never happened again.

CrossMan92, 2020-04-25:
> I had the same problem.
> I updated to d983789<https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/commit/d983789d72edfa047e4dd1c96eabf5124b088cb2> and the error never happened again.

Hello
Thank you your answer. But can you explain me what is d983789?

Thank you

Enviado do Outlook Mobile<https://aka.ms/blhgte>

mad73923, 2020-04-30:
Thats the commit hash which points to the state / version of source at which the LCD error was fixed for me.
This version is from December 5th 2019, so its pretty old. If you checkout the current master and build the binary, you'll get a newer version than this.

CrossMan92, 2020-04-30:
I'm have updated my marlin with the last marlin updates.
By the way do you have any problem with bltouch, by bl touch doesn't read well the mesh of the bed. Sometimes I am printing and the nozzle dives in the bed. I have antclabs bl touch V 3.1

Enviado do Outlook Mobile<https://aka.ms/blhgte>