bigtreetech/BIGTREETECH-SKR-mini-E3 #233: SKR Mini E3 V1.2 doesn't recognize SDCard after using an SDCard Adapter

shumakerr, 2020-04-26:
### Description

I used an SDCard Adapter for the microSD Card. (https://www.amazon.com/gp/product/B07PY7QCWV/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1)

Using the SDCard Adapter, it did not recognize the microSD Card. I removed the adapter and now the microSD card isn't recognized by the board. "Attach Media" does not work now and I am not able to flash my firmware.

I have Marlin 2.0.5.3 flashed right now and my SDCARD_CONNECTION to ONBOARD.

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

alloreq, 2020-05-04:
Hi,
I have te same problem, when I use the adapter, skr "see" the card (i tried few different card) but when I remove the adapter, skr don't see my microsd card's no more.
Did you solve it?

shumakerr, 2020-05-04:
Yes, I gave in and de-soldered the SDCard in the board and found one of the pins was broken off. I ordered another SKR mini card and threw the adapter in the trash. Using the system right now as-is with not being able to modify firmware.

KirovAir, 2020-10-22:
@shumakerr Dit you ever repaired the board? I think I have the exact same problem..

alloreq, 2020-10-24:
Yes i did, one of the pins inside the sd card slot was broken, i resolder
new sd card slot and workig good until now.