bigtreetech/BIGTREETECH-SKR-mini-E3 #234: My 3D printer is just ignoring what my BLtouch measures

Reznik007, 2020-04-27:
My 3D printer is just ignoring what my BLtouch measures. I have Ender 3 with BTT SKR mini e3 v1,2 and newest firmware for using BLtouch also as Z homing a I think that printer just use first Zmin and print on that measure.
I´m sorry for my english, but I hope I described my problem well.

If there is someone who could share to me his own working firmware on Ender 3 with BLtouch I will be very excited.
Thanks a lot for every response.

lsn1perl, 2020-04-27:
Following as well as I am curious. Mine probes the center. It also probes the bed 9 times after the first middle probe but judging from my first layer, I'd say it isn't taking into account the 3x3 grid leveling.

Reznik007, 2020-04-27:
Yep same what I´m saying. So it isn´t only my problem. So we have good chance to some repair XDD

Thinkersbluff, 2020-04-27:
Is Bed Leveling "On" , after you probe the bed?
What Probe Z offset value are you using?
 If you just level the bed and then edit the mesh, do the values all look similar?  Do those values change if you probe a second time?
I am not having any problems now, but I have seen my extruder hit a clip as it probed the back corners and that sometimes stored a bad value in that corner.
Are you still on an older version of Marlin, with the EEPROM emulation bug?

Reply-to: reply@reply.github.com
Cc: subscribed@noreply.github.com

Reznik007, 2020-04-27:
Yes bed leveling is on. I have marlin 2.0.5.3. Clips aren´t that problem. I don´t quite understand these questions: If you just level the bed and then edit the mesh, do the values all look similar?  Do those values change if you probe a second time?

Would you please explain?
Thanks

HVJ17, 2020-04-27:
What is your start gcode?

This sounds like it might be the issue for you:

Any arguments left out of G29 will use the default values set in Configuration.h.

**G28 disables bed leveling.** Follow with M420 S to turn leveling on, or use RESTORE_LEVELING_AFTER_G28 to automatically keep leveling on after G28.

You need to either use M420 S or start code or set your config to have RESTORE_LEVELING_AFTER_G28 .

This is a good order if you cannot edit the firmware.
G28
G29
M500
M420 S1

I

Reznik007, 2020-04-27:
This is my gcode script:
 ; Ender 3 Custom Start G-code
G92 E0 ; Reset Extruder
G28 ; Home all axes
G29 ; BLTouch mesh generation
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish

Thinkersbluff, 2020-04-27:
Sure - if you think the board is not storing the bed leveling values, I wondered whether they were changing each time you level.  I expect there would be slight differences, confirming at least the probe is being read into the mesh.
If some values are significantly different from others, it might indicate that the probe is not triggering at a consistent distance from the bed, as I have seen when my extruder sits down on a bed clip while probing.
If the bed corners are not levelled, the probe may not compensate sufficiently.
If the probe Z offset is not calibrated correctly, the probe will set the nozzle gap incorrectly.
If the probe is wired to the board as a probe, not a Z limit switch, the probe may be confounded by hitting the z limit before it touches the bed.  Shooting in the dark, here, hoping something proves helpful, even if I get the specifics wrong...

Cc: badback42@outlook.com; comment@noreply.github.com

Reznik007, 2020-04-28:
If I level the bed measurements are every time different. So what can I do with that?

Reznik007, 2020-04-28:
I don´t know why but I´m not able to level my left upper corner, every time when I try level it It´s still different from other corners. Although I level the bed manually.

Thinkersbluff, 2020-04-28:
Since the mesh values are changing, and there are no probe errors, It sounds like the probe itself is ok.  If you have the probe test on your menu, I would try that also, but it is probably ok.
If you cannot level one corner of the bed, you need to solve that first, before the BLTouch can help.

Reznik007, 2020-04-28:
And do you have idea how I should solve that corner? Because I don´t XD 

And btw I really much appreciate your help m8.

Q-Diddy, 2020-04-28:
I am experieing the same results since updating to 2.0.5.3 Marlin from 2.0.X. I am running a bltouch with a skr mini E3 on ender 3. New z offset is not adhering when print starts. Restarted printer and same result.

Thinkersbluff, 2020-04-28:
Corners go up when spring tension is released and down when  the spring is compressed.  Are you compressed so far there is no adjustment room left, and the corner is too high, or so uncompressed there is no tension left to raise the corner, so the corner is too low?

Reznik007, 2020-04-28:
Nope, when I move with wheel on left down side the left upper side is moving too or I don´t know. I will try get some new springs and some mirror instead of that glass I get from stock and It must be solved. If not there is some problem in firmware for 100%.

HVJ17, 2020-04-28:
This is expected. Moving any one of the wheels moves the entire platform. This is why it's best level the bed at your printing temperature with no filament loaded. Just everything heated.

Reznik007, 2020-04-28:
Ok thanks and that gcode is right?

Reznik007, 2020-04-29:
I used different firmware from HVJ17 and now is my bed leveling and I have best first layers.

I thank to everyone who tryes to help me and have a nice day.
Reznik007

closed 2020-04-29 by Reznik007